snabbco/snabb #346: Optimize straightline branch

lukego, 2015-01-20:
Goal: Optimize the [straightline](https://github.com/virtualopensystems/snabbswitch/tree/straightline) branch to have performance that is consistently competitive with master.

* Key benchmark: `loadgen-snabb-nic-guest` to forward 256-byte packets `NIC -> Snabb -> VM -> Snabb -> NIC`. Target is 10G line rate of 4.4 Mpps when running 3 parallel instances on the same NUMA node of grindelwald.
* Additional benchmark: iperf runs in the NFV application selftests to be consistent with master.

Current state at the time of writing:

* Benchmark is functional on branch. `loadgen` currently broken; using the one from master.
* Variance in performance that is not accounted for.
* JIT behavior is suboptimal (~15% interpreted + ~5% GC)
* Snabb Switch `struct link` and QEMU Virtio-net `vring` both reduced from 8192 elements to 256. Operational impact not fully understood yet. Could reduce throughput by starvation or cause drops by overflow. (This is both to reduce the cache footprint of the working set and to match upstream QEMU. @n-nikolaev will post the branch.)
* Many other parts of Snabb Switch are broken on this branch due to `struct packet` redesign.

Questions? ask here and @n-nikolaev has answers :-).

Fixes should be Pull Requested to the branch cited above for @n-nikolaev to merge into the straightline branch. Once everything is working he will pull request it to master.

n-nikolaev, 2015-01-20:
The "recommended" QEMU version to be used is this one. It is basically current upstream with a single patch that helps reconnecting vhost-user.

https://github.com/virtualopensystems/qemu/tree/straightline

dpino, 2015-01-20:
I'm having some trouble running ```loadgen```.

* I added the remote ```vosys``` in my snabbswitch cloned repo.
* I checked the ```vosys/straighline``` branch and built ```snabb```.
* If I understood correctly, I should now switch to ```snabbco/master``` to run ```loadgen```.
* I run ```loadgen``` but it gets stalled.

```
SNABB=./scripts/bench_env/../..//snabb
Locking 0000:01:00.0
Locking 0000:01:00.1
Connect to guests with:
telnet localhost 5500
```

It also gets stalled even if I build ```snabb``` from ```snabbco/master``` and run ```loadgen```. An older master works though, so I discard it's a problem with the NIC.

With regard to QEMU I'm using ```/home/dpino/workspace/snabbswitch/src/snabb```, which is a copy of ```/home/luke/hacking/snabb-nfv/qemu/x86_64-softmmu/qemu-system-x86_64```.

n-nikolaev, 2015-01-20:
`straightline` `loadgen` is broken.You must use the master one and run it by hand in another console (`tmux`, `screen` etc.). I use it like this:
`sudo /home/luke/hacking/pristine/snabbswitch/src/snabb /home/luke/hacking/pristine/snabbswitch/src/digns/loadgen/loadgen /opt/bench/256.pcap 0000:82:00.1`

If you want to use the integrated bench script `src/scripts/bench_env/loadgen-snabb-nic-guest.sh`

You'll need to change your `~/bench_env.conf`:
`RUN_LOADGEN=    `

dpino, 2015-01-20:
Thanks n-nikolaev, that worked. I'm still a bit confused, shouldn't I run the snabb that I just build from the straightline branch, instead of /home/luke/hacking/pristine/snabbswitch/src/snabb, if I want to try the changes in the straightline branch?

n-nikolaev, 2015-01-20:
dpino, can you please post full configs and step by step what do you do to run the test? Lets figure it out together.

andywingo, 2015-01-20:
I should note that we found that the precise way in which you iterate through a pcap file can affect results.  In pflua we now use this helper: https://github.com/Igalia/pflua/blob/master/src/pf/savefile.lua#L64

It builds and returns a regular Lua table of packet pointers.

dpino, 2015-01-20:
OK, here's what I'm doing.

First about the master branch. Currently I'm working in grindelwald, using NIC 0000:01:00. I have a cloned repo of ```snabbco/snabbswitch``` at ```/home/dpino/workspace/snabbswitch/src```. I have a local branch (master-old) pointing to a an old commit in master. If I build and run loadgen-snabb-nic-guest.sh on that branch it works:

```
snabbswitch/src (master-old) $ make
snabbswitch/src (master-old) $ sudo ./scripts/bench_env/loadgen-snabb-nic-guest.sh

SNABB=./scripts/bench_env/../..//snabb
Locking 0000:01:00.0
Locking 0000:01:00.1
Connect to guests with:
telnet localhost 5500
On 0000:01:00.0 got 3.636
Rate(Mpps):	3.636
Waiting QEMU processes to terminate...
Finished.
```

This is my ~/bench_conf.sh file.

```
RUN_LOADGEN=true

# Snabb NFV traffic process configuration file
NFV_CONFIG=/home/dpino/workspace/snabbswitch/src/test_fixtures/nfvconfig/test_functions/snabbnfv-bench.port

# Trace file defining which packets to replay as a traffic source.
PCAP=/opt/bench/256.pcap

# How many packets to run for the benchmark.
NFV_PACKETS=100e6

# Telnet port numbers for VM serial ports.
# (VMs will use consecutive ports starting from this base)
TELNET_PORT0=5500
TELNET_PORT1=5600

NFV_PCI0="0000:01:00.0"
NFV_PCI1="0000:01:00.1"

GUEST_MEM=1024
QEMU=/home/dpino/workspace/snabb-nfv/qemu/x86_64-softmmu/qemu-system-x86_64

SNABB="/home/dpino/workspace/snabbswitch/src/snabb"

```

If I switch to ```snabbco/master``` and do the same, the benchmark gets stalled (it never ends).

```
snabbswitch/src (master-old) $ make
snabbswitch/src (master-old) $ sudo ./scripts/bench_env/loadgen-snabb-nic-guest.sh

SNABB=./scripts/bench_env/../..//snabb
Locking 0000:01:00.0
Locking 0000:01:00.1
Connect to guests with:
telnet localhost 5500
```

Now, with regard to the ```straightline``` branch, running ```sudo /home/luke/hacking/pristine/snabbswitch/src/snabb /home/luke/hacking/pristine/snabbswitch/src/designs/loadgen/loadgen /opt/bench/256.pcap 0000:01:00.1``` works, so that should be the expected result.

```
loadgen
oink
scan
scanning
scanned
scanned
0000:01:00.1
conf
waiting...
main
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	3,313,786	GOTCL (TX octets)	861,482,440
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	4,464,796	GOTCL (TX octets)	1,160,833,440
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	4,463,589	GOTCL (TX octets)	1,160,530,800
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	4,464,251	GOTCL (TX octets)	1,160,719,820
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	4,463,976	GOTCL (TX octets)	1,160,622,580
```

I added remote ```vosys``` (https://github.com/virtualopensystems/snabbswitch.git) and checked out branch vosys/straightline. I built snabb in that branch and run the master loadgen script:

```
snabbswitch/src (straightline) $ make
snabbswitch/src (straightline) $ sudo ./snabb /home/luke/hacking/pristine/snabbswitch/src/designs/loadgen/loadgen /opt/bench/256.pcap 0000:01:00.1
...hacking/pristine/snabbswitch/src/designs/loadgen/loadgen:4: module 'core.buffer' not found:
	no field package.preload['core.buffer']
	no file './core/buffer.so'
	no file '/usr/local/lib/lua/5.1/core/buffer.so'
	no file '/home/dpino/workspace/snabbswitch/deps/luajit/usr/local/lib/lua/5.1/core/buffer.so'
	no file '/usr/local/lib/lua/5.1/loadall.so'
	no file './core.so'
	no file '/usr/local/lib/lua/5.1/core.so'
	no file '/home/dpino/workspace/snabbswitch/deps/luajit/usr/local/lib/lua/5.1/core.so'
	no file '/usr/local/lib/lua/5.1/loadall.so'
stack traceback:
	core/main.lua:185: in function <core/main.lua:183>
	[C]: in function 'require'
	...hacking/pristine/snabbswitch/src/designs/loadgen/loadgen:4: in main chunk
	[C]: in function 'dofile'
	core/main.lua:109: in function <core/main.lua:50>
	[C]: in function 'xpcall'
	core/main.lua:190: in main chunk
	[C]: in function 'require'
	[string "require "core.main""]:1: in main chunk
```

n-nikolaev, 2015-01-20:
You'll need to change your ~/bench_env.conf:
`RUN_LOADGEN=   `

i.e. unset the bench env to run loadgen from its own tree (which is still broken).
The lukes pristine loadgen you shoudl ron on another consle and leve it there running while you;re testing , no need to kill it every time.

lukego, 2015-01-20:
@n-nikolaev is it correct that you can leave both loadgen and the guest running and only restart the Snabb Switch process? (I mean: it will be happy to keep connecting to the same loadgen and guest?)

I have been restarting the guest in my tests but that may be overkill.

n-nikolaev, 2015-01-20:
Yes you can - loadgen is made to be untoched (remeber wht we did in DT - external loadgen)
Just unset `RUN_LOADGEN` in your `bench_env.conf`.
As for the guest - also true, that is why we have the _reconnect_ feature.

dpino, 2015-01-20:
@n-nikolaev OK, cool, thanks.

So doing a recap this is what I should do:

1) In a different terminal, run ```sudo /home/luke/hacking/pristine/snabbswitch/src/snabb /home/luke/hacking/pristine/snabbswitch/src/digns/loadgen/loadgen /opt/bench/256.pcap 0000:82:00.1```. This generates traffic.

2) In another terminal, from the ```straightline``` branch and run ```sudo ./scripts/bench_env/loadgen-snabb-nic-guest.sh```. The var ```RUN_LOADGEN``` in ```bench_env.conf``` should be empty because there's another snabb instance running loadgen.

n-nikolaev, 2015-01-20:
Yes - exactly.
Actually, a working `loadgen` on `straightline` is near completion, may be a quick review and will be integrated soon.

dpino, 2015-01-20:
@n-nikolaev OK, I got it running, but I'm not sure whether is working OK.

In one terminal I got loadgen running:

```
Transmissions (last 1 sec):
0000:84:00.1	TXDGPC (TX packets)	4,464,328	GOTCL (TX octets)	1,160,723,200
Transmissions (last 1 sec):
0000:84:00.1	TXDGPC (TX packets)	3,704,678	GOTCL (TX octets)	963,222,780
Transmissions (last 1 sec):
0000:84:00.1	TXDGPC (TX packets)	4,463,986	GOTCL (TX octets)	1,160,636,620
Transmissions (last 1 sec):
0000:84:00.1	TXDGPC (TX packets)	4,464,379	GOTCL (TX octets)	1,160,733,080
Transmissions (last 1 sec):
```

In another one, I launch `sudo ./scripts/bench_env/loadgen-snabb-nic-guest.sh` but it never ends, I never got a summary report, as I used to.

I'm using 0000:84:00 at this moment cause either 0000:01:00 and 0000:03:00 seemed not to work. I got the following while running loadgen in 0000:01:00 (```sudo /home/luke/hacking/pristine/snabbswitch/src/snabb /home/luke/hacking/pristine/snabbswitch/src/designs/loadgen/loadgen /opt/bench/256.pcap 0000:01:00.1```):

```
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
```

Here's my ```~/bench_cong.sh```:

```
RUN_LOADGEN=

# Snabb NFV traffic process configuration file
NFV_CONFIG=/home/dpino/workspace/snabbswitch/src/test_fixtures/nfvconfig/test_functions/snabbnfv-bench.port

# Trace file defining which packets to replay as a traffic source.
PCAP=/opt/bench/256.pcap

# How many packets to run for the benchmark.
NFV_PACKETS=100e6

# LuaJIT profiler settings
# export NFV_PROF=z2F
# export NFV_DUMP=+rsT

# Telnet port numbers for VM serial ports.
# (VMs will use consecutive ports starting from this base)
TELNET_PORT0=5500
TELNET_PORT1=5600

# Port(s) to use for VMs
NFV_PCI0="0000:84:00.0"
NFV_PCI1="0000:84:00.1"

# Uncomment these to test with many ports in parallel instead:
# NFV_PCI0="0000:01:00.0 0000:03:00.0" # Not working
# NFV_PCI1="0000:01:00.1 0000:03:00.1" # Not working

GUEST_MEM=1024
QEMU=/home/dpino/workspace/snabb-nfv/qemu/x86_64-softmmu/qemu-system-x86_64

SNABB="/home/dpino/workspace/snabbswitch/src/snabb"
```

Sorry for the inconvenience.

n-nikolaev, 2015-01-21:
As Luke proposed:

    Nikolay: 82 84 88
    Diego: 01 03
    Luke: 05

I don't mind share some of the boards though.
Note that yesterday I was running test on 82 too, so maybe conflicts were the cause of the behaviour.
I can leave 82 for you if it suits you best - no problem for me.

dpino, 2015-01-21:
@n-nikolaev Sorry, I was using it temporary for testing. I'd consider using 82 but I need to know whether 01 and 03 are working correctly. This is what I'm getting:

```
sudo /home/luke/hacking/pristine/snabbswitch/src/snabb /home/luke/hacking/pristine/snabbswitch/src/designs/loadgen/loadgen /opbench/256.pcap 0000:0X:00.X
```

**0000:01:00.1**

```
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
Transmissions (last 1 sec):
0000:01:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
```

**0000:01:00.0**

```
Transmissions (last 1 sec):
0000:01:00.0	TXDGPC (TX packets)	3,315,437	GOTCL (TX octets)	861,911,180
Transmissions (last 1 sec):
0000:01:00.0	TXDGPC (TX packets)	4,464,610	GOTCL (TX octets)	1,160,783,260
Transmissions (last 1 sec):
0000:01:00.0	TXDGPC (TX packets)	4,463,673	GOTCL (TX octets)	1,160,550,560
```

**0000:03:00.1**

```
Transmissions (last 1 sec):
0000:03:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
Transmissions (last 1 sec):
0000:03:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
Transmissions (last 1 sec):
0000:03:00.1	TXDGPC (TX packets)	0	GOTCL (TX octets)	0
```

**0000:03:00.0**

```
Transmissions (last 1 sec):
0000:03:00.0	TXDGPC (TX packets)	3,314,888	GOTCL (TX octets)	861,769,740
Transmissions (last 1 sec):
0000:03:00.0	TXDGPC (TX packets)	4,464,237	GOTCL (TX octets)	1,160,702,920
Transmissions (last 1 sec):
0000:03:00.0	TXDGPC (TX packets)	4,463,963	GOTCL (TX octets)	1,160,608,280
```

n-nikolaev, 2015-01-21:
OK so the `straightline` branch now has all Luke's updates + a small patch from me.
This effectively makes loadgen working in the tree. 
 
 * Please update and remove  your `RUN_LOADGEN= ` line from the `bench_env.conf`.
 * Use 82 - I dropped it from my PCI list
 * can you please verify also this issue https://github.com/SnabbCo/snabbswitch/issues/347
 for me it means that changing the file `src/test_fixtures/nfvconfig/test_functions/snabbnfv-bench.port` from `port_id = "A",` to `"port_id = "a",` gives me better numbers in the tests.

dpino, 2015-01-21:
@n-nikolaev OK, done. Thanks. I added a comment in https://github.com/SnabbCo/snabbswitch/issues/347.

lukego, 2015-03-01:
Straightline branch was successfully optimized to the level of master before merge.

closed 2015-03-01 by lukego