snabbco/snabb #347: [straightline] Link name performance impact

n-nikolaev, 2015-01-20:
Using the `straightline` branch to run performance tests.

`snabbnfv-bench` script runs slower if the `port_id` is set to certain names.h

EXAMPLE:
Using `src/test_fixtures/nfvconfig/test_functions/snabbnfv-bench.port` with `./src/scripts/bench_env/loadgen-snabb-nic-guest.sh`. I run this 10 times and here are the results.

    Rate(Mpps):     2.677
    Rate(Mpps):     2.69
    Rate(Mpps):     2.846
    Rate(Mpps):     2.697
    Rate(Mpps):     2.674
    Rate(Mpps):     2.73
    Rate(Mpps):     2.716
    Rate(Mpps):     2.816

I then change the `port_id` in `src/test_fixtures/nfvconfig/test_functions/snabbnfv-bench.port`  from 
`port_id = "A",` to `port_id = "a",` and I get generally higher results:

    Rate(Mpps):     3.453
    Rate(Mpps):     3.39
    Rate(Mpps):     3.392
    Rate(Mpps):     3.426
    Rate(Mpps):     3.382
    Rate(Mpps):     3.446
    Rate(Mpps):     3.537
    Rate(Mpps):     3.504

My impression is that since `port_id` is part of the link name, it influences the performance (hash coliisions etc.) when accessing the links by name.

dpino, 2015-01-21:
Confirmed, I get similar results:

**A (uppercase)**

```
Rate(Mpps):	2.74
Rate(Mpps):	2.843
Rate(Mpps):	2.737
Rate(Mpps):	2.835
Rate(Mpps):	2.723
```

**a (lowercase)**

```
Rate(Mpps):	3.555
Rate(Mpps):	3.527
Rate(Mpps):	3.463
Rate(Mpps):	3.571
Rate(Mpps):	3.544
```

dpino, 2015-01-23:
Since yesterday I cannot manage to run loadgen in grindelwald. To discard there was something wrong with my local branch I switched to vosys/straigtline.

First, I selected a NIC it's guaranteed to be working fine:

```bash
$ sudo SNABB_TEST_INTEL10G_PCIDEVA=0000:05:00.0 SNABB_TEST_INTEL10G_PCIDEVB=0000:05:00.1 ./snabb -t apps.intel.intel_app

link report
1177760 sent on nicBm1.tx -> sink_ms.in3 (loss rate: 0%))
2 sent on source_ms.out -> repeater_ms.input (loss rate: 0%))
0 sent on nicAs.tx -> sink_ms.in1 (loss rate: 0%))
2355945 sent on repeater_ms.output -> nicAs.rx (loss rate: 0%))
1177727 sent on nicBm0.tx -> sink_ms.in2 (loss rate: 0%))
selftest: ok
```

I checked ixgbe driver is not loaded.

```bash
$ lsmod | grep ixgbe
```

Apparently there's no other snabbswitch instance running in the machine, so I discard the NIC is being used by another process.

Here's my ~/bench_conf.sh

```bash
PCAP=/opt/bench/256xff.pcap

NFV_CONFIG=/home/dpino/workspace/snabbswitch/src/test_fixtures/nfvconfig/test_functions/snabbnfv-bench.port

NODE_BIND0=0
SNABB_LOG0=snabblog.txt
NFV_PCI0="0000:05:00.0"
NFV_SOCKET0=/tmp/dpino_vhost_A.sck

NODE_BIND1=1
SNABB_LOG1=/dev/null
SNABB_LOG1=loadgen.log
NFV_PCI1="0000:05:00.1"
NFV_SOCKET1=/tmp/dpino_vhost_B.sock
NFV_PACKETS=10e6

# QEMU
GUEST_MEM=1024
HUGETLBFS=/hugetlbfs
KERNEL=/opt/bench/bzImage-no-virtio-net
QEMU=/home/dpino/workspace/snabb-nfv/qemu/x86_64-softmmu/qemu-system-x86_64

# Guest instance #0
TAP0=tap0
TELNET_PORT0=7111
GUEST_IP0=192.168.2.10
GUEST_MAC0=52:54:00:00:00:00
IMAGE0=/home/dpino/virt/ubuntu-trusty.img
BOOTARGS0="earlyprintk root=/dev/vda rw console=ttyS0 ip=$GUEST_IP0 hugepages=32"

# Guest instance #1
TAP1=tap1
TELNET_PORT1=7112
GUEST_IP1=192.168.2.11
GUEST_MAC1=52:54:00:00:00:01
IMAGE1=/home/dpino/virt/ubuntu-trusty.img1
BOOTARGS1="earlyprintk root=/dev/vda rw console=ttyS0 ip=$GUEST_IP1"
```

Any hint appreciated. Thanks :)

n-nikolaev, 2015-01-23:
Can you use `82`.

dpino, 2015-01-23:
Same thing. I was using 05 temporally because I got the following result running the intel_app selftest on 82:

```
Transmitting bidirectionally between nicA and nicB
2475 sent on source2.out -> nicB.rx (loss rate: 0%))
0 sent on nicB.tx -> sink.in2 (loss rate: 0%))
0 sent on nicA.tx -> sink.in1 (loss rate: 0%))
2475 sent on source1.out -> nicA.rx (loss rate: 0%))
sq_sq: missing packets
```

The selftest on 05 looked better. Yesterday, when it was working, I was using 82.

n-nikolaev, 2015-01-23:
@lukego any ideas?

dpino, 2015-01-23:
I got it working on 05 by using a ~/bench_conf.sh similar to Luke's. It looks like this:

```bash
PCAP=/opt/bench/256xff.pcap

NFV_CONFIG=/home/dpino/workspace/snabbswitch/src/test_fixtures/nfvconfig/test_functions/snabbnfv-bench.port

# LuaJIT profiler settings
# export NFV_PROF=vl

NODE_BIND0=0
SNABB_LOG0=snabblog.txt
NFV_PCI0="0000:05:00.0"
NFV_SOCKET0=/tmp/dpino_vhost_A.sck

NODE_BIND1=1
SNABB_LOG1=/dev/null
SNABB_LOG1=loadgen.log
NFV_PCI1="0000:05:00.1"
NFV_SOCKET1=/tmp/dpino_vhost_B.sock
NFV_PACKETS=10e6

# QEMU
GUEST_MEM=1024
HUGETLBFS=/hugetlbfs
KERNEL=/opt/bench/bzImage-no-virtio-net
QEMU=/home/dpino/workspace/snabb-nfv/qemu/x86_64-softmmu/qemu-system-x86_64

# Guest instance #0
TAP0=tap0
TELNET_PORT0=5500
GUEST_IP0=fe80::5054:ff:fe00:0
GUEST_MAC0=52:54:00:00:00:00
IMAGE0=/opt/bench/ubuntu-trusty.img
BOOTARGS0="earlyprintk root=/dev/vda rw console=ttyS0 ip=$GUEST_IP0 hugepages=32"

# Guest instance #1
TAP1=tap1
TELNET_PORT1=5600
GUEST_IP1=fe80::5054:ff:fe00:1
GUEST_MAC1=52:54:00:00:00:01
IMAGE1=/media/disk/bench-env-chur/vmb.img
BOOTARGS1="earlyprintk root=/dev/vda rw console=ttyS0 ip=$GUEST_IP1"
```

Interestingly running loadgen using this conf I got the following results:

**port_id: A (uppercase)**

```
Rate(Mpps): 3.631
Rate(Mpps): 3.542
Rate(Mpps): 3.607
Rate(Mpps): 3.543
Rate(Mpps): 3.577
```

**port_id: a (lowercase)**

```
Rate(Mpps): 3.354
Rate(Mpps): 3.406
Rate(Mpps): 3.401
Rate(Mpps): 3.407
Rate(Mpps): 3.435
```

Both results are similar, although A is slightly better, which compare to the previous results is a little bit strange :/

lukego, 2015-01-23:
FWIW: To me it seems entirely possible that the link name issue is really nothing to do with link names but something more nebulous e.g. use of uninitialized memory that is subtly influenced. I would not be surprised if there are 100 different things that can be done to provoke the good/bad behaviour. (I had previously seen significantly lower performance when enabling profiling: could be that this was the same issue and actually nothing to do with profiling directly.)

lukego, 2015-01-23:
FYI: I created accounts for you guys on `interlaken`. I don't see any urgent need to move from grindelwald but it is there if you want to use it.

The accounts should be equivalent to your grindelwald accounts. If you want to create an account for a colleague then please go ahead. To login `ssh -p 2030 lab1.snabb.co`.

lukego, 2015-02-19:
@dpino is this issue resolved now? (That is: does it represent something we need to fix on master?)

dpino, 2015-02-23:
Lowercase / uppercase of port_id is no longer an issue, performance is the same in both cases.

**port_id="a"**
```
Rate(Mpps):	4.424
Rate(Mpps):	4.419
Rate(Mpps):	4.418
Rate(Mpps):	4.38
Rate(Mpps):	4.412
```

**port_id="A"**
```
Rate(Mpps):	4.423
Rate(Mpps):	4.411
Rate(Mpps):	4.428
Rate(Mpps):	4.425
Rate(Mpps):	4.426
```

lukego, 2015-02-26:
Thanks for measuring!

closed 2015-02-26 by lukego