snabbco/snabb #366: sysctl required for 1GB hugepages (bad error message)

lukego, 2015-02-09:
Snabb Switch allocates HugeTLB pages with shared memory system calls. This will fail if the maximum allowed shared memory mapping size is smaller than the HugeTLB page size, which may be the default scenario if your kernel is configured with 1GB huge pages.

The symptom of this problem is that Snabb Switch fails to allocate a HugeTLB page (`shmget(2)` returns `-EINVAL`) and prints this message:

    Failed to allocate a huge page for DMA

and the solution is to increase the shared memory mapping limit with this command:

    sudo sysctl -w kernel.shmmax=1073741824

This is listed as a bug because the error message is unhelpful. Snabb Switch should specifically check for this problem and print a more helpful error message.

n-nikolaev, 2015-02-10:
The current value of `kernel.shmmax` can be read from here `/proc/sys/kernel/shmmax` and then compared to the huge page size.

jeffbrl, 2015-02-10:
I'd like to give this one a shot. I will make a proposal in this issue's comments before writing the code.

jeffbrl, 2015-02-10:
Use this logic after receiving `-EINVAL` from `shmget`.

If `sysinfo`'s `totalmem` struct member < 256 MB:

"Cannot allocate 128 MB of shared memory for snabbswitch. System memory is too low."

else:

"Cannot allocate 128 MB of shared memory for snabbswitch. Increase shared memory using 'sudo sysctl -w kernel.shmmax=1073741824'. Use the '-p' flag or edit sysctl.conf for persistence across reboots."

lukego, 2015-04-29:
Here is another idea:

We could automatically increase the kernel `shmmax` setting and print a message to say that we are doing so. (Kind of like the way we automatically provision new huge pages.) If we are worried about users disliking this then we could add a `--no-change-kernel-settings` argument to print an error in these cases instead of automatically correcting them.

I note that on Ubuntu 15.04 the limit seems to be increased to essentially infinity:

```
$ sudo sysctl -a | grep shmmax
kernel.shmmax = 18446744073692774399
```

PR #469 (2015-05-13): memory: Force the kernel to enable working HugeTLB allocation
If huge page allocation fails then do one of two things:

- Increase sysctl kernel.shmmax if the maximum shared memory
  allocation size is less than the size of one huge page. (That is a
  misconfigured kernel from our perspective.) Fixes #366.

- Increase sysctl vm.nr_hugepages otherwise to have the kernel reserve
  a new huge page that we can allocate. (Try this up to three times
  per allocation, as before.)

To change system-wide sysctl values is a fairly presumptuious thing to
do. For this reason we now print clear messages ab [... cut]
commit cfbebe55 (2015-05-13): memory: Force the kernel to enable working HugeTLB allocation

If huge page allocation fails then do one of two things:

- Increase sysctl kernel.shmmax if the maximum shared memory
  allocation size is less than the size of one huge page. (That is a
  misconfigured kernel from our perspective.) Fixes #366.

- Increase sysctl vm.nr_hugepages otherwise to have the kernel reserve
  a new huge page that we can allocate. (Try this up to three times
  per allocation, as before.)

To change system-wide sysctl values is a fairly presumptuious thing to
do. For this reason we now print clear messages a [... cut]
closed 2015-05-13 by lukego