snabbco/snabb #368: Intel driver selftest reliability

lukego, 2015-02-11:
The Intel10G driver needs to consistently pass selftest on all machines. This issue has been discussed at length on the mailing list, and this Issue is simply to track the lifetime of the bug. Can't wait to close it :-)

Here is a log with 100 runs on the current master on `interlaken`: https://gist.github.com/lukego/bda37923f504e0c4f71f.

lukego, 2015-02-15:
Closing this issue because the fix from Javier works very well for me. On `interlaken` I tested each of the four NICs 20 times and had no failures. (Previously I had many.)

Awesome if that is the end of it!

Please report any suspicious NIC failures ocuring after commit 0f98bd6. cc @javierguerragiraldez @n-nikolaev @dpino @eugeneia

closed 2015-02-15 by lukego

eugeneia, 2015-02-16:
Just ran the selftest 100 times on davos and had 2/100 failures. Big improvements here!

eugeneia, 2015-02-19:
I want to document one very weird (*possibly* Intel_app related) failure case here even though I have no idea why it happens yet:

It happened twice so far: SnabbBot fails the NFV selftest universally because "PING failed". PING usually never fails, it really means packets don't go through the hardware I guess. The weird thing is it all works when I run the selftest manually, only SnabbBot triggers the fail. Which sucks because SnabbBot and me we do the same thing. So there is a chance that I am just really unlucky and never trigger the "timing bug?" manually but SnabbBot hits it consistently??? My solution: Reboot davos, unbind all 10-G nics, try again. So yeah... what?

Maybe someone sees something I don't see here. The fact that rebooting "helps" tells me its hardware related, but that I can't reliably reproduce it (but SnabbBot seems to do) puts me off really.

commit 0f98bd6a (2015-02-15): Merge remote-tracking branch 'origin/pr/373' into intel10g-init

Javier's improvements to 82599 driver initialization reliability.

Fixes #368.