snabbco/snabb #401: Failed to run test_sprayer.lua

ghjung, 2015-03-13:
Hi,

I'm trying to learn and use Snabb Switch as a novice. 
When I follow "Getting Started Guide", I got the message when I do "sudo ./snabb snsh test_sprayer.lua"; 
 "apps/sprayer/sprayer.lua:33: variable 'deref' is not declared"

Then, I found that there is difference between the example "srayer.lua" in document and src/apps/sprayer.lua" in the source. Actually there is no "deref" function in core.packet but sprayer.lua is calling packet.deref().
```lua
function Sprayer:process_packet(i, o)
   local p = link.receive(i)

   -- drop every other packet
   if self.packet_counter % 2 == 0 then
      link.transmit(o, p)
   else
      packet.deref(p)
   end
end
```

It was working when I remove the line to call "packet.deref(p)" .
It might be very minor issue but I'd like let you guys know.

Thanks.

javierguerragiraldez, 2015-03-13:
The packet API has changed significantly not long ago (the `straightline` merge), this line should be changed to `packet.free(p)`.

ghjung, 2015-03-16:
Okay. Are you going to fix it? then I will just close this case. 

Thanks.

lukego, 2015-03-16:
Thanks for reporting! I have made a new issue (#405) to track this.

closed 2015-03-16 by lukego