snabbco/snabb #476: Find an alternative to isolcpus

lukego, 2015-05-13:
Snabb Switch processes generally want their own CPU core to use exclusively. The recommendation is to use the Linux `isolcpus` boot parameter to reserve cores for these processes and then bind to them using `taskset -c <core> snabb ...`. The `isolcpus` parameter ensures that the kernel will not schedule work onto these cores automatically. This works well, but it is not necessarily convenient to have to reboot the kernel with specific CPU reservations to allocate resources for Snabb Switch processes.

Let us find the next best alternative to recommend in cases where the user does not want to reserve a core but does want to achieve close to maximum performance even when the machine is under heavy load.

The options would include:

- Running with high priority on `nice`.
- Running with realtime priority.
- Dynamically reserving cores using Linux scheduler calls.

lukego, 2015-05-14:
Could be that `cpuset(7)`(http://man7.org/linux/man-pages/man7/cpuset.7.html) is a good solution. There is much discussion of this as a potential successor to isolcpus on the interwebs and it sounds like it can cover our use case of excluding other processes (and kernel threads and interrupt handlers) from using "our" cores.

lukego, 2015-05-14:
The kernel documentation: [Documentation/cgroups/cpusets.txt](https://www.kernel.org/doc/Documentation/cgroups/cpusets.txt).

mwiget, 2016-02-22:
A simple but effective temporary solution is the use of tasket on the current shell. Processes spawned from that shell inherit the cpu mask:

```
[mwiget@st:~]$ taskset -cp 14 $$
pid 21198's current affinity list: 0,4-8,12-15
pid 21198's new affinity list: 14

[mwiget@st:~]$ bash

[mwiget@st:~]$ taskset -cp $$
pid 21213's current affinity list: 14
```

I learned this the hard way when I realised that Docker containers ignore the kernels isolcpu settings:

```
[mwiget@st:~]$ taskset -cp $$
pid 21220's current affinity list: 0,4-8,12-15

[mwiget@st:~]$ docker run -ti --name ubuntu ubuntu:14.04.4
root@63c70c480477:/# taskset -cp $$
pid 1's current affinity list: 0-15
root@63c70c480477:/# exit
exit
```

So my solution to the problem is setting the cpu list on the provisioning shell launched within the container, plus set the same taskset in /root/.bashrc, so debugging shells launched into a running container also adhere to the requested cpu affinity:

```
taskset -p $AFFINITY_MASK $$
echo "taskset -p $AFFINITY_MASK \$\$" >> /root/.bashrc
```

BTW Docker has an option to set "--cpuset-cpus=<cpu list>" with "docker run ...", but doing so also prevents the Container to use any cpu outside that list. In my snabbvmx case, I wanted to have qemu and anything but snabb adhere to a specific cpu list, while assigning snabb another cpu.

wingo, 2016-02-22:
I think I would add a "-c" option to Snabb itself (handled in src/core/main.lua), which does a `S.sched_setaffinity` to the core in question.  Initially we can allow just one core in the list.  WDYT?

I also want to investigate the impact of `S.sched_setscheduler(0, "fifo", 1)` at some point, but I need to build some metrics to let me know the impact of this.