snabbco/snabb #480: Choose default directory for runtime files (counters, etc)

lukego, 2015-05-14:
Let us choose a well-defined default naming scheme for temporary data files created by Snabb Switch processes. This will make it easy to store information (e.g. counters from traffic processes) and to read and aggregate it (e.g. from snabbtop #472 or netsnmp).

lukego, 2015-05-14:
For a root directory one idea would be to look at the [Filesystem Heirarchy Standard](http://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard) and see which standard location best matches our requirements. For example, if we would like a ramdisk then `/var/run/` could be suitable.

How about the naming scheme? Subdirectories named by pid, for example?

eugeneia, 2015-05-21:
@lukego I hacked something up that generally works: https://github.com/eugeneia/snabbswitch/compare/snabbtop...eugeneia:snabbtop-fs

It implements the `/var/run/snabb/<pid>/<resource>` scheme. `snabb top` will either select the only running instance or expect the user to specify one (by pid). Both `core.app` and `snabb top` agree on using a shared "resource" ("core-stats").

There is one (recurring) problem though: We need a reliable way to run cleanup code on abnormal exits. In this case its important that the <pid> directory is deleted when `snabb` is killed so that we don't leave behind stale resources. That probably mean reading a `signalfd` during `pace_breathing`?

lukego, 2015-05-21:
I like this scheme.

Perhaps we could even have an optional name that would create/update a symlink? So you could write `snabb top snabbnfv-traffic-01:00.0` to attach to the latest incarnation of the named process. (symlink like `/var/run/snabbnfv-traffic-0 -> 90210`).

Cleanup is interesting. Yes, we need to clean up old files to avoid using unbounded space, particularly on a RAM disk. On the other hand, the state of a given process might actually be especially interesting after it has crashed/terminated. So it might be nice to have a scheme that would somehow expire old directories to keep the space usage within bounds but without being too eager to delete data?

lukego, 2015-05-21:
Here is another idea:

Suppose that we used a symbolic name instead of a PID. Then we could avoid leakage by using a reasonably small set of names. The decision then would be what happens when we start and the directory already exists: do we delete it, archive it, or use it?

eugeneia, 2015-05-22:
I'll play the UNIX's advocate here. I think using the PID (publicly) is smart because the process starting Snabb can get it without much knowledge about Snabb internals. Furthermore the calling process is generally in a better spot to manage the instances resources:

* It can easily tell when Snabb exits
* It can create symlinks for arbitrary purposes (snabb top doesn't care whether the identifying string is a PID or "foo")
* It can archive / snapshot the resources
* Enabling arbitrary management of the instance resources would require as little as an option to `snabb`: "Don't cleanup the instance directory."

While I see the merits of the behaviors you propose, I think they don't need to be implemented by Snabb Switch. I'd say: have a simple and functional default behavior and enable independent extension. The idea here is: Operators love simple interfaces and flexibility.

lukego, 2015-05-24:
There is a precedent for keeping it simple and leaving such details to others: the init scripts. Snabb NFV is not pushing people to use standard startup scripts and instead leaving that to people doing packaging for distributions. They are the experts on how they want things done (sysv, upstart, systemd, logging, file ownership, etc) and we don't care so much at the moment. The best approach seems to be to let people take care of this downstream and then pull good solutions upstream in the future if that makes sense.

I am not sure about this case though. As a user I want a simple and uniform way to monitor and control the Snabb Switch processes that are running. The processes can be started in a bunch of different ways:

1. Throw-away instance started in an interactive shell.
2. Persistent instance started in a shell running under tmux or screen.
3. Started by script e.g. init.d or bench_env.

If we push work onto the code that is starting Snabb Switch processes that it would seem to complicate all of those scenarios and make it harder to control Snabb Switches in a consistent way that is independent of how they started. That sounds negative to me.

I would also like to drastically simplify the bench_env scripts. They are complex and a lot of that comes from boring details like keeping track of which processes have been started and making sure they are terminated together. So if we would do more in the Snabb Switch core it could potentially make these scripts simpler too.

The notion I have in mind is a bit like the Erlang `-sname foo` argument that assigns a name to a process that is unique to the local machine. That seems to work fairly well in practice.

lukego, 2015-05-25:
Here is something I did not really consider: it could be that it is hard to automatically choose a suitable name.

For example, if the traffic process would self-assign itself the name `snabbnfv-traffic-01:00.0` then that seems simple on the surface but how will the name change if we have one process handling multiple ports or one port shared by multiple processes? Then we are potentially making life more complicated than if the startup scripts would manually assign a name that suits them (e.g. `nfv-1`).

So I start to come around to your way of thinking.

I wonder if we could have a command like `snabb gc` that would clean up old directories in a good way and could be easily called from scripts or cronjobs that want to adopt a default behavior?

lukego, 2015-05-25:
I suppose also that over time `snabb top` could  be made quite flexible about how it chooses which instance to monitor. For example instead of always having to be told the pid or symlink name perhaps it could do a simple search of the available directories. Then naming becomes less of a big deal if a user can say `snabb top nfv 01:00.0` and it finds the right instance heuristically. That may not be the best example but at least I see now that we are not so heavily dependent on a symbolic name or lack thereof.

lukego, 2015-05-26:
Closing this issue. I think you have all the context to decide what you think is the best way to do this for `snabb top`.

closed 2015-05-26 by lukego