snabbco/snabb #484: build with installed libraries

comotion, 2015-05-15:
If I have a system which already provides ljsyscall and pflua, the snabbswitch build system will still insist on fetching and building these. How could I go about building snabbswitch with my pre-installed libraries?
On a related note, should snabbswitch always be built static or can I link the dependencies dynamically?

lukego, 2015-05-15:
Currently we always build statically.

We have considered moving away from this model (see e.g. [here](https://groups.google.com/d/msg/snabb-devel/2kNiyRvIl5g/oxhbGpfU5lcJ)) towards dynamically linking or even breaking Snabb Switch into separate [LuaRocks](https://luarocks.org/) components, but to me it seems like the static linking model is more practical for our current needs.

One major challenge with using system-supplied versions of LuaJIT and pflua is how to get a compatible version? There is no published stable version of either that we can nominate and in practice it has been important to be able to pull specific fixes at least from LuaJIT. I also like the idea that we could extend these projects and immediately make use of those extensions before they land upstream.

So submodules seem very practical for the immediate future, at least until LuaJIT starts making releases from its v2.1 development branch that we could depend on.

lukego, 2015-05-16:
The [DMA address tag check failed](https://groups.google.com/d/msg/snabb-devel/jY_W6Z7_CFI/0PTsmcUH3rgJ) issue highlights the way we depend on LuaJIT.

We added some code that depended on bringing in a specific bug fix in LuaJIT. This broke Snabb Switch for anybody using an earlier version of LuaJIT, including everybody who had not updated and rebuilt the luajit submodule in our repository. The solution we adopted was both to update our LuaJIT submodule to a known-good version and also to update `src/Makefile` to check that this is actually built.

This has worked okay so far as I know. People have since stopped reporting problems caused by an out-of-date LuaJIT. It does mean that we have a hard dependency on the specific submodule versions that we link to though.

One piece of the puzzle that is missing is a plan for when to pull in updates to our submodules (luajit, ljsyscall, pflua). Currently we do this when we need something new but not as a matter of routine.

comotion, 2015-05-18:
I see, I can relate to both the need to build statically and the need to depend on a particular version of the dependencies. What I am after is a way for the build system to allow me to use system-provided libraries not as a default, but as an option in the "I know what I am doing" situation, and certainly without disabling any dependency checks.

Context here is that I am trying to package snabbswitch for a small embedded linux distro, where space is at a premium.

Perhaps the build system can support some environment variable to tell it to use a particular library file instead?

lukego, 2015-05-25:
Can you easily include patches in your packaging? If so I would suggest initially working on a branch that updates the Makefile to suit your needs. Once you are happy with it we can look for the best way to make the master branch support this use case and reduce/avoid the maintenance burden of maintaining a branch.

Space optimization is interesting. I would like to start working on shrinking the `snabb` executable back to its target size of 1MB from the ~1.5MB that it is today. I am not sure that dynamically linking with LuaJIT will turn out to be a good trade-off though due to the "I know what I am doing" assumption that the user will always have a suitable version of LuaJIT installed at all times. There may be simpler ways to shave a few hundred KB off the object size.

lukego, 2015-05-25:
I suppose there is also the risk of Snabb Switch packaged for a particular distro diverging too much from the mainline. The best solution I can think of for this would be to publish the branch in `src/doc/branches.md` so that the current state of the package is transparent and we can see what changes are currently needed compared with the master repo.

lukego, 2015-05-28:
I don't think it makes sense for the master branch to offer to use the system-installed LuaJIT at this time. If LuaJIT were making releases from the v2.1 branch and we could refer to a well-known minimum version then this might make sense. However, LuaJIT v2.1 is not making releases and I would not be confident to run Snabb Switch with a version older than what is on master at any given time.

If anybody really wants to do this they can of course modify the Makefile to stop statically linking LuaJIT. That seems to me like roughly the right level of difficulty to discourage people from doing this lightly and potentially creating problems for themselves.

Closing this issue for now.

closed 2015-05-28 by lukego