snabbco/snabb #505: Build issues

majek, 2015-05-27:
1) basename: invalid option -- 'a'

```
C(SSE2)   obj/arch/sse2_c.o
C(AVX2)   obj/arch/avx2_c.o
ASM       obj/apps/example/asm_dasc.o
basename: invalid option -- 'a'
Try `basename --help' for more information.
make[1]: *** [program/programs.inc] Error 123
```

2) For some reason cloning http://github.com/SnabbCo/luajit.git fails. I had to replace that with http2:// in .gitmodules, deps/pflua/.gitmodules,

lukego, 2015-05-27:
Can you give some more information please?

1. What platform are you on? (OS/distro/version)
2. Can you paste the output when it fails?

lukego, 2015-05-28:
Please reopen if you want to follow up on this with more information about your environment.

Currently we are depending on a GNU toolchain (where `basename -a` exists) and this is the first I have heard about `http://` not being a suitable method for accessing Github.

majek, 2015-05-28:
We are on debian squeeze:

```
$ apt-cache policy coreutils
coreutils:
  Installed: 8.5-1
  Version table:
 *** 8.5-1 0
        500 http://ftp.us.debian.org/debian/ squeeze/main amd64 Packages
```

lukego, 2015-05-28:
Thank you for reporting this. It seems from browsing [Debian manpages](http://manpages.debian.org/cgi-bin/man.cgi) that `basename -a` is a new feature that arrived in Debian Jessie. I had assumed it had been around forever.

I will look into a fix.

I suppose this is a good motivation to support efforts like #483 that are generally reducing our dependence on GNUisms.

lukego, 2015-05-28:
Does #510 fix this for you?

The easy way to check should be to checkout the `next` branch where I have merged this along with other features for the next release:

```
$ git checkout next
$ make
```

commit eaa408d8 (2015-05-28): Makefile: Avoid nonportable 'basename -a'

'basename -a' seems be a relatively new feature of GNU coreutils that
is not available in recent Debian releases.

Resolves #505.
PR #510 (2015-06-01): Makefile: Avoid nonportable 'basename -a'
`basename -a` seems be a relatively new feature of GNU coreutils that is not available in recent Debian releases.

Resolves #505.
closed 2015-06-01 by eugeneia