snabbco/snabb #509: Running cleanup functions on exit signals

eugeneia, 2015-05-27:
We would like to be able to run cleanup code before exiting due to "deadly signals". I have implemented one possible solution based on ljsyscall (signalfd) here:

https://github.com/SnabbCo/snabbswitch/compare/master...eugeneia:graceful-exit

It allows to add cleanup functions to be run (`on_exit`), the trapping of exit signals (`trap_exit`) and a function to poll for exit signals on the signalfd and eventually exit cleanly (`poll_exit`).

It is hooked into Snabb Switch by running `poll_exit` preiodically during `engine.main`. The problem is that Snabb Switch programs don't spend all their time in `engine.main`, some don't spend any time there. This would mean that programs spending significant time outside of `engine.main` would have to periodically call `signal_exit.poll_exit` to stay responsive to the trapped signals.

Another problem is that `signalfd(2)` can't queue `SIGKILL`.

My solution seems inadequate and incomplete. I would like to kick off a discussion about implementing this feature (or simply abandoning it) here.

lukego, 2015-05-28:
Could we use [`atexit`](http://man7.org/linux/man-pages/man3/atexit.3.html)?

ljsyscall has a Lua interface to to this I believe, @n-nikolaev showed me how to use it once to call a Lua function on shutdown. However, I don't see it available right at this moment (see justincormack/ljsyscall#180).

lukego, 2015-05-28:
Looks like I misunderstood `atexit()`: it is only for orderly shutdowns.

Thought: it is not possible to handle all kinds of errors in the process itself. For example, heap corruption, `kill -9`, out-of-memory killer can all break any in-process cleanup logic. So there needs to be a separate mechanism to cover those cases. If we are going to have a separate cleanup mechanism anyway, why do anything on process termination at all? That would seem to only be an optimization to make cleanup happen faster.

Joe Armstrong argues persuasively that error handling should always be done outside of the process that is crashing. If interested see "Error Handling Philosophy: Let some other process fix the error" in his thesis [Making reliable distributed systems in the presence of software errors](http://www.erlang.org/download/armstrong_thesis_2003.pdf).

lukego, 2015-05-28:
Other thought: what are the use cases for this mechanism? (are there things we need to do for which signal-driven cleanup is the best solution?)

eugeneia, 2015-05-28:
@lukego There was some PCI related cleanup that fell into the same category.

lukego, 2015-05-28:
Crazy idea that may or may not have a precedent...

Could we fork() a process that will detect when the parent process dies and perform the cleanup? That would seem to be immune to most kind of errors that I can think of.

eugeneia, 2015-05-28:
That would work for the runtime files case but not so much for the "disable DMA" case (unless we persist which devices are open). I generally like this idea because we can add it later on and just stop thinking about it right now. (I can deal with the stale runtime files issue right now by running `snabb gc` in `bench_env` and/or manually).

lukego, 2015-05-28:
Sounds good. I reckon we will be able to handle the DMA case too. (Though on that specific case maybe we *will* want to do it in the main process so that it happens synchronously before the kernel has the chance to reuse memory that will be DMA'd.)

eugeneia, 2016-06-03:
Note to self: never delete branches......

eugeneia, 2016-09-21:
Closing because #930 landed.

closed 2016-09-21 by eugeneia