snabbco/snabb #518: Call for help for 82599 ixgbe driver traffic mirror implementation

pavel-odintsov, 2015-06-14:
Hello, folks!

I found you have implemented traffic mirror capability for 82599. I need complete traffic mirror from one virtual port another for my toolkit FastNetMon.

Unfortunately, standard driver ixgbe haven't this option but it's very important for me.

Could you help me with docs and code examples for this option? Thanks!

lukego, 2015-06-15:
Hi Pavel,

Can you explain a bit more context about how your software works today and how you want to change it?

You are using ixgbe? with PF_RING? and VMDq or SR-IOV? The problem you want to solve is lack of an ioctl/ethtool trick to enable mirroring between queues in this situation? Or you want to do a Snabb-based prototype of FastNetMon?

pavel-odintsov, 2015-06-15:
Hello, Luke!

I want to implement scheme when netmap or PF_RING ZC enabled analyzer could be run on same interface with already working Linux with network services.

[image]

pavel-odintsov, 2015-06-15:
Now my tool working really anywhere. It could work inside KVM (PCI-E passthrough), on top of physical NIC, on virtual function inside KVM VM or on physical hardware. 

But I want capture on wire speed without any changes from network infrastructure side. 

Intel decided do not add this function to the driver: https://sourceforge.net/p/e1000/bugs/480/ Very sad. You are my last resort.

Yes, I working with ixgbe, it's most popular, flexible, reliable and awesome card :) Sure, I will try with SnabbSwitch idea, it's on my roadmap too. But before this I need finish some tasks..

lukego, 2015-06-16:
So if I understand correctly you are looking for a way to capture all packets sent/received on the NIC without interfering with the normal traffic processing by e.g. the Linux kernel. The mechanism you have in mind is to provision new hardware RX queues in "mirror" mode that your application can attach to. This is supported in hardware, but the problem is that the standard Linux driver has no way to enable it. So you are looking for the simplest way to enable this.

Is that correct?

I am interested in this topic: How to make multiple applications play nice when they all want to access the NIC hardware directly. In principle SR-IOV could be a nice way to take care of this but in practice Intel's implementation is pretty limited. So it seems like a mechanism operating directly on TX/RX queues makes sense.

The full solution would have a few components:

1. Driver to initialize the NIC. For example, ixgbe, or a derivative like the one from PF_RING/DNA, netmap, or Intel's [bifurcated driver](http://events.linuxfoundation.org/sites/events/files/slides/LinuxConEurope_DPDK-2014.pdf).
2. Interface to setup queues the way we want them e.g. provision new mirror ports. For example, ethtool talking with a kernel driver.
3. Applications that independently attach to individual TX/RX queues. For example, PF_RING, netmap, Snabb Switch, etc.

This would be a neat ecosystem. Then Snabb Switch could have "apps" to attach to an RX or TX queue and they would not really care how the queue was provisioned: by our own NIC driver or by ethtool/ixgbe on the host. It should work for your application too.

So what is missing? If it is only an ethtool command to provision a mirror queue then the best solution might be to implement that and submit it to ethtool/ixgbe. Even if Intel do not have a roadmap to implement this themselves it could still be the best solution.

If you really need something in a hurry there should be multiple possibilities for a quick hack, including using the Snabb Switch driver to add new mirror queues after ixgbe has setup the NIC. This kind of hack usually works to a first approximation but is fragile and I would not especially recommend it.

I would be curious to hear the perspective of Intel engineers who are working with the bifurcated driver and related areas. Let us see if we can attract their attention and get their feedback.

lukego, 2015-06-16:
... actually perhaps the direct hack would not be so bad with SR-IOV VFs when you already have a queue dedicated to your application and you only need to enable mirror mode on it. Let me ponder..

lukego, 2015-06-16:
Does your application already support operating on a VF instead of a full NIC? (I am not up-to-date on the latest internals of netmap and PF_RING to know whether they support this.)

pavel-odintsov, 2015-06-16:
I'm really happy about your interest in this approach!

I need check it, I have multiple machines with PF_RING and Netmap with 82599 NIC's. And will check it shortly.

pavel-odintsov, 2015-06-16:
Negative for PF_RING, detailed ticket there: https://github.com/ntop/PF_RING/issues/15

pavel-odintsov, 2015-06-16:
And awesome news from netmap, it working perfectly: http://www.stableit.ru/2015/06/how-to-run-netmap-on-ixgbe-virtual.html

I could run two netmap instances on same physical NIC, one per each virtual NIC.

And ticket to Luigi: https://github.com/luigirizzo/netmap/issues/63

lukego, 2015-06-16:
If you are using netmap then you already have a modified ixgbe driver. Could be straightforward to add mirroring support directly to the netmap ixgbe? Then you would have the whole solution in one place.

pavel-odintsov, 2015-06-16:
Yes, I have custom driver for ixgbe: https://github.com/pavel-odintsov/ixgbe-linux-netmap

But I have no knowledge what I should fix.

lukego, 2015-06-16:
I would suggest asking the netmap and/or ixgbe community about that. If that turns out to be a dead-end for some reason then we can look into how to help you with Snabb Switch. However, it seems like overkill to use Snabb Switch just to poke a register on a device controlled by a different driver.

Are you familiar with the [82599 data sheet](http://www.intel.com/content/dam/www/public/us/en/documents/datasheets/82599-10-gbe-controller-datasheet.pdf)? That is a really nice document, once you get to know it, and there you can find details of how to enable mirroring in the driver.

pavel-odintsov, 2015-06-16:
Luke, do you have support for mirrored VF's in Snabb Switch? Maybe I could port it from LUA to C?

lukego, 2015-06-16:
Here is our code to enable mirroring: https://github.com/SnabbCo/snabbswitch/blob/master/src/apps/intel/intel10g.lua#L987.

I don't think this particular feature has test coverage so beware in case there is a bug. The code probably only makes sense with reference to the data sheet.

pavel-odintsov, 2015-06-16:
Awesome! I will try to dig into ixgbe docs. I have sent letter to ixgbe community and send copy to you.

Have you finished any performance tests? Could two applications on different VFs work in wire speed? Could we achieve wire speed for 10GE for mirror interface in both directions?

lukego, 2015-06-16:
I have not tested performance of this feature. We implemented it in the driver but do not currently use it in any applications.

I would imagine the PCIe bus will the the bottleneck. Packets will have to be DMA'd twice. The PCIe bus has around 16 Gbps of bandwidth but there is additional overhead to be counted in that. I would guesstimate you should see ballpark 2/3 of line rate.

pavel-odintsov, 2015-06-16:
2/3 of wire speed songs awesome too. Really nice approach if we could implement it. Thank you for your attention and your huge help! :)

lukego, 2015-06-16:
Good luck and let us know how you go :-)

pavel-odintsov, 2015-06-16:
Sure, thanks!

lukego, 2015-06-19:
Closing issue since there is no action pending on Snabb Switch.

closed 2015-06-19 by lukego

morphyno, 2015-11-05:
I have a question in regard to Snabb Switch's code above to enable port mirror on 82599. How is Snabb interfacing with the driver via lua? Did you have to compile the existing driver any differently?

pavel-odintsov, 2015-11-06:
Hello!

They have wrote driver from the scratch with LUA :)

ghostli123, 2016-06-24:
Hello! Have you figure out the issue of how to enable and control traffic mirroring in 82599 ixgbe driver? Thanks!