snabbco/snabb #522: Design idea: Split I/O apps into separate init, transmit, receive

lukego, 2015-06-18:
Crazy idea... I am not sure yet whether it is good, bad, impossible, or necessary.

First the idea and then the rationale.

**Idea**

Let us split each I/O app into three separate parts for initialization, transmit to one queue, and receive from one queue. For example, `intel10g` becomes `intel10g.nic`, `intel10g.tx`, `intel10g.rx`.

Here are some example app networks:

two nics, bump in the wire:

    nic0 (app that initializes a nic)
    nic1
    nic0RX0 --> myapp0 --> nic1TX0  (rx app connected to tx app)
    nic1RX0 --> myapp1 --> nic0TX0  (other direction)

one nic, traffic dispatched in hardware (RSS, VMDq, SR-IOV):

    nic {rss/vmdq/sriov=true} (init nic and setup dispatching)
    nicRX0 --> myapp0 --> nicTX0
    nicRX1 --> myapp1 --> nicTX1  (processing pipelines..)
    nicRXN --> myappN --> nicTXN

one nic, traffic dispatched in software, separate TX queues:

       nic
                              .--> myapp0 -> nicTX0
    nicRX0 --> mydispatcher --+--> myapp1 -> nicTX1
                              `--> myappN -> nicTXN

**Why?**

1. Support distributing I/O throughout a parallelized app network that is executing in many different processes. For example, the "bump in the wire" example could have a separate process for each direction of traffic.
2. Support 100G NICs. For example, in the traffic dispatched in software example we could use one dedicated core to dispatch the 100G traffic stream on RX. On TX we could use multiple queues to let the hardware recombine the traffic.
3. Play nice with other drivers. How about if the kernel driver could be setup with `ethtool` to send traffic to a new RX/TX queue that we would attach to with an app? (This seems to be a use case that Intel want to support with their "bufurcated" kernel driver but I am not sure of the details.)

Generally it seems like if we don't do this then it will prevent us from creating well parallelized app networks. Transmit and receive on a given port will often be independent operations operating on different packets and ideal for running on separate cores.

**Challenges**

If we adopt the notion that transmit and receive should be handled by different apps then it will require some hacking:

1. Drivers will need to be split up: Intel, Solarflare, Virtio, Tap, Raw, etc. I have a very rough draft of an Intel (I350) driver based on this model that seems basically promising. I am not sure whether the others will be hard/impossible?
2. This would make app networks more complex. The app network designer would need to create all the apps that they need (zero or more NIC, RX, and TX apps) and would need to manually ensure that the configuration is consistent i.e. that the NIC is initialized to dispatch traffic to the right queues.

**Summary**

That's the idea. Feedback welcome. Is this good, bad, necessary, or impossible?

lukego, 2015-06-18:
NB: The I350 driver code linked above is very quirky. I am just playing around with ideas there for now.

javierguerragiraldez, 2015-06-22:
I've been checking what would be needed to split the Intel10g driver like that.  Currently, each 'vf' device object holds a reference to the 'pf' device object, but there's almost no function call through this reference, it's mostly to access some 'common area' registers to do some setup, like MAC address, enabling/disabling queues, etc.

I think the simplest would be that any setup that needs access to those common registers would be done directly to the 'pf' app, and the 'vf' apps would just accept what is done.  Something like this:

    nic0 args={vmdq='32*4', pools={
        [1]={mac='52:54:00:01:01:01', vlan=18, rate_limit=1e9},
        [2]={mac='52:54:00:02:02:02', vlan=18, rss='ipv6'},
        [3]={mac='52:54:00:10:10:10', vlan=1, mirror_pools={1,2}},
    }

    nic0rx1 args={rxpool=1},
    nic0tx1 args={txpool=1},
    nic0rx2a args={rxpool=2, queue=0},
    nic0rx2b args={rxpool=2, queue=1},
    nic0rx2c args={rxpool=2, queue=2},
    nic0rx2d args={rxpool=2, queue=3},
    ......

There, the 'vf' apps would only deal with moving packets to/from a single queue ring and don't care about initialization.  seems really doable, and quite simple to parallelize.  it even seems like we could finally get some use for RSS: just set a separate but equal 'app network' for each queue in a single queue pool.  in the example i'm using 4 queues per pool, instead of the more common 64*2 scheme.  the four "pool 2" apps would be connected to different instances of, say, a tunnelling/compression app, so the CPU/heavy work would be distributed among 4 cores.

plajjan, 2015-06-22:
RSS yay \o/

lukego, 2015-07-04:
Playing around with this idea a bit. Here is a diagram of a hypothetical design for an Ethernet switch showing two different ways that each port could be implemented: with separate TX and RX apps feeding into parallel paths of the app network ("Separate RX and TX") vs with a series of apps each handling bidirectional traffic ("Unified RX+TX"). I imagined that each port would use two processes (cores/hyperthreads) and I used a distinct color for apps that would run together in the same process.

[image]

I suppose it would be hard to argue that the new-style "Separate RX and TX" design is fundamentally better. It has more apps and that probably makes it both more parallelisable and also more complex. The extra parallel opportunities don't seem that exciting because the more traditional design seems to have plenty too. So at first glance it seems like more complexity with no really certain benefit.

So for the moment this leads me to think we may be better off sticking much closer to the existing intel_app that we have. It seems like we could pretty straightforwardly extend that to cover all scenarios we are talking about:

- Option to attach multiple apps to the same NIC from different processes.
- Option to send RSS.
- Option for each app to use 0..N RX queues and 0..M TX queues.

The next radical possibility, bouncing even further back in the other direction, is that we could make links in the app network bidirectional. That would make them more like physical ethernet links and so constructing app networks would be more closely related to constructing real networks. Just a thought.

This was also an interesting exercise in using Graphviz to sketch app networks. It was kind of a pain but maybe that gets better with practice. I have some arrows pointing the wrong way because it was a pain to try to fix. Here is the source for `dot` file - tips welcome:

```
digraph labswitch {
  graph [style=dotted ranksep=0.3]
  node [style=filled]
  subgraph cluster_port1 {
    label="Separate RX and TX"
    node [color=lightblue]  RX1 -> RXFilter1 -> RXRatelimit1 -> Switch1
    node [color=tomato]     TX1 -> TXFilter1 -> TXRatelimit1 -> Switch1
  }
  subgraph cluster_port2 {
    label="Unified RX+TX"
    node [color=violet] NIC2 Ratelimit2
    node [color=green2] Switch2 Filter2
    Switch2 -> Filter2 -> Ratelimit2 -> NIC2
    edge [constraint=false]
    NIC2 -> Ratelimit2 -> Filter2 -> Switch2
  }
  rankdir=LR;
  Switch1 -> Switch2
  Switch2 -> Switch1
}
```

lukego, 2015-07-04:
Gonna close this for now. I will experiment with the labswitch program.

closed 2015-07-04 by lukego