snabbco/snabb #529: Understand how numactl/taskset affects Snabb Switch

eugeneia, 2015-06-26:
We still have this yet unexplored issue of Snabb NFV failing if the Snabb/VM processes are not bound to the correct NUMA nodes. In the past we fixed this issue popping up in SnabbBot by doing:

```
taskset -c 6 ~/snabbswitch-max/src/scripts/snabb_bot/snabb_bot.sh
```

Note that on davos CPU 6 in on node 1, the VM and Snabb processes are bound to that node by `numactl`. I do not think I ever understood why that worked (or how it affected SnabbBot), but the effect was reproducible, without it we would get a "PING failed". Also note that this "fix" was highly dependent on the process tree, it wouldn't work if we `taskset` the underlying NFV `selftest.sh` process, nor does the problem occur deterministically when running `selftest.sh` by hand.

Now with the new `test_env`, the behavior is the other way around. We get ping failures with the `taskset` and it passes without. Long story short: I have no idea whats going on, and that really bothers me. There is also this new non-deterministic  "ND failed" which I believe to be caused by the same issue.

In case anybody wonders: The failing PRs that suddenly pass is me switching on and off the `taskset` used in the SnabbBot invocation depending on whether the PR uses `bench_env` or `test_env`. Yeah......

Help?

lukego, 2015-06-26:
Can you tell me exactly how to reproduce this at the shell on `davos`? I'll take a look.

eugeneia, 2015-06-26:
@lukego Try this (on a branch with `test_env`):

```
cd snabbswitch
flock -x -n /home/max/snabb_bot/lock taskset -c 6 /home/max/snabb_bot/tasks/test.sh
flock -x -n /home/max/snabb_bot/lock /home/max/snabb_bot/tasks/test.sh
```

lukego, 2015-06-29:
I am rebooting davos with the IOMMU disabled in order to eliminate that as a potential cause. The default behavior of Linux when the IOMMU is disabled has been inconsistent at best.

sleinen, 2015-06-29:
On Mon, Jun 29, 2015 at 8:31 AM, Luke Gorrie <notifications@github.com>
wrote:

> The default behavior of Linux when the IOMMU is disabled has been
> inconsistent at best.

Do you mean, "...when the IOMMU is *enabled* has been inconsistent at best"?
-- 
Simon.

lukego, 2015-06-29:
Er, right :).

@eugeneia  I have disabled the IOMMU and the problem is not immediately reproducible for me. I am not sure if it is actually fixed though. Can you reproduce it now?

eugeneia, 2015-06-30:
That seems to have  fixed it. The ND failure was actually caused by a race between the check and Snabb applying the config and kicking off the ND resolution. See https://github.com/SnabbCo/snabbswitch/pull/531

closed 2015-06-30 by eugeneia

lukego, 2015-07-02:
Is this fixed by #531 or by the IOMMU change? (If the IOMMU is believed to be innocent then I would like to re-enable it so that we don't falsely blame it for voodoo problems.)

NB: For some reason it seems like the default kernel in `grub` for davos is not bootable. I brought it up manually last time. So at the moment if you reboot it and it doesn't come up then that is probably why :-)

eugeneia, 2015-07-02:
It was the IOMMU change, #531 fixes an unrelated problem with the test suite that I falsely believed to be related.

eugeneia, 2015-07-02:
It would be nice to gain insight in how IOMMU affects our code. From what I could gather its a feature geared towards virtual machines efficiently using hardware, and is indeed related to DMA. Could it be a safety feature of IOMMU that prevents successful DMA from the `snabb` process or the NIC (depending on which NUMA node its running on)?

https://en.wikipedia.org/wiki/IOMMU

http://www.mjmwired.net/kernel/Documentation/Intel-IOMMU.txt

Above is the only Linux documentation I can find about IOMMU, it has one interesting bit:

```
Fault reporting

DMAR:[DMA Write] Request device [00:02.0] fault addr 6df084000
DMAR:[fault reason 05] PTE Write access is not set
DMAR:[DMA Write] Request device [00:02.0] fault addr 6df084000
DMAR:[fault reason 05] PTE Write access is not set
```

So I assume we should look out for error messages like these in syslog (I checked davos and the only DMAR it has is this one: `Jun 29 09:13:29 davos kernel: [    0.000000] ACPI: DMAR 000000007e00b300 000190 (v01 A M I   OEMDMAR 00000001 INTL 00000001)`

lukego, 2015-07-02:
On Ubuntu 12.04 the IOMMU was disabled by default and when you enabled it then DMA would be blocked with messages like the one you show above. Quite predictable.

On Ubuntu 14.04 the IOMMU was now enabled by default and the behavior has been much less clear cut. Often everything works fine and occasionally we see strange behavior such as >50% loss of throughput or these test failures you saw.

I would like to identify the right kernel hacker / mailing list to follow this up with. Trouble is, I believe the kernel people would all suggest using the `vfio` mechanism to access PCI devices, but I don't feel that this kernel feature is mature enough for us yet. (That could also be something to discuss.)

There are actually three modes to the IOMMU: on, off, and passthrough. I wonder if passthrough is really okay? That would be convenient because then ideally it should not affect Snabb Switch but QEMU could still use the IOMMU to make whole PCI devices available to virtual machines (an important use case).

lukego, 2015-07-02:
(See also #455.)