snabbco/snabb #532: Include packetblaster/DPDK based NFV benchmark in NFV selftest

eugeneia, 2015-06-30:
In my [test-env-dpdk](https://github.com/eugeneia/snabbswitch/tree/test_env-dpdk) branch I have added the building blocks for the packetblaster/DPDK setup. You can test it like so:

```
cd snabbswitch/src
sudo bash -rcfile program/snabbnfv/test_env/test_env.sh
packetblaster 0000:84:00.1 64
qemu_dpdk 0000:84:00.0 vhost_A.sock 1234
snabb 0000:84:00.0 "snabbnfv traffic 0000:84:00.0 test.port vhost_%s.sock"
```

Everything *seems* to work but I can't see any packets reaching snabb. The snabb config looks like this:

```
return {
  { vlan = nil,
    mac_address = "00:00:00:00:00:00",
    port_id = "A",
    ingress_filter = nil,
    gbps = nil,
    tunnel = nil
  },
}
```

I had to add these lines to the VMs `/etc/rc.local` to get DPDK to run:

```
# Setup hugepages (required by DPDK)
mount -t hugetlbfs nodev /hugetlbfs
echo 64 > /sys/devices/system/node/node0/hugepages/hugepages-2048kB/nr_hugepages
```

Any hints? Are the ports `0000:84:00.0` and `0000:84:00.1` on davos connected?

lukego, 2015-07-03:
Howdy! Sorry I didn't dig into this yet. Here are some quick ideas:

Yes, davos is cabled. (Every 10G port in the lab is cabled to itself except for some quad-port Broadcom NICs that we haven't done anything with.)

Is the destination MAC address in the traffic you are sending 00:00:00:00:00:00 like in the config file? That is required to make the NIC send them to that VM's RX queue. If you have 00:00:00:00:00:00 everywhere already then *maybe* there is something special/reserved about that address. I'd recommend replaying a trace that sends packets to FF:FF:FF:FF:FF:FF which is broadcast to all queues. (I believe Nikolay has been testing with packets containing all 0xFF -- should be able to find those files on grindelwald even.)

Try running `snabbnfv traffic -D 10` to get a debug report every 10 seconds? The statistics registers in the NIC often provide a hint to what is happening.

eugeneia, 2015-07-06:
Aha!! It works! Was indeed a mac address thing. Ideally I would want a NFV config that forwards ALL packets to a port. E.g. wildcard MAC.

eugeneia, 2016-01-06:
Closing because #547 landed.

closed 2016-01-06 by eugeneia