snabbco/snabb #555: Idea: Layer-2 switch scalable design

lukego, 2015-07-10:
This is an idea for how we could parallelize the [labswitch](https://groups.google.com/d/msg/snabb-devel/gdMF5Vmd-YI/J3mkO2apkbIJ) application to give it suitable performance characteristics.

Goals:

1. Support many ports e.g. 32 x 10G.
2. Support line rate with small packets on all ports in simple configurations.
3. Support additional features (tunnels, filters, etc) with overhead only on the ports that use them.

The idea is as follows:

* One snabb process per 10G of traffic (one 10G port or multiple 1G ports).
* Local Layer-2 switch app for each port.
* Layer-2 switch apps all connected together in a fully meshed topology.

Here is the idea in picture form. Each color (red, brown, blue) is a distinct process.

[image]

Design notes:

* Loops between Layer-2 switches will have to be managed to avoid [broadcast storms](https://en.wikipedia.org/wiki/Switching_loop). Suggestion is that the Layer-2 switch will have a rule that it does not send multicast packets between switch interconnect ports (curly lines). That is, multicast packets are "flooded" by the source Layer-2 switch to all of the other Layer-2 switches, which then transmit only to their own locally attached NICs.
* This should scale. In a 32 x 10G configuration each CPU core could be processing only 10 Gbps of traffic i.e. the packets received and transmitted to the local NICs.
* The design depends on processing functions to be able to run per-port including switching. If we wanted to add routing support (for example) it would also have to fit this model.

lukego, 2015-07-10:
@eugeneia @javierguerragiraldez what do you think?

javierguerragiraldez, 2015-07-10:
Sounds like a very concrete case of eating our own dogfood.  In theory such a setup would be able to support _any_ network function, so it becomes an incentive to write then and make them actually work.

The inter-process mesh seems heavy at first, but a full mesh of 32 nodes needs 992 unidirectional links.  assuming we use the current two-256-packet rings, it's just 2*256*sizeof(ptr)*992, less than 4MB in rings.  Each process just manages 31 inputs, 31 outputs and a whatever local configuration (i guess typically just one input and one output more).

eugeneia, 2015-07-10:
Oh we are using scanners now? I love scanners!

This maps fairly well to my existing `labswitch` branch. Currently the bottleneck in my design is that all L2 ports share the same `bridge` app, which obviously wont scale. Now I have this possibly stupid idea for a design, where every L2 port has its own `bridge` app, which is connected to every other port's `bridge` app in an N:N mesh.

[image]

This is analogous to a physical "everyone-to-everyone" wiring, except we don't get to maintain a cable matrix by hand. The mesh ports obviously need to be in a split horizon group in order to avoid cycling packets around. See [apps/bridge/README.md](https://github.com/SnabbCo/snabbswitch/blob/1da904f7eced9384adbe4e46670892c3cfd5fad3/src/apps/bridge/README.md).

lukego, 2015-07-12:
@eugeneia yeah that makes perfect sense to me. @alexandergall was way ahead once again with already supporting the split horizon groups in the bridge app :) I hadn't made that connection yet.

eugeneia, 2015-07-13:
I implemented the sketched design in my `lab-switch` branch. It works as imagined but pays a ~30% performance toll, which is to be expected since each packet travels through 30% more links I guess?

lukego, 2015-07-13:
There is probably a bunch of optimization to do. I would suggest defining a standardized benchmark that we can use for reference. Then we can define a meaningful performance target and gradually work on making the application reach it.

One idea would be to blast 64-byte packets through the switch and aim for ~30 million packet per second per process capacity (~15M from NIC + ~15M from other switch apps). That is the performance we would expect from a hardware switch.

The lab switch will also need to be very accountable for any packets that are dropped, even if the are dropped by the NIC rather than the snabb process. Otherwise it would be very confusing to involve in performance tests of other applications. This is basically the same requirement as Alex's SNMP monitoring support.

Getting to the right performance may be a long process. To start with I think the important thing is to have an easy way to measure performance and evaluate optimizations. Perhaps it would even be meaningful to have multiple related benchmarks e.g. separately exercising the NIC and the L2 switch app.

Is that useful feedback?

lukego, 2015-07-13:
Note: we have often started from 10x less performance than we want but progress has been fast once we set a target and have a common benchmark. So don't be discouraged in any way by whatever the starting point is.

alexandergall, 2015-07-14:
The learning part of the current bridge implementation does incur a substantial overhead (I included some estimates in the comments of the PR). It's a function of the parameters of the Bloom filter as well as the number of output ports assigned to each input port. Storing of the source and destination MAC addresses of a packet in a Bloom filter storage cell can take up to a few hundred cycles (but it's O(1) with respect to the number of objects in the filter and the lookup operation should be pretty fast), i.e. that alone will limit the packet rate substantially.

I did not yet perform any serious performance tests with that thing, apart from my VPLS app, which has a fairly modest performance target, so I'm very curious how it will do under heavy scrutiny :)

da4089, 2015-07-14:
But ... not STP between the bridge apps?

eugeneia, 2015-07-14:
@da4089 By STP you mean [Spanning Tree Protocol](https://en.wikipedia.org/wiki/Spanning_Tree_Protocol)? I do not think STP applies here. This is all virtual, we do not pay a cost for having more links, just for hops. Maybe I am not getting it, what do you think would applying STP give us?

da4089, 2015-07-14:
Yes.  STP (in one variant or another) is the traditional way of avoiding cycles between interconnected L2 segments. If the plan is to match each port with its own bridge, and to fully mesh them, STP might be suitable.

javierguerragiraldez, 2015-07-14:
Cycles are avoided by the split horizon tagging of internal links.  It's a much cheaper (and deterministic) algorithm.  Suitable here because topology is fixed.

javierguerragiraldez, 2015-07-14:
Regarding performance, I think it will be a challenge but doable.  My first (and so far, only) inter-process tests gave me 3Mpps, but I suspect that it was pushing too few packets per cycle.  I guess that the first hardware bottleneck is the constant cache-line locks and flushes that occur when modifying the ring indexes.  Other microbenchmarks have shown that there's an upper limit of 10~20 Mcycles/sec per shared resource.  If each cycle moves more than 10 packets, even a figure as low as 3 Mcycles/sec would give us the desired throughput.

lukego, 2015-07-14:
Curious, what topology would STP be likely to settle on for this network? (Would it chain all the L2 switch apps together in a chain? Or put one in the middle as a hub? Or arrange them in a tree?) My mental model is that STP will operate by selecting a set of links to deactivate i.e. by applying an eraser to the diagrams.

If we had a bridge app with STP support already then it would be interesting to test. I do tend to agree the static setup should be simple/fast/robust though.

eugeneia, 2015-07-14:
I added a snabbmark for snabblab in my `lab-switch` branch:

```
snabbmark labswitch <npackets> <packet-size> [<nstreams>]
    Test labswitch performance by sending <npackets> of <packet-size>
    over <nstreams> through the switch. The default for <nstreams> is 1.
```

The results are... improvable. Will talk about the results tomorrow.

alexandergall, 2015-07-15:
STP would put certain links in "blocking" mode (not forward onto nor accept packets from the link) to create a tree topology with a dynamically seleted bridge at the root (there are some knobs to influcence the decision who becoms the root bridge). With virtual links, it's much easier to set up a full mesh, which will also allow the packets to travel more directly than with STP, where there can be multiple hops in the path from one bride to another.

lukego, 2015-07-15:
> Oh we are using scanners now? I love scanners!

I am starting to love them too :-). I switched now from taking a photo of the diagram with my phone to a shiny new [USB scanner](http://www.amazon.com/Canon-Products-LiDE220-Document-Scanner/dp/B00LN0NUGC). (Diagram above updated.)

Good to know that the high DPI mode is there if I really need to do justice to my crayon artistry too:

[image]

Goodbye whiteboard and goodbye Sketchbook Pro w/ drawing tablet?

eugeneia, 2015-07-15:
```
./snabb snabbmark labswitch 2e8 64 1
Stream 1:
Sending from 01:00:00:00:00:01 to 01:00:00:00:00:02
Sending from 01:00:00:00:00:02 to 01:00:00:00:00:01
Switched 200.0 million packets in 57.30 seconds (rate: 3.5 Mpps).
link report:
         114,293,295 sent on bridge_stream_1_a.l2 -> sink_1_a.rx (loss rate: 0%)
          85,720,035 sent on bridge_stream_1_a.stream_1_b -> bridge_stream_1_b.stream_1_a (loss rate: 24%)
          85,720,035 sent on bridge_stream_1_b.l2 -> sink_1_b.rx (loss rate: 0%)
         114,293,295 sent on bridge_stream_1_b.stream_1_a -> bridge_stream_1_a.stream_1_b (loss rate: 0%)
         114,293,295 sent on source_1_a.tx -> bridge_stream_1_a.l2 (loss rate: 0%)
         114,293,295 sent on source_1_b.tx -> bridge_stream_1_b.l2 (loss rate: 0%)
```

[image]

```
./snabb snabbmark labswitch 2e8 64 2
Stream 1:
Sending from 01:00:00:00:00:01 to 01:00:00:00:00:02
Sending from 01:00:00:00:00:02 to 01:00:00:00:00:01
Stream 2:
Sending from 02:00:00:00:00:01 to 02:00:00:00:00:02
Sending from 02:00:00:00:00:02 to 02:00:00:00:00:01
Switched 200.0 million packets in 95.82 seconds (rate: 2.1 Mpps).
Stream 1: 76.93  million packets
Stream 2: 123.08  million packets
link report:
          46,157,040 sent on bridge_stream_1_a.l2 -> sink_1_a.rx (loss rate: 25%)
          30,771,360 sent on bridge_stream_1_a.stream_1_b -> bridge_stream_1_b.stream_1_a (loss rate: 0%)
          30,771,360 sent on bridge_stream_1_a.stream_2_a -> bridge_stream_2_a.stream_1_a (loss rate: 0%)
          15,386,190 sent on bridge_stream_1_a.stream_2_b -> bridge_stream_2_b.stream_1_a (loss rate: 49%)
          30,771,360 sent on bridge_stream_1_b.l2 -> sink_1_b.rx (loss rate: 50%)
          30,771,360 sent on bridge_stream_1_b.stream_1_a -> bridge_stream_1_a.stream_1_b (loss rate: 0%)
          30,771,360 sent on bridge_stream_1_b.stream_2_a -> bridge_stream_2_a.stream_1_b (loss rate: 0%)
          30,771,360 sent on bridge_stream_1_b.stream_2_b -> bridge_stream_2_b.stream_1_b (loss rate: 0%)
          61,542,465 sent on bridge_stream_2_a.l2 -> sink_2_a.rx (loss rate: 33%)
                 255 sent on bridge_stream_2_a.stream_1_a -> bridge_stream_1_a.stream_2_a (loss rate: 0%)
                 255 sent on bridge_stream_2_a.stream_1_b -> bridge_stream_1_b.stream_2_a (loss rate: 0%)
          30,771,360 sent on bridge_stream_2_a.stream_2_b -> bridge_stream_2_b.stream_2_a (loss rate: 0%)
          61,541,955 sent on bridge_stream_2_b.l2 -> sink_2_b.rx (loss rate: 20%)
          30,771,360 sent on bridge_stream_2_b.stream_1_a -> bridge_stream_1_a.stream_2_b (loss rate: 0%)
          30,771,360 sent on bridge_stream_2_b.stream_1_b -> bridge_stream_1_b.stream_2_b (loss rate: 0%)
          30,771,360 sent on bridge_stream_2_b.stream_2_a -> bridge_stream_2_a.stream_2_b (loss rate: 0%)
          30,771,360 sent on source_1_a.tx -> bridge_stream_1_a.l2 (loss rate: 0%)
          30,771,360 sent on source_1_b.tx -> bridge_stream_1_b.l2 (loss rate: 0%)
          30,771,360 sent on source_2_a.tx -> bridge_stream_2_a.l2 (loss rate: 0%)
          30,771,360 sent on source_2_b.tx -> bridge_stream_2_b.l2 (loss rate: 0%)
```

[image]

javierguerragiraldez, 2015-07-15:
I'm not sure your first diagram is correct...
this is how i'm reading the output:

stream A->B:

    114,293,295 sent on source_1_a.tx -> bridge_stream_1_a.l2 (loss rate: 0%)
    85,720,035 sent on bridge_stream_1_a.stream_1_b -> bridge_stream_1_b.stream_1_a (loss rate: 24%)
    85,720,035 sent on bridge_stream_1_b.l2 -> sink_1_b.rx (loss rate: 0%)

stream B->A:

    114,293,295 sent on source_1_b.tx -> bridge_stream_1_b.l2 (loss rate: 0%)
    114,293,295 sent on bridge_stream_1_b.stream_1_a -> bridge_stream_1_a.stream_1_b (loss rate: 0%)
    114,293,295 sent on bridge_stream_1_a.l2 -> sink_1_a.rx (loss rate: 0%)

eugeneia, 2015-07-15:
@javierguerragiraldez Yeah there was a swapped label in the first diagram, sorry. Should be fixed now.

eugeneia, 2015-07-15:
Some thoughts about the results: The first case with only one stream is pretty straight forward, except for 25% packet loss on a link. The results are deterministic, so there seems to be some kind of math behind when links will drop packets. We should figure that out.

The second case is disappointing for multiple reasons, first of all there is heavy packet loss on multiple links again. Then there are only 120M packets sent from the sources and 200M received on the sinks. This is because many packets cross the "stream border", e.g. go astray. This means that either I am using the learning bridge app the wrong way, or the learning bridge app does not work a I expected. This also leads to reduced overall performance. Ideally we would see very few packets crossing the dashed line.

@alexandergall thoughts?

javierguerragiraldez, 2015-07-15:
ah, ok.

i don't get the second one... i'm assuming the mac addresses are like:

1A: 01:00:00:00:00:01
1B: 01:00:00:00:00:02
2A: 02:00:00:00:00:01
2B: 02:00:00:00:00:02

if so, then this is how i read the output:

stream 1A->1B>

          30,771,360 sent on source_1_a.tx -> bridge_stream_1_a.l2 (loss rate: 0%)
          30,771,360 sent on bridge_stream_1_a.stream_1_b -> bridge_stream_1_b.stream_1_a (loss rate: 0%)
          30,771,360 sent on bridge_stream_1_b.l2 -> sink_1_b.rx (loss rate: 50%)

the 50% loss means that the 1B bridge was getting as many "stray" packets as expected ones.  most of them seem to come from the 2B bridge.

stream 1B->1A:

          30,771,360 sent on source_1_b.tx -> bridge_stream_1_b.l2 (loss rate: 0%)
          30,771,360 sent on bridge_stream_1_b.stream_1_a -> bridge_stream_1_a.stream_1_b (loss rate: 0%)
          46,157,040 sent on bridge_stream_1_a.l2 -> sink_1_a.rx (loss rate: 25%)

25% loss while delivering more packets than it supposedly receives... again, bridge 2B seem to be the source.

stream 2A->2B:

          30,771,360 sent on source_2_a.tx -> bridge_stream_2_a.l2 (loss rate: 0%)
          30,771,360 sent on bridge_stream_2_a.stream_2_b -> bridge_stream_2_b.stream_2_a (loss rate: 0%)
          61,541,955 sent on bridge_stream_2_b.l2 -> sink_2_b.rx (loss rate: 20%)

this one delivers a bigger fraction, the extra packets seem to be 30M from bridge 1B and 15M from 1A.  a funny point, those 15M from 1A to 2B register a 49% loss, so it was supposed to be 30M

stream 2B->2A:

          30,771,360 sent on source_2_b.tx -> bridge_stream_2_b.l2 (loss rate: 0%)
          30,771,360 sent on bridge_stream_2_b.stream_2_a -> bridge_stream_2_a.stream_2_b (loss rate: 0%)
          61,542,465 sent on bridge_stream_2_a.l2 -> sink_2_a.rx (loss rate: 33%)

now the extra packets are from bridge 1A and 1B, 30M each.

"stray packets"

          30,771,360 sent on bridge_stream_1_a.stream_2_a -> bridge_stream_2_a.stream_1_a (loss rate: 0%)
          15,386,190 sent on bridge_stream_1_a.stream_2_b -> bridge_stream_2_b.stream_1_a (loss rate: 49%)
          30,771,360 sent on bridge_stream_1_b.stream_2_a -> bridge_stream_2_a.stream_1_b (loss rate: 0%)
          30,771,360 sent on bridge_stream_1_b.stream_2_b -> bridge_stream_2_b.stream_1_b (loss rate: 0%)
                 255 sent on bridge_stream_2_a.stream_1_a -> bridge_stream_1_a.stream_2_a (loss rate: 0%)
                 255 sent on bridge_stream_2_a.stream_1_b -> bridge_stream_1_b.stream_2_a (loss rate: 0%)
          30,771,360 sent on bridge_stream_2_b.stream_1_a -> bridge_stream_1_a.stream_2_b (loss rate: 0%)
          30,771,360 sent on bridge_stream_2_b.stream_1_b -> bridge_stream_1_b.stream_2_b (loss rate: 0%)

it seems that the bridge 2A is the only one that send most of the packets it get from source 2A to the intended destination (bridge 2B), and only 255 packets to each of the other two (1A and 1B).  the other three bridges receive 30M packets locally and send (or try to send) all 30M to the other three bridges.  it does sound like either a misconfiguration, or maybe a small bug that results in losing most of the split horizon tags.

javierguerragiraldez, 2015-07-15:
sorry, small brain glitch: it seems the issue is in the MAC address learning or table applying, not split horizon

eugeneia, 2015-07-15:
> i'm assuming the mac addresses are like:
> 
> 1A: 01:00:00:00:00:01
> 1B: 01:00:00:00:00:02
> 2A: 02:00:00:00:00:01
> 2B: 02:00:00:00:00:02

That is correct.

javierguerragiraldez, 2015-07-15:
on a third sight, if you sum all the packets that each bridge gets from the other bridges, it corresponds to the number of packets it tries to send to its local sink.

for example, bridge 1A receives 30M from bridge 1B, 255 from 2A and 30M from 2B, that's 60M.  it delivers 46M after 25% loss, that is it tries to send 60M to sink 1A.

so, it seems that there _was_ 30M packets from source 2B intended for sink 1A.

eugeneia, 2015-07-15:
All packets that reach the bridge will be forwarded to its sink: Either a packet matches the source MAC of the port (from the packets sent by the source) and passes it to the sink, or it does not match the source MAC of the port in which case it gets flooded to all ports, e.g. the sink.

alexandergall, 2015-07-15:
Didn't have time to look at it in detail yet, but there is at least one gotcha: the MAC addresses starting with 01 happen to have the multicast bit set, so they are always flooded to all ports.  Try with a proper unicast MAC address.

eugeneia, 2015-07-15:
Wow I am an amateur :)

alexandergall, 2015-07-15:
Darn, but there is also a bug in the learning bridge! The section

       if not is_mcast then
         bf:store_value(mem, 6, eth_dst)
      end

should be

      if not is_mcast then
         bf:cell_clear(eth_dst)
         bf:store_value(mem, 6, eth_dst)
      end

Will submit a fix.

eugeneia, 2015-07-15:
With non multicast MAC addresses and a little fix on the benchmark side[1], but without Alex's bug fix (what does it do?), the result now looks like this, which is nicer indeed:

```
snabb snabbmark labswitch 2e8 64 2
Stream 1:
Sending from 00:00:00:00:01:01 to 00:00:00:00:01:02
Sending from 00:00:00:00:01:02 to 00:00:00:00:01:01
Stream 2:
Sending from 00:00:00:00:02:01 to 00:00:00:00:02:02
Sending from 00:00:00:00:02:02 to 00:00:00:00:02:01
Switched 200.0 million packets in 109.33 seconds (rate: 1.8 Mpps).
Stream 1: 100.00  million packets
Stream 2: 100.00  million packets
link report:
          50,002,185 sent on bridge_stream_1_a.l2 -> sink_1_a.rx (loss rate: 0%)
          50,001,930 sent on bridge_stream_1_a.stream_1_b -> bridge_stream_1_b.stream_1_a (loss rate: 0%)
                 255 sent on bridge_stream_1_a.stream_2_a -> bridge_stream_2_a.stream_1_a (loss rate: 0%)
                 255 sent on bridge_stream_1_a.stream_2_b -> bridge_stream_2_b.stream_1_a (loss rate: 0%)
          50,001,930 sent on bridge_stream_1_b.l2 -> sink_1_b.rx (loss rate: 0%)
          50,001,930 sent on bridge_stream_1_b.stream_1_a -> bridge_stream_1_a.stream_1_b (loss rate: 0%)
                 255 sent on bridge_stream_1_b.stream_2_a -> bridge_stream_2_a.stream_1_b (loss rate: 0%)
                 255 sent on bridge_stream_1_b.stream_2_b -> bridge_stream_2_b.stream_1_b (loss rate: 0%)
          50,001,930 sent on bridge_stream_2_a.l2 -> sink_2_a.rx (loss rate: 0%)
                 255 sent on bridge_stream_2_a.stream_1_a -> bridge_stream_1_a.stream_2_a (loss rate: 0%)
                 255 sent on bridge_stream_2_a.stream_1_b -> bridge_stream_1_b.stream_2_a (loss rate: 0%)
          50,001,930 sent on bridge_stream_2_a.stream_2_b -> bridge_stream_2_b.stream_2_a (loss rate: 0%)
          50,001,930 sent on bridge_stream_2_b.l2 -> sink_2_b.rx (loss rate: 0%)
                 255 sent on bridge_stream_2_b.stream_1_a -> bridge_stream_1_a.stream_2_b (loss rate: 0%)
                 255 sent on bridge_stream_2_b.stream_1_b -> bridge_stream_1_b.stream_2_b (loss rate: 0%)
          50,001,930 sent on bridge_stream_2_b.stream_2_a -> bridge_stream_2_a.stream_2_b (loss rate: 0%)
          50,001,930 sent on source_1_a.tx -> bridge_stream_1_a.l2 (loss rate: 0%)
          50,001,930 sent on source_1_b.tx -> bridge_stream_1_b.l2 (loss rate: 0%)
          50,001,930 sent on source_2_a.tx -> bridge_stream_2_a.l2 (loss rate: 0%)
          50,001,930 sent on source_2_b.tx -> bridge_stream_2_b.l2 (loss rate: 0%)
```

[1]:  I was counting packets received by the sinks, now I count packets transmitted to or dropped by the sink links. I believe that is a better metric, since it doesn't involve the sink app.

eugeneia, 2015-07-15:
@alexandergall WIth your bug fix I get better performance but worse overall results (e.g. stray packets):

```
snabb snabbmark labswitch 2e8 64 2 (WITH ALEX' FIX)
Stream 1:
Sending from 00:00:00:00:01:01 to 00:00:00:00:01:02
Sending from 00:00:00:00:01:02 to 00:00:00:00:01:01
Stream 2:
Sending from 00:00:00:00:02:01 to 00:00:00:00:02:02
Sending from 00:00:00:00:02:02 to 00:00:00:00:02:01
Switched 200.0 million packets in 89.20 seconds (rate: 2.2 Mpps).
Stream 1: 150.00  million packets
Stream 2: 50.00  million packets
link report:
          43,750,860 sent on bridge_stream_1_a.l2 -> sink_1_a.rx (loss rate: 41%)
          25,000,455 sent on bridge_stream_1_a.stream_1_b -> bridge_stream_1_b.stream_1_a (loss rate: 0%)
                 255 sent on bridge_stream_1_a.stream_2_a -> bridge_stream_2_a.stream_1_a (loss rate: 0%)
                 255 sent on bridge_stream_1_a.stream_2_b -> bridge_stream_2_b.stream_1_a (loss rate: 0%)
          25,000,455 sent on bridge_stream_1_b.l2 -> sink_1_b.rx (loss rate: 66%)
          25,000,455 sent on bridge_stream_1_b.stream_1_a -> bridge_stream_1_a.stream_1_b (loss rate: 0%)
                 255 sent on bridge_stream_1_b.stream_2_a -> bridge_stream_2_a.stream_1_b (loss rate: 0%)
                 255 sent on bridge_stream_1_b.stream_2_b -> bridge_stream_2_b.stream_1_b (loss rate: 0%)
          25,000,455 sent on bridge_stream_2_a.l2 -> sink_2_a.rx (loss rate: 0%)
          25,000,455 sent on bridge_stream_2_a.stream_1_a -> bridge_stream_1_a.stream_2_a (loss rate: 0%)
          25,000,455 sent on bridge_stream_2_a.stream_1_b -> bridge_stream_1_b.stream_2_a (loss rate: 0%)
          25,000,455 sent on bridge_stream_2_a.stream_2_b -> bridge_stream_2_b.stream_2_a (loss rate: 0%)
          25,000,455 sent on bridge_stream_2_b.l2 -> sink_2_b.rx (loss rate: 0%)
          25,000,455 sent on bridge_stream_2_b.stream_1_a -> bridge_stream_1_a.stream_2_b (loss rate: 0%)
          25,000,455 sent on bridge_stream_2_b.stream_1_b -> bridge_stream_1_b.stream_2_b (loss rate: 0%)
          25,000,455 sent on bridge_stream_2_b.stream_2_a -> bridge_stream_2_a.stream_2_b (loss rate: 0%)
          25,000,455 sent on source_1_a.tx -> bridge_stream_1_a.l2 (loss rate: 0%)
          25,000,455 sent on source_1_b.tx -> bridge_stream_1_b.l2 (loss rate: 0%)
          25,000,455 sent on source_2_a.tx -> bridge_stream_2_a.l2 (loss rate: 0%)
          25,000,455 sent on source_2_b.tx -> bridge_stream_2_b.l2 (loss rate: 0%)
```

alexandergall, 2015-07-15:
Forget my "fix". I wasn't thinking straight.  Sorry for the confusion.

eugeneia, 2015-07-15:
Yup, and that's whats happening. :+1:

alexandergall, 2015-07-15:
When I did some test runs, I sometimes saw flooding behaviour, i.e. the lookup of destiantion MAC addresses failed when it should have succeeded. The effect goes away when I make subtle changes somewhere else in the code or when I turn off JIT.  It looks similar to issues with optimizations I've had a while back when I wrote that code. I thought it was fixed with a bug-fix by Mike, but apparently not all of it :(

Do you see this at all?

alexandergall, 2015-07-16:
The last issue aside (I'll have to ponder this a bit), I've made some crude performance checks with the 2-bridge topology. In that case, the learning is just overhead. When I comment out that code, I get a baseline of about 30Mpps. I use `perf stat -e cycles` to get the total number of CPU cycles for the entire application and divide that by the number of processed packets, which gives a sort of per-packet cost. In my case, I get a little over 80 cycles per packet.

When I add just the code for the Bloom-filter encoding of the destination MAC address for each packet, I'm up to 590 cycles per packet, which is way more than it should be and doesn't match my previous tests at all :(

The encoding involves two passes of the Murmur hash function over the MAC address.  My previous tests estimated each pass at about 25 cycles. In this test, the first pass alone apparently amounts to about 240 cycles, while the second pass adds just 60 cycles (which is still too much). I thought maybe a caching effect, but `perf`'s cache-miss counters don't suggest that.

This is hard to analyze and it probably depends a lot on the dynamics of the compiler as well, particularly due to the inner loops of the Bloom filter.

lukego, 2015-07-16:
I wonder if another interesting variant to compare would be using a Lua table for the forwarding table and representing MAC address keys as Lua numbers (48-bit is okay as a double-float). That would probably have the same cache behavior as the bloom filter but execute as a LuaJIT builtin lookup primitive. Could help to isolate where the cost is coming from. (I have a hard time being really confident in interpreting `perf` counters based on my past experience.)

alexandergall, 2015-07-16:
@eugeneia the loss on some of the inter-bridge links comes from the fact that the learning bridge app processes only one source port per call of push(), see the comment there. I wanted to avoid yet another loop, but this may have been misguided, don't really know. If you iterate over all source ports, the packet rate becomes completely symmetric and there shouldn't be any loss.

eugeneia, 2015-07-16:
@alexandergall I noticed that too, but its not a huge issue. I don't have any loss either way on my current `lab-switch` head. I can't reproduce the non-deterministic flooding behavior you mentioned. I am currently working on replacing the bloom filter with an under-engineered table based implementation (inspired by Luke's suggestion) to see if that has any merit.

eugeneia, 2015-07-16:
So after a reckless attempt to reduce the cost of the learning bridge app I have learned that the order of expensiveness is as follows:

1. The hash function
2. Lookup in MAC table / filter
3. Lookup of destination port filter

I came to this conclusion by ripping out the bloom filter, replacing it with a Lua table and a dummy hash function for MAC addresses with neglectable cost. What is then left as the main cost is mapping the destination port to its MAC table and the source port to its destination port, e.g. plain Lua table lookups.

eugeneia, 2015-07-16:
My current feeling is that the learning bridge app is too generic for this particular use case. I don't see us getting 30Mpps optimizing it (but I might be wrong!). My next step would be to design a switching app catering just for lab switch and see how fast I can get it.

alexandergall, 2015-07-17:
The performance of the Bloom filter in this context is clearly disappointing.  It behaves completely different when compared to being run in isolation. This is the tracing JIT working against us this time, I guess :( Run in an isolated test, execution of both hash functions takes about 40 cycles, which is pretty good. I have no idea why it takes hundreds of cycles in this case.

Each round in the Bloom filter loop takes 5-10 cycles, so with 12 rounds in the default configuration (1000 MAC addresses with fals-positive rate <0.01%), storing of an address still takes around 150 cycles.

I don't have a hunch about what performance to expect from a different approach using an optimized standard MAC address table.  One thing to keep in mind is that any such algorithm will have some non-trivial scaling property as the number of stored addresses increases, where as the Bloom filter overhead is constant.

A disadvantage of the Bloom Filter is that a destination has to be looked up in every possible outgoing port, where as a standard address table lookup would yield the output port immediately.

The whole thing is pretty much an experiment, nobody uses this in a switch :)  It would still be neat if I could get it to perform well. I'm sure that it would give us an edge at least in the case where a large number of MAC addresses need to be learned.

Anyway, to get 30Mpps on a 2.4GHz CPU, you need to get it all done within 80 cycles. I would have thought that's impossible, but maybe not?

lukego, 2015-07-17:
30 Mpps would be valuable because then the switch should not be a limiting factor in any of our benchmarking activities. Each port could handle all packet sizes on both ingress and egress. Then we could hook up all of the lab ports to the switch and do all performance tests confident in the knowledge that packet drops will be due to the system under test and not the switch in the middle.

Today we sometimes do performance tests with packetblaster spamming 64-byte packets. This is handy when looking for applications' absolute limits on packet rate. It would be a pity if the lab switch prevented this kind of testing.

There are still quite a few tricks in are arsenal if needed...
- Optimize the hell out of the bottlenecks e.g. rewrite in assembler or get help from Mike Pall :-)
- Use two Snabb processes per port e.g. NIC-I/O and L2-Bridge running on separate hyperthreads.
- Require a fast CPU e.g. E5-2690V2 that has 10 x 3GHz cores.
- Hardware tricks e.g. offload switching onto the NIC or use multiple TX queues so that each process can transmit directly to any output port.

I see a lot of value in the exercise of seeing what performance we can achieve and what kind of tricks we have to use to get there. This experience should also help to inform other people considering Snabb based projects about what performance targets are easy / hard / impossible. I also think the lab switch would be a really cool application for other people to use too :).

lukego, 2015-07-17:
I wrote a really quick script to sanity-check parallelizing the labswitch benchmark:

```bash
#!/bin/bash
for core in "$@"; do
    sudo taskset -c $core  ./snabb snabbmark labswitch 10e6 64 &
done
for core in "$@"; do
    wait
done
```

The results I see are:

- 6.6 seconds running one process.
- 6.6 seconds running two processes on different cores.
- 13.6 seconds running two processes on a pair of hyperthreads.

So on this specific benchmark (max's lab-switch branch @ 7a2db8e8) there doesn't seem to be any value in hyperthreading. That could easily change in the future though e.g. if one process would be stalled on cache misses while the other could make progress. I would love to be better with [ocperf](https://lwn.net/Articles/556983/) to understand from very specific performance counters what work the CPU is bottlenecked on.

alexandergall, 2015-07-20:
I've got one step closer to understanding the bad performance of the Bloom filter code in the context of the labswitch. It appears that performance goes down the drain as soon as the JIT-ed code of the learning bridge contains a side trace. This is the distinguishing feature between fast and slow runs in my tests.

When perftool support is enabled in LuaJIT, `perf top` will usually show three traces using the bulk of the cycles (roughly corresponding to the main loop, the snabbmark loop and the main loop in the learning bridge). When one of these traces is a side trace of one of the others (indicated by, e.g. `TRACE 10/4` in the JIT dump file, which says that the trace is called from side exit #4 in trace 10 in this case), a penalty of several hundred cycles per packet appears, even if the side trace is basically doing nothing.

My current guess is that somehow the overhead of switching between traces in this scenario has a pretty large overhead (maybe due to stack slot loads or something like that). I wonder whether this has something to do with the "trace stitching" code that has been disabled in recently by Mike due to subtle bugs, if I remember correctly.

I don't really understand when exactly a side trace is created. For example, when I call the murmur hash function directly from within the learning bridge rather than through the bloom filter module, there is no side trace and only the expected 20 additional cycles due to the hash function.

Fascinating.... and frightening :)

lukego, 2015-07-20:
Indeed. I wonder if we will end up adopting programming practices like un-nesting our loops e.g. having separate loops over the available packets to learn their MAC addresses, lookup their output ports, and transmit them. I am warming up for another round of LuaJIT and Xeon investigations :).

eugeneia, 2015-07-21:
I replaced the learning bridge with a minimal switching app[1] with the following results:

```
Switched 200.1 million packets in 12.10 seconds (rate: 16.5 Mpps).
[Stream 1: 200.1 million packets]

Switched 200.0 million packets in 12.03 seconds (rate: 16.6 Mpps).
Stream 1: 100.02  million packets
Stream 2: 100.02  million packets

Switched 200.0 million packets in 14.15 seconds (rate: 14.1 Mpps).
Stream 1: 66.67  million packets
Stream 2: 66.67  million packets
Stream 3: 66.67  million packets

Switched 200.0 million packets in 16.11 seconds (rate: 12.4 Mpps).
Stream 1: 50.01  million packets
Stream 2: 50.01  million packets
Stream 3: 50.01  million packets
Stream 4: 50.01  million packets
```

The profiler report looks like this:

```
99%  Compiled
  -- 23%  switch.lua:lookup < switch.lua:method < app.lua:with_restart < app.lua:breathe < app.lua:main
  -- 15%  link.lua:receive < switch.lua:method < app.lua:with_restart < app.lua:breathe < app.lua:main
  -- 11%  switch.lua:insert < switch.lua:method < app.lua:with_restart < app.lua:breathe < app.lua:main
  --  7%  link.lua:transmit < switch.lua:method < app.lua:with_restart < app.lua:breathe < app.lua:main
  --  6%  switch.lua:hash < switch.lua:method < app.lua:with_restart < app.lua:breathe < app.lua:main
  --  6%  snabbmark.lua:method < app.lua:with_restart < app.lua:breathe < app.lua:main < snabbmark.lua:labswitch
  --  6%  ethernet.lua:is_mcast < switch.lua:method < app.lua:with_restart < app.lua:breathe < app.lua:main
  --  5%  link.lua:transmit < snabbmark.lua:method < app.lua:with_restart < app.lua:breathe < app.lua:main
  --  5%  packet.lua:free < basic_apps.lua:method < app.lua:with_restart < app.lua:breathe < app.lua:main
```

alexandergall, 2015-07-21:
Ha, I actually understand why side traces are created and how to avoid it in this case. Once again, it's a subtle detail but I've learned something new and it seems very clear now :) I'll write it up and post it to the ML when I find the time.

I've speeded up the hash function a bit. It now takes only ~15 cycles to pass over a MAC address in the ideal case, i.e. when all constants can be pre-loaded to registers outside the loop. When executed as part of a larger trace, like in this case, some operands are fetched from memory/spill slots due to register pressure. This increases the cycle budget by a little over a factor of 2. I don't know if that could be avoided.

eugeneia, 2015-07-21:
@alexandergall I might have noticed a bug in the learning bridge app: It checks if the destination is a multicast address to decide whether to learn the source address. If I understand correctly, that doesn't

eugeneia, 2015-07-21:
I accidentally clicked "close", sorry!

alexandergall, 2015-07-21:
@eugeneia No, the source address is always learnt. The check for the multicast address avoids storing the destination in a bloom filter object.

eugeneia, 2015-07-21:
@alexandergall Yeah I realized that while writing the comment and then accidentally hit close&comment, then I deleted my comment... sorry for the lost context!

alexandergall, 2015-07-21:
Ah, ok :) I should have seen that but I'm in a hacking frenzy and don't pay enough attention right now :)

eugeneia, 2015-07-29:
I want to wrap up my experiments here a bit. My current [lab-switch](https://github.com/eugeneia/snabbswitch/tree/lab-switch) branch uses a simple [switch app](https://github.com/eugeneia/snabbswitch/blob/lab-switch/src/apps/layer2/switch.lua) that manages to duplex switch 18 Mpps on grindelwald.

It can reach this throughput  because instead of hashing it simply uses the last three bytes of MAC addresses as learning keys (which as far as I can tell is good enough and has the nice side effect of yielding a reasonable upper limit of entries in the mac table: 2^24). It also does less, e.g. its an 1:N switch instead of an N:N bridge. One can get another performance boost out of its `samplegap` option which allow to specify that only 1 in N incoming packets is used to learn addresses. With `samplegap=200` I get 20 Mpps switched.

A test on real hardware (two packetblasters, two NICs, one labswitch instance with one intel app per NIC) gives me 7 Mpps. I don't think that's too bad is it? Unsure.

alexandergall, 2015-07-31:
Nice, though I do need the N:N case with split-horizon for my use case. How do you avoid loops without split-horizon? Do you just assume that the topology is loop-free?

I'm a bit uneasy about how colliding (truncated) MAC addresses are handled. In the worst case, the wrong host gets all the packets and the legitimate host none at all. That's a subtle failure that will drive you nuts when you hit it. And you can't detect it either, because the upper three bytes have been thrown away.

lukego, 2015-07-31:
I really like the "learn from every Nth packet" sampling optimization. I don't know what the worst-case band will be on overflooding but I suspect it is very modest.

I agree with @alexandergall that the truncated MAC method sounds very problematic. If you have two hosts on different ports with colliding MAC addresses then service will be badly disrupted for one or both.

Generally it is okay to sometimes send packets to too many ports but not to too few ports or to the wrong port.

lukego, 2015-07-31:
> A test on real hardware (two packetblasters, two NICs, one labswitch instance with one intel app per NIC) gives me 7 Mpps. I don't think that's too bad is it? Unsure.

Performance is never really good or bad in isolation. I think it's only meaningful to discuss in the context of a particular application.

For a lab switch that would connect the servers with each other and a 10 Gbps internet uplink, perhaps running heavy traffic (e.g. distributed filesystem like CEPH), I would estimate that 7 Mpps is okay. That could be a nice initial application for the lab switch.

In the future we can take on use cases that would require more performance e.g. being in the middle between 64-byte load generators and systems under test.

eugeneia, 2015-07-31:
> Nice, though I do need the N:N case with split-horizon for my use case. How do you avoid loops without split-horizon? Do you just assume that the topology is loop-free?

It's not a drop-in replacement for `bridge`, but rather a replacement for `bridge` in the `labswitch` use-case. I needed the split-horizon semantics in `bridge` because `labswitch` used it as a 1:N switch (every port has its own switch). I don't need the split-horizon semantics now because there are no loops as far as I can tell: every packet that reaches a switch is delivered to the port, the packets sent by the port are delivered to one or all of the N links selectively. Does that make sense?

> I'm a bit uneasy about how colliding (truncated) MAC addresses are handled. In the worst case, the wrong host gets all the packets and the legitimate host none at all. That's a subtle failure that [...]

> I agree with @alexandergall that the truncated MAC method sounds very problematic. If you have two hosts on different ports with colliding MAC addresses then service will be badly disrupted for one or both.

I am uneasy about this too as I am unable to prove/test real-world effect, here is the reasoning though:

* For the collision case to happen there has to be two devices of the same vendor with the same last three bytes on the network, which is not impossible but unlucky depending on the number of devices.
* Because every port has its own switch, the collision only causes a problem if the colliding devices talk to the same third port.
* Because of learning (last learned address overwrites table entry) the collision only causes a problem if above happens at the same time.

For a lab setup with 10-100 MAC addresses I don't see a problem, 1000 MAC addresses could cause slight packet loss depending on traffic, with 10000 MAC addresses you will almost definitely have a collision that will prevent the ports with the colliding MAC addresses to talk to the same third port at the same time. It really depends on what you expect from the switch.

lukego, 2015-07-31:
@eugeneia This is way too much voodoo. People want to be able to account for every packet when they are troubleshooting networking problems. If the argument for this optimization is so complex then people will have to think really hard about it every time they are troubleshooting any packet loss problem anywhere. People don't want to have to worry about this kind of stuff in their networks.

Also: Is there not a case where two hosts A and B have colliding MAC addresses, neither is actively transmitting, and one of them will receive all the incoming packets while the other receives none?

eugeneia, 2015-08-06:
Would it help if the failure case was better? E.g. I've tested modifying the table to store `key -> {mac, port}` and lookup to assert `key == key(mac) and mac == mac'` which is not much slower. This ensures that collisions in the key space never lead to packets being send to the wrong port. If keys collide they simply overwrite the previous record, e.g. unlearning it.

blakedotfr, 2015-12-19:
This is an excellent read & a fascinating discussion.

There are two L2 architectures that I recommend studying if you want to take this further:

 - RFC7432 Ethernet VPN.  One core concept:  the "mac default" route, that sends broadcast/multicast/unknown-unicast traffic to the "gateway(s)" which then act as a sort of central MAC database so the network can scale to a very large number of nodes without flooding issues.  Also, packets are only looked up at ingress, & then have an MPLS label attached; this avoids egress lookup (egress router just pops & sends).

 - RFC6329 Shortest Path Bridging.  Instead of using spanning-tree, IS-IS is extended to support a MAC address family.  SPF can be calculated on a per-VLAN or per-MAC basis.  In the case of per-MAC, you can mutlipath load-balance without fearing loops.  IS-IS distributes the bridging table around the network, so unknown-unicast flooding is reduced.