snabbco/snabb #572: Idea: Garbage collector for packets

lukego, 2015-07-21:
Here is an idea: To use a simple garbage collector to put freed packets back onto the freelist. (based on #565 and discussions with Juho Snellman.)

The idea is to introduce a packet garbage collector to give us these benefits during multiprocess execution:
1. Allow any process to free any packet.
2. Allow reclaiming packets that were being processed by a crashed process.
3. Allow reclaiming packets that were on the freelist of a crashed process.
4. Allow unused DMA memory to be detected and freed to the operating system.

Sounds good so far?

Here is the idea, which is not fully fleshed out and may be a little oversimplified.

First we add `owner` and `user` metadata fields to `struct packet`:
```c
struct packet {
  pid_t owner; // process that collects and allocates this packet
  pid_t user;  // process that is currently processing this packet
  int length;
  char data[10240];
}
```

When a process allocates a packet in new DMA memory it initializes the `owner` field with its own PID. Each time a packet is sent over a link the `user` field is set to the PID of the process that will receive from that link. We will need to add a `pid_t receiver` to `struct link` and then do:

```lua
function link.transmit (l, p)
   p.user = l.receiver
   ...
end
```

Now it starts to get interesting:

To free a packet we simply set its `user` field to a process that doesn't exist:

```lua
function packet.free (p)
   p.user = 0
end
```

and we rely on the `owner` process to notice this eventually and recycle the packet. How does that happen? Well, the owner periodically performs a garbage collection by scanning every packet it has ever allocated and checking whether the `user` points to a valid PID or not: if not then the packet is added back onto the freelist. This GC would be done at a regular interval e.g. once per breath or each time the packet freelist runs empty.

How about when a process crashes while using some packets? The packets will be reclaimed automatically: the GC will see that the `user` PID is not a running process and treat it the same as `0` i.e. know that the packet is not in use and put it back onto the free list.

How about reclaiming packets from the freelist of a dead process? That could also be detected by a larger GC sweep: scan every packet ever allocated by any process and recycle them if the `owner` PID is not a running process.

and returning packets to the OS e.g. after all traffic processes have terminated? Scan all packets allocated by all processes and free every HugeTLB memory block that does not contain any packets owned or used by a running process.

That's the idea.

I am assuming that we could come up with a simple and efficient routine for doing the garbage collection. also that all packet memory will be globally available via shared memory files (see also #571). There would seem to be quite some tricks available if needed e.g. running the GC incrementally, allocating some extra packets so that GC is not needed so frequently, or making `packet.free()` more cooperative with providing information that the GC can access quickly.

This might become a little tricky and need to the revised over time (e.g. for new applications that have massive amounts of packets live) but at least it would be an isolated module with a simple interface.

I suspect that a simple brute-force GC would be okay to start with. Likely a process will only have up to around 10,000 packets to keep track of. The packets can also be laid out in memory consistently and in principle that means the CPU cache prefetcher would detect the pattern of memory access ("stride") and make a linear scan across the packets fairly cheap. But this is speculation...

pavel-odintsov, 2015-07-21:
Awesome idea! Like it :) We could pass packets over multiple "iterations" and free they up when work finished.

andywingo, 2015-10-30:
What would happen if the owner goes to alloc a packet and there are none available?  Just return?

I would also wonder about PID collisions.  A GC would have to be super-reliable, and so you'd have to consider this possibility.  Perhaps instead of PID you could use cookies read from /dev/urandom when a Snabb process starts up, and reify the cookie -> PID association into ramfs at startup so that tools could know what apps are running.