snabbco/snabb #583: Design idea: String Theory

jgroom33, 2015-08-03:
Packet processing is equivalent to string manipulation

Packets are variable length strings.  Networking protocols have defined the layout of these strings.  Network protocols are the string schema.

The possible actions on a string are:
Create
     data Packet generator
     control packet creation (for peering)
Remove
     Strip vlan header, pop MPLS label, remove Layer 3 header
Update
     Everything else
Delete
     Used for bandwidth control

Since the update section encompasses so much, here’s some additional information:
Swapping one MAC for another MAC is an update
Translating a vlan is an update
Set ethertype is an update

The goal would be to implement these actions at the last second.  Potentially, they would even be exposed as part of the config.  In networking, we’re accustomed to implementing a protocol.

The challenge with discerning when to swap a MAC (sending packet out a new port) is defined in the flow.

In addition to the actions above, the following two implementations are necessary:
Store
Flow

The flow component is where everything is joined.
Wikipedia "The early C version has recently been upgraded to use C++ and Boost, and is now called CppFBP - it also contains an interface for the Lua scripting language, allowing FBP processes to be implemented as Lua routines.[28] This implementation also uses threads - however, it needs further tweaking (as of May 2015) to improve the human factors. 

A flow implements.
It provides the ability to learn a ‘protocol’:
     ARP implementation
     Read LLDP info from a string based on matching an ether type and storing info in the database
It provides the ability to process complex frame structure
     MAC in MAC
     QinQ

Intersection with SDN
This section proposes that current incarnation of SDN is not flexible enough for the what will happen next.  The popular idea of SDN makes a fundamental assumption.  It aims to separate the control and data planes.  It assumes  there is an underlying ontology.  Open flow assumes there are underlying tables to manipulate.  Another view (possibly an opinion) is that SDN attempts to operate based on the ‘golden packet’.  This is the concept that all packets conform to this construct and that packet flow is defined by a standardized process.

String theory has no preconceived thoughts of a ‘golden packet’ or how to store any information during control protocol interaction.

For study:
The intersection of data store and flow

lukego, 2015-08-07:
I very much like the idea of treating network traffic as an infinite stream of bytes.

I feel like networking has traditionally been regarded as a system programming problem (interrupts, spinlocks, etc) but on modern x86 it is more of a High Performance Computing problem (don't let the processing pipeline stall).

To process traffic as a sequential stream of bytes - like unix tools processing strings - would seem like a very interesting paradigm and extremely well suited to modern CPUs.

Is that at all in line with how you are thinking?

jgroom33, 2015-08-10:
Exactly.  If this is applied to the scalable switch [thread](https://github.com/SnabbCo/snabbswitch/issues/555) the concept would be manifest as follows:
A switch is a table that contains a set of strings (mac addresses)
An arbitrary set of data streams from an arbitrary list of ports can be attached to that table.  It should be possible to connect (broadcast between) 2 switch tables together if desired.

The ?only? thing that makes this memory table a switch is what occurs when a match is triggered (send frame to matching port) and what occurs when a match is not triggered (broadcast to all ports).

It should be the responsibility of something else to decide whether a loop does/should exist.  If it is desired to implement spanning tree within the (tables) switches of the heap, then that is one possible implementation.

I think that implementing FBP will support the necessary flexibility.  turning on/off a flow would be synonymous to enabling/disable a control protocol.

note: flow is a difficult term to use in describing the above concept as it contains a definition already related to packet transport.  A flow is not meant in that context above.

alexandergall, 2015-08-10:
I don't think it's relevant in your context, but I just want to make sure there are no misconceptions: you seem to assume that a switch manipulates the destination address of  a packet during forwarding (that's what you mean by "swap a MAC" if I understand correctly). That's not how a switch works, i.e. the packet is forwarded unmodified.

Unfortunately, I'm unfamiliar with the conecpt of FBP, so I can't comment on its application to packet processing, at least not with a substantial amount of reading :)

jgroom33, 2015-08-10:
Agreed. Mac swap is a routing function.

A switch does similar actions:
Change tpid on egress frames (0800,8100,88a8) - though, this is also an
insert
Vlan translate (stamp)

I'm curious if all functions can be generically implemented

Can a 'config' be a list of grep/pipe/modify commands?

jgroom33, 2015-11-18:
This presentation is promoting a similar concept:
http://opennetsummit.org/blog/2015/11/webinar-replay-programming-the-network-dataplane-in-p4/

My personal preference would be to design the flows in a visual environment similar to the goals of http://noflojs.org/

closed 2021-05-28 by jgroom33