snabbco/snabb #592: Low performance for VM2VM in v2015.08

virtualopensystems-nnikolaev, 2015-08-12:
We are using the following script for VM2VM performance evaluation:

```lua
local VhostUser = require("apps.vhost.vhost_user").VhostUser

local c = config.new()
config.app(c, "vh1", VhostUser, {socket_path="/tmp/vh1.sock", is_server=false})
config.app(c, "vh2", VhostUser, {socket_path="/tmp/vh2.sock", is_server=false})

config.link(c, "vh1.tx -> vh2.rx")
config.link(c, "vh2.tx -> vh1.rx")

engine.configure(c)
engine.main()
```
This script is run using `snsh` and attached to the NUMA node where the 2 VMs are running.
Inside the VMs we use a packet generator based on DPDK from here http://dpdk.org/browse/apps/pktgen-dpdk/ The performance test is to send 64byte UDP packets from one VM and measure the received ones using `pktgen-dpdk` in the other VM.

<pre>
+-----+---------+     +-----+---------+                  
|     |         |     |     |         |                  
|VM1  | PKTGEN  |     |VM2  | PKTGEN  |                  
|     |         |     |     |         |                  
|     +---------+     |     +---------+                  
|     |         |     |     |         |                  
|     |  DPDK   |     |     |  DPDK   |                  
|     |         |     |     |         |                  
+-----+-----+---+     +-----+---+-----+                  
            |                   |                        
            |                   |                        
+-----------v-------------------v-----+                  
|                                     |                  
|           SnabbSwitch               |                  
|                                     |                  
|       snsh vhost2vhost.lua          |                  
|                                     |                  
+-------------------------------------+                  

</pre>

The test are run on a `E5-2637 v3 @ 3.50GHz` dual socket machine.

When using `v2015.07.1` we are seeing about `12.7 Mpps` passed to the receiving machine.
With the latest `master` aka `v2015.08` the same performance test shows `9.3 Mpps`.

lukego, 2015-08-12:
Great catch!

@eugeneia has recently added a performance regression on the CI that I would expect to catch this. It is running the classic DPDK packet forwarding benchmark (loadgen->snabbnfv->vm->snabbnfv). However, this test may have been enabled for the first time after the regression was introduced.

@eugeneia can your regression test detect the performance difference between `v2015.07` and `v2015.08`? That would be very cool :-).

lukego, 2015-08-12:
@eugeneia also is there a simple instruction for how others (e.g. me) can run that test case manually?

eugeneia, 2015-08-12:
@lukego I assume you want to test `src/program/snabbnfv/packetblaster_bench.sh` (which is self documenting). Since that does not exist in `v2015.07` my approach would be to backport #547 into `v2015.07`.

lukego, 2015-08-13:
> which is self documenting

Point taken :-).

The challenge to quick testing now is that I am on nixos and test-env assumes that quite a lot of diverse software is installed and available e.g. everything needed to compile QEMU. I suppose that this is the problem that the Dockerized test environment will solve.

lukego, 2015-08-13:
In the near future this kind of problem should be the perfect use case for Docker based testing (#588). If the test environment is based on Docker then it should be a one-liner for anybody to run and reproduce the problem. I would just type `docker run vosys/vm2vmbench` and the rest is automatic.

That would be wonderful because otherwise it can take anything from an hour to a day to setup a test environment by hand when dealing with complex software e.g. installing DPDK inside the guests and so on.

nnikolaev-virtualopensystems, 2015-08-13:
I did some bisecting and I found which commit causes this low perfromance in our test scenario.

This is the commit after which the performance drops down:
7063865 Revert "Merge branch 'restore-snabbtop' into next"

This is really strange as this commit only remove some code related to statistics in the `app.lua`.

This is the last known good commit:
8ecc0c0 ipv6:pton: Fix error on invalid address

Any hints how to profile with `snsh` will be useful.

lukego, 2015-08-14:
@nnikolaev-virtualopensystems Great detective work. I also don't immediately see why this specific commit would be the issue. Let me braindump a little.

There has previously been a performance regression related to snabbtop that we tried to resolve with #568. That was an extremely hard to profile issue: we were accessing many counters on 4096-byte aligned addresses and CPU cache could not handle that. The backstory of finding that problem is on #558.

Our solution was to use a double-buffer to access the counters less frequently. However, we still write them once every 100 breathe() loops and maybe that is too often. Could be worth changing `app.breathe()` to call `counter.commit()` less often e.g. every 10000 loops and see if that affects the results.

@eugeneia has added a standard DPDK-based packet forwarding benchmark on the master branch now. I have tried to reproduce the issue with that but haven't seen it yet (but can be that I am making a git mistake when backporting it onto 8ecc0c0). It would be very useful if we could reproduce the problem using that test since it is the easiest setup for multiple people to run and also it is running automatically on the CI. If you can see the problem using that benchmark, or an adaptation of that benchmark to be more like your environment (one-way packet send instead of two-way), then that could help me to reproduce it over here (just one idea).

(Hopefully once we move our test environments over to Docker images it will be easier to run the test with an arbitrary version of the software. Currently the test scripts and the code are living in the same directory tree and it is not so easy to run a new test with older software from before the test was added. I know, I know, I am expecting Docker to solve all of our problems at once, we will see :))

Here is how that test runs:

```
$ sudo TESTPCI0=0000:01:00.0 TESTPCI1=0000:01:00.1 TELNET_PORT=5000 program/snabbnfv/packetblaster_bench.sh
...
load: time: 1.00s  fps: 4,756,445 fpGbps: 2.778 fpb: 218 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,935,922 fpGbps: 5.803 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,862,646 fpGbps: 5.760 fpb: 255 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,944,067 fpGbps: 5.807 fpb: 255 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,957,900 fpGbps: 5.815 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,932,911 fpGbps: 5.801 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,922,261 fpGbps: 5.795 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,924,639 fpGbps: 5.796 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,919,184 fpGbps: 5.793 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,941,215 fpGbps: 5.806 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,945,911 fpGbps: 5.808 fpb: 255 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,508,429 fpGbps: 5.553 fpb: 248 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,984,065 fpGbps: 5.831 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,972,250 fpGbps: 5.824 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,949,254 fpGbps: 5.810 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,983,303 fpGbps: 5.830 fpb: 256 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,204,803 fpGbps: 5.376 fpb: 242 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,355,580 fpGbps: 5.464 fpb: 242 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,196,086 fpGbps: 5.371 fpb: 249 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 9,951,668 fpGbps: 5.812 fpb: 256 bpp: 64   sleep: 0   us
load: time: 0.91s  fps: 9,988,466 fpGbps: 5.833 fpb: 256 bpp: 64   sleep: 0   us

Processed 100.0 million packets in 20.40 seconds (6400007744 bytes; 2.51 Gbps)
Made 890,951 breaths: 112.24 packets per breath; 22.90us per breath
Rate(Mpps):     4.901

lukego, 2015-08-14:
For profiling with `snsh` here is one idea:

    snabb snsh -jp=z -p snabbnfv ...

which should tell you how time is spent in "zones" (z) which is separately for each app i.e. which part is getting slower of vhost-user vs intel_app vs engine.

You could also profile with perf stat e.g.

    perf stat -e instructions,cycles,L1-dcache-misses,cache-misses,dTLB-load-misses,dTLB-store-misses,iTLB-load-misses

which may lead to some interesting information. For example if you see many more `instructions` for the same work then it might point to the JIT. If you see lower instructions per cycle it might point to the CPU.

I have been using a personal blog lately to explore low-level profiling issues around both LuaJIT and Xeon E5. I am starting at the most basic and working my way up to Snabb Switch. If you are interested there are four entries starting with lukego/blog#5. Could make LuaJIT slightly less mysterious :).

eugeneia, 2016-03-14:
I am closing this because it seems to be a stale issue. Please reopen/object in case this is still relevant.

closed 2016-03-14 by eugeneia