snabbco/snabb #611: LuaJIT: Improve trace-granularity profiling

lukego, 2015-09-07:
I want to improve LuaJIT's profiler to help me understand what compiled traces are relevant to performance of an app or program, how those traces fit together, and what code is executed in each trace. I have been exploring these issues in tiny examples like lukego/blog#8 but really I want this to be easy to do in the context of real Snabb Switch applications.

The problem historically is a "needle in a haystack" issue. If you enable `-jdump` then you end up with thousands of lines of output that you have to make sense of. Even if you know the numbers of the traces that use most CPU it is really a manual effort to understand what code they represent and why we have the set of traces that we do. I want to improve this. (Really what I want is to be able to understand what machine code is running for each app and be able to account for why but this will have to come in baby steps.)

I have extended the LuaJIT profiler now to add a significant chunk of what I want. The code is in commit 12b46070c428613d28eb91b14d1272f8116ea168 and the output looks like this (running on the rate_limiter selftest as an example):

```
$ sudo ./snabb snsh -jdump=im,x -jp=v -t apps.rate_limiter.rate_limiter
47%  TRACE  16 (14/4)   ->14       rate_limiter.lua:82
22%  TRACE  15          ->loop     basic_apps.lua:82
11%  TRACE  14          ->loop     rate_limiter.lua:73
10%  TRACE  12          ->loop     basic_apps.lua:26
 4%  TRACE  18 (12/13)  ->14       basic_apps.lua:25
 4%  TRACE  24 (19/10)  ->12       app.lua:293
```

This immediately looks suspicious: the rate limiter has two traces and the side-trace is burning a lot of CPU cycles. This could potentially be a bad case like lukego/blog#8 if there is an unbiased branch deciding whether to keep or drop packets. It could be tempting to experiment with either making the whole app compile down to one trace or splitting the two cases (forward vs. drop) into separate root traces instead of having a side trace.

I will send this LuaJIT patch as a PR to the snabbswitch repository once #609 lands on `next` to make this possible. Meanwhile if you want to experiment with it you can apply the patch manually somehow.

(This could be interesting for you, @alexandergall.)

alexandergall, 2015-09-07:
Very useful, thanks! Now I don't have to collect this information by hand via `perf top` and the dump file any more. 

I've just rewritten my learning bridge app using a more standard MAC table instead of a Bloom filter. The first version suffered from the "hot side trace" syndrome as well. I'll try to write it all up because I learned a lot during this exercise.  The upshot is that it is non-trivial to influence the compiler's decisions which traces to select, i.e. it's easy to say that we would like to have separate traces for something, but it can be very hard to achieve and the effect can be easily lost when the compiler's strategy changes in a future release.  For the bridge app, I ended up having to write some code in C to hide it from the compiler. A bit ugly but it made all the difference.

kbara, 2015-09-22:
This is a wonderful patch. As a convenience for anyone who wants to try it out before it's merged, I've created https://github.com/Igalia/snabbswitch/tree/lukegosubtree2_nutmeg_and_611profiling , which consists of your subtree2 branch, cherry-picked PRs 619, 620, and 621 (profiling enhancements that PR 617, your subtree2 branch, mentions), and this patch, since that takes a while to put together. No guarantees, but it seems to work for me:
```
 % sudo ./snabb snsh -jdump=+rs,x -jp=v -t apps.rate_limiter.rate_limiter
48%  TRACE  13          ->loop     basic_apps.lua:27
40%  TRACE  17          ->loop     basic_apps.lua:83
 7%  TRACE  16          ->loop     rate_limiter.lua:73
 4%  TRACE  19 (16/4)   ->16       rate_limiter.lua:82
```

lukego, 2015-09-23:
Thanks!

The next step I am working on is to separately track the "loop" part of a trace from the "straight" part. The idea being that it is very interesting to know whether a trace is spending its time staying inside the loop (probably efficient with lots of work "hoisted" out of the way) or in the straight part (probably expensively being re-entered from a side-trace). Really rough patch here: https://gist.github.com/lukego/d801cc7ad5fa68d884fb

Example output on rate_limiter:

```
   97  TRACE  17:LOOP (13/4)   ->13       rate_limiter.lua:82
   55  TRACE  10:LOOP          ->loop     basic_apps.lua:26
   54  TRACE  13               ->loop     rate_limiter.lua:73
   11  TRACE  39:LOOP (13/17)  ->15       link.lua:70
    9  TRACE  22:LOOP (16/2)   ->10       timer.lua:46
    8  TRACE  10               ->loop     basic_apps.lua:26
```

However I am not satisfied with the code yet. I "back tested" it on the example from lukego/blog#8 and expected to see three lines reported (Trace A, Trace A:LOOP, and Side Trace B) but actually I see only this:

```
  409  TRACE   3               ->loop     s2.lua:5
```

So there is something wrong in my mental model. I have been digging into how the profiler works and what I see is that there is not a 1:1 relationship between a SIGPROF and a callback to the profiler. Instead multiple samples are being coalesced together into a smaller number of callbacks. I suspect this is costing resolution i.e. all samples are being counted as the same code location in this example even though multiple traces are executing. (Could be that the results for rate_limiter are also misleading for the same reason but that it is less obvious.)

I am pursuing an idea for improving the situation :).

eugeneia, 2016-01-06:
@lukego What is the status here. Did any code related to this issue land?