snabbco/snabb #613: Long NIC initialization for X540-AT2

pavel-odintsov, 2015-09-10:
Hello, folks!

I'm using FireHose app for traffic processing. 

And I have two NIC models:
```bash
lspci |grep Eth
03:00.0 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
03:00.1 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
04:00.0 Ethernet controller: Intel Corporation Ethernet Controller 10-Gigabit X540-AT2 (rev 01)
04:00.1 Ethernet controller: Intel Corporation Ethernet Controller 10-Gigabit X540-AT2 (rev 01)
```

And I'm using following code for tests:
```bash
cd /usr/src
git clone https://github.com/FastVPSEestiOu/fastnetmon.git
cd fastnetmon/src/tests/snabb
```

My current SnabbSwitch branch is "next". I have checked "master" branch too.

And when I have specified X540-AT2 NIC in --input I have really HUGE time for NIC "Initialization":
```bash
time /usr/src/snabbswitch/src/snabb firehose  --input 0000:04:00.0 --input 0000:04:00.1 /usr/src/fastnetmon/src/tests/snabb/capturecallback.so
Loading shared object: /usr/src/fastnetmon/src/tests/snabb/capturecallback.so
Initializing NIC: 0000:04:00.0
!!! 10 seconds lag here !!!
Initializing NIC: 0000:04:00.1
!!! And 10 seconds lag here !!!
Initializing callback library
Processing traffic...
^CWe caught SINGINT and will finish application

real	0m32.881s
user	0m1.276s
sys	0m0.980s
```

But when I have switched to 82599 NIC for --input everything goes really fast:
```bash
time /usr/src/snabbswitch/src/snabb firehose --input 0000:03:00.0 --input 0000:03:00.1  /usr/src/fastnetmon/src/tests/snabb/capturecallback.so
Loading shared object: /usr/src/fastnetmon/src/tests/snabb/capturecallback.so
Initializing NIC: 0000:03:00.0
Initializing NIC: 0000:03:00.1
Initializing callback library
Processing traffic...
^CWe caught SINGINT and will finish application

real	0m2.184s
user	0m0.772s
sys	0m0.020s
```

Do you have any ideas what wrong with X540-AT2?

pavel-odintsov, 2015-09-10:
So I looked on strace and found so much nanosleep calls:
```bash
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
nanosleep({0, 100000}, NULL)            = 0
```

lukego, 2015-09-10:
There does seem to be a long delay before these NICs reach the "link up" state. I don't know why exactly but we made our selftest routine accept this lag.

Can you start actually passing traffic more quickly with the kernel driver? Could be that we can accelerate the init somehow.

pavel-odintsov, 2015-09-10:
Hello!

Thanks for answer! Will add this cards to blacklist! :)

So I could test it with netmap and share results.