snabbco/snabb #622: Mystery solved? Less than 14.8 Mpps w/ 64-byte packets

lukego, 2015-09-16:
Since forever we have noticed that when we tested with the minimum frame size we typically don't see the theoretical maximum packet rate of ~14.8 Mpps. Instead we see a lower rate typically around 12-13 Mpps. This has always been a little mystifying but because it has not been a practical problem for any application we have mostly ignored it. However, we are reminded of it any time we run `packetblaster` with a 64-byte packet input file.

I have looked into this a little bit now: first googling for a hypothesis and then running a small test to confirm. I think I have the answer. drumroll...

The problem seems to be that the 82599 cannot achieve full line rate on 64-byte packets with only one TX queue. I have seen multiple mailing list threads suggesting this.

I tested a multiqueue transmit by connecting one Source app to two Intel apps sharing the same port (VMDq mode) and then I do see the full 14.8 Mpps rate in total (half on each app).

app network fragment:

```
config.app(c, "source", require("apps.basic.basic_apps").Source)                                                                      
config.app(c, "nic1", Intel82599, { pciaddr = "0000:02:00.0", vmdq=true, mac="00:00:00:00:00:01" })                                 
config.app(c, "nic2", Intel82599, { pciaddr = "0000:02:00.0", vmdq=true, mac="00:00:00:00:00:02" })                                 
config.link(c, "source.tx -> nic1.rx")
config.link(c, "source.tx -> nic2.rx")
```

and result after running for 100 seconds:

```
link report:
         743,653,162 sent on source.tx -> nic1.rx (loss rate: 0%)
         743,643,612 sent on source.tx -> nic2.rx (loss rate: 0%)
```

which yields 14.872787 Mpps.

Compare with a single app/queue for transmit:

```
link report:
       1,290,189,769 sent on source.tx -> nic1.rx (loss rate: 0%)
```

i.e. 12.9 Mpps.

So if correct then what does this mean? I would say that for most apps it actually does not matter in practice but it is nice to have an explanation anyway. For certain apps though it could actually be worth using multiple queues (something to consider for packetblaster).

lukego, 2015-09-17:
Oh actually it looks like one queue is enough when descriptor prefetch is enabled. Here is a preliminary patch that does the job:

```diff
diff --git a/src/apps/intel/intel10g.lua b/src/apps/intel/intel10g.lua                                                                
index a3b1584..437082d 100644                                                                                                         
--- a/src/apps/intel/intel10g.lua                                                                                                     
+++ b/src/apps/intel/intel10g.lua                                                                                                     
@@ -421,6 +421,7 @@ function M_sf:init_transmit ()
    self.r.HLREG0:set(bits{TXCRCEN=0})                                                                                                
    self:set_transmit_descriptors()                                                                                                   
    self.r.DMATXCTL:set(bits{TE=0})                                                                                                   
+   self.r.TXDCTL:set(bor(bits{Enable=25, SWFLSH=26, hthresh=8}, 32))                                                                 
    return self                                                                                                                       
 end                                                                                                                                  
```

then I see 14.878 Mpps with just a source feeding one Intel app.

Have to verify this and check out the receive side too.

lukego, 2015-09-17:
@alexandergall Sorry if I am duplicating work that you are already doing. I know you mentioned wanting to solve this mystery and that kind of captured my imagination too :).

alexandergall, 2015-09-17:
I didn't have time to look at it yet, so I'm very happy about your progress :) It looks like you nailed it down pretty well.

alexandergall, 2015-09-17:
Thinking about it a bit more, I'm pretty sure that we're dealing with two distinct problems. One is on the sending side, which you solved. The other is on the receiving side, because I already verified that the receiving NIC is logging input drops. There doesn't seem to be a similar pre-fetching mechanism on the RX side, but I haven't looked very hard yet.

closed 2015-10-05 by eugeneia