snabbco/snabb #627: QEMU: Upgrade to 2.4.0

lukego, 2015-09-25:
I suggest that we change our recommended QEMU version from 2.1 (+ mandatory patches) to 2.4 (+ optional patches).

People have reported errors with various kinds of VMs (e.g. management ports on the Juniper vMX) and I believe this is actually resolved by upgrading to newer QEMUs that have fixes and improvements to vhost-user. I have verified this in manual testing with half a dozen operating systems: I often see problems with QEMU 2.1 but no problems with QEMU 2.4.

The standard upstream QEMU should be fine except for one aspect: it does not allow the vswitch to restart and reconnect to a virtual machine. (If you restart Snabb Switch then the VMs will not process traffic until you restart them too.) This is unfortunate: I like being able to restart at any time.

I have prepared a rebased QEMU branch that includes our patch for enabling reconnect: https://github.com/SnabbCo/qemu/tree/v2.4.0-snabb. The one patch is quite small and should actually apply cleanly to any QEMU version: https://github.com/SnabbCo/qemu/commit/c9cea8f431c929f70a9371f4b379ab66c15c5293.

The steps involved in "upgrading" our recommended QEMU would be:

1. Update the Getting Start Guide with new instructions on selecting a QEMU.
2. Update the CI test scripts to use the new recommended QEMU.
3. Post a note to the mailing list to ask people to check whether updating QEMU resolves their problems.

cc @mwiget @eugeneia

eugeneia, 2015-09-28:
I am having trouble getting a submodule pointing to https://github.com/SnabbCo/qemu to checkout the `v2.4.0-snabb` branch. How would I go about “updating” a submodule?

What I tried went as follows:

```
(cd to/submodule
    git fetch origin v2.4.0-snabb:v2.4.0-snabb
    git checkout v2.4.0-snabb)
git add to/submodule
git comit -m "blah"
git push
```

So far so good, but on another repository I was not able to sync:

```
git pull --rebase
git submodule update to/submodule
fatal: reference is not a tree: c9cea8f431c929f70a9371f4b379ab66c15c5293
Unable to checkout 'c9cea8f431c929f70a9371f4b379ab66c15c5293' in submodule path 'to/submodule'
```

I tried all the variants of this procedure I could find on the net and each failed with the error message above.

nnikolaev-virtualopensystems, 2015-09-28:
In similar situations `git submodule sync` helped me.

lukego, 2015-09-28:
Generally `git fetch` in the submodule directory should pick up missed commits too? (Surely not the official solution.)

eugeneia, 2015-09-29:
I think the issue is that I added the submodules with `--depth 1` because I wasn't feeling like downloading the whole kernel repository. The downside seems to be that clones of the repository have to manually fetch new revisions.

I have pushed an updated `eugeneia/snabb-nfv-test` image (tag `latest`) containing the `v2.4.0-snabb` version of QEMU.

lukego, 2015-09-29:
If submodules are a pain could consider a shell command to `git clone` instead.

eugeneia, 2015-12-17:
The [snabbco/nfv-dpdk2.1](https://hub.docker.com/r/snabbco/nfv-dpdk2.1/) image contains QEMU 2.4 and the newest DPDK. All without patches. I reckon we need to work on #665 to start recommending QEMU 2.4.

closed 2016-05-13 by eugeneia