snabbco/snabb #643: Solarflare driver segfaults on lib.nfv.config selftest

eugeneia, 2015-10-14:
Solarflare driver segfaults on  `lib.nfv.config` selftest. To reproduce run `make lib.nfv.config` with a Solarflare NIC. Result:

```
selftest: lib.nfv.config
testing:	program/snabbnfv/test_fixtures/nfvconfig/switch_nic/x
engine: start app id0_Virtio
engine: start app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
testing:	program/snabbnfv/test_fixtures/nfvconfig/switch_filter/x
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
engine: start app id0_Filter_in
load: time: 0.30s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
testing:	program/snabbnfv/test_fixtures/nfvconfig/switch_qos/x
engine: stop app id0_Filter_in
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
load: time: 0.30s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
testing:	program/snabbnfv/test_fixtures/nfvconfig/switch_tunnel/x
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
engine: start app id0_Tunnel
engine: start app id0_ND
Sending neighbor solicitation for next-hop 2::2
load: time: 0.28s  fps: 3         fpGbps: 0.000 fpb: 0   bpp: 86   sleep: 100 us
testing:	program/snabbnfv/test_fixtures/nfvconfig/scale_up/y
engine: stop app id0_Tunnel
engine: stop app id0_ND
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
engine: start app id1_Virtio
engine: start app id1_NIC
Opened SolarFlare interface p10p1 (MAC address 01:90:f5:ef:aa:67, MTU 1514)
load: time: 0.29s  fps: 6         fpGbps: 0.000 fpb: 0   bpp: 86   sleep: 100 us
testing:	program/snabbnfv/test_fixtures/nfvconfig/scale_up/x
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
engine: reconfig app id1_NIC
Opened SolarFlare interface p10p1 (MAC address 01:90:f5:ef:aa:67, MTU 1514)
engine: start app id63_Virtio
engine: start app id48_NIC
Opened SolarFlare interface p10p1 (MAC address 30:90:f5:ef:aa:67, MTU 1514)
engine: start app id10_NIC
Opened SolarFlare interface p10p1 (MAC address 0a:90:f5:ef:aa:67, MTU 1514)
engine: start app id52_Virtio
engine: start app id6_NIC
Opened SolarFlare interface p10p1 (MAC address 06:90:f5:ef:aa:67, MTU 1514)
engine: start app id45_NIC
Opened SolarFlare interface p10p1 (MAC address 2d:90:f5:ef:aa:67, MTU 1514)
engine: start app id34_NIC
Opened SolarFlare interface p10p1 (MAC address 22:90:f5:ef:aa:67, MTU 1514)
engine: start app id35_NIC
Opened SolarFlare interface p10p1 (MAC address 23:90:f5:ef:aa:67, MTU 1514)
engine: start app id40_NIC
Opened SolarFlare interface p10p1 (MAC address 28:90:f5:ef:aa:67, MTU 1514)
engine: start app id61_Virtio
engine: start app id61_NIC
Opened SolarFlare interface p10p1 (MAC address 3d:90:f5:ef:aa:67, MTU 1514)
engine: start app id60_Virtio
engine: start app id49_Virtio
engine: start app id59_Virtio
engine: start app id12_NIC
Opened SolarFlare interface p10p1 (MAC address 0c:90:f5:ef:aa:67, MTU 1514)
engine: start app id42_NIC
Opened SolarFlare interface p10p1 (MAC address 2a:90:f5:ef:aa:67, MTU 1514)
engine: start app id38_NIC
Opened SolarFlare interface p10p1 (MAC address 26:90:f5:ef:aa:67, MTU 1514)
engine: start app id24_NIC
Opened SolarFlare interface p10p1 (MAC address 18:90:f5:ef:aa:67, MTU 1514)
engine: start app id58_Virtio
[... 130 lines cut ...]
engine: start app id22_NIC
Opened SolarFlare interface p10p1 (MAC address 16:90:f5:ef:aa:67, MTU 1514)
engine: start app id15_NIC
Opened SolarFlare interface p10p1 (MAC address 0f:90:f5:ef:aa:67, MTU 1514)
engine: start app id18_Virtio
engine: start app id20_NIC
Opened SolarFlare interface p10p1 (MAC address 14:90:f5:ef:aa:67, MTU 1514)
engine: start app id2_NIC
Opened SolarFlare interface p10p1 (MAC address 02:90:f5:ef:aa:67, MTU 1514)
engine: start app id9_NIC
Opened SolarFlare interface p10p1 (MAC address 09:90:f5:ef:aa:67, MTU 1514)
engine: start app id19_Virtio
engine: start app id29_Virtio
engine: start app id17_NIC
Opened SolarFlare interface p10p1 (MAC address 11:90:f5:ef:aa:67, MTU 1514)
engine: start app id11_NIC
Opened SolarFlare interface p10p1 (MAC address 0b:90:f5:ef:aa:67, MTU 1514)
engine: start app id55_NIC
Opened SolarFlare interface p10p1 (MAC address 37:90:f5:ef:aa:67, MTU 1514)
engine: start app id7_NIC
Opened SolarFlare interface p10p1 (MAC address 07:90:f5:ef:aa:67, MTU 1514)
engine: start app id14_NIC
Segmentation fault (core dumped)
EXITCODE: 139
```

dpino, 2015-10-19:
I'm not sure about how to run it with _make_. I run it as:

```bash
$ sudo SNABB_TEST_INTEL10G_PCIDEVA=0000:86:00.0 ./snabb snsh -t program.snabbnfv.nfvconfig
```

This is what I got. Apparently it seems to work fine. Machine is grindelwald.

```
selftest: lib.nfv.config
testing:	program/snabbnfv/test_fixtures/nfvconfig/switch_nic/x
engine: start app id0_Virtio
engine: start app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
testing:	program/snabbnfv/test_fixtures/nfvconfig/switch_filter/x
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
engine: start app id0_Filter_in
load: time: 0.30s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
testing:	program/snabbnfv/test_fixtures/nfvconfig/switch_qos/x
engine: stop app id0_Filter_in
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
load: time: 0.27s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
testing:	program/snabbnfv/test_fixtures/nfvconfig/switch_tunnel/x
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
engine: start app id0_Tunnel
engine: start app id0_ND
Sending neighbor solicitation for next-hop 2::2
load: time: 0.29s  fps: 3         fpGbps: 0.000 fpb: 0   bpp: 86   sleep: 100 us
testing:	program/snabbnfv/test_fixtures/nfvconfig/scale_up/y
engine: stop app id0_Tunnel
engine: stop app id0_ND
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
engine: start app id1_Virtio
engine: start app id1_NIC
Opened SolarFlare interface p10p1 (MAC address 01:90:f5:ef:aa:67, MTU 1514)
load: time: 0.29s  fps: 7         fpGbps: 0.000 fpb: 0   bpp: 86   sleep: 100 us
testing:	program/snabbnfv/test_fixtures/nfvconfig/scale_up/x
engine: reconfig app id0_NIC
Opened SolarFlare interface p10p1 (MAC address 00:90:f5:ef:aa:67, MTU 1514)
engine: reconfig app id1_NIC
Opened SolarFlare interface p10p1 (MAC address 01:90:f5:ef:aa:67, MTU 1514)
engine: start app id63_Virtio
engine: start app id48_NIC
Opened SolarFlare interface p10p1 (MAC address 30:90:f5:ef:aa:67, MTU 1514)
engine: start app id10_NIC
Opened SolarFlare interface p10p1 (MAC address 0a:90:f5:ef:aa:67, MTU 1514)
engine: start app id52_Virtio
engine: start app id6_NIC
Opened SolarFlare interface p10p1 (MAC address 06:90:f5:ef:aa:67, MTU 1514)
engine: start app id45_NIC
Opened SolarFlare interface p10p1 (MAC address 2d:90:f5:ef:aa:67, MTU 1514)
engine: start app id34_NIC
Opened SolarFlare interface p10p1 (MAC address 22:90:f5:ef:aa:67, MTU 1514)
engine: start app id35_NIC
Opened SolarFlare interface p10p1 (MAC address 23:90:f5:ef:aa:67, MTU 1514)
engine: start app id40_NIC
Opened SolarFlare interface p10p1 (MAC address 28:90:f5:ef:aa:67, MTU 1514)
engine: start app id61_Virtio
engine: start app id61_NIC
Opened SolarFlare interface p10p1 (MAC address 3d:90:f5:ef:aa:67, MTU 1514)
engine: start app id60_Virtio
engine: start app id49_Virtio
engine: start app id59_Virtio
engine: start app id12_NIC
Opened SolarFlare interface p10p1 (MAC address 0c:90:f5:ef:aa:67, MTU 1514)
engine: start app id42_NIC
[... 138 lines cut ...]
engine: start app id15_NIC
Opened SolarFlare interface p10p1 (MAC address 0f:90:f5:ef:aa:67, MTU 1514)
engine: start app id18_Virtio
engine: start app id20_NIC
Opened SolarFlare interface p10p1 (MAC address 14:90:f5:ef:aa:67, MTU 1514)
engine: start app id2_NIC
Opened SolarFlare interface p10p1 (MAC address 02:90:f5:ef:aa:67, MTU 1514)
engine: start app id9_NIC
Opened SolarFlare interface p10p1 (MAC address 09:90:f5:ef:aa:67, MTU 1514)
engine: start app id19_Virtio
engine: start app id29_Virtio
engine: start app id17_NIC
Opened SolarFlare interface p10p1 (MAC address 11:90:f5:ef:aa:67, MTU 1514)
engine: start app id11_NIC
Opened SolarFlare interface p10p1 (MAC address 0b:90:f5:ef:aa:67, MTU 1514)
engine: start app id55_NIC
Opened SolarFlare interface p10p1 (MAC address 37:90:f5:ef:aa:67, MTU 1514)
engine: start app id7_NIC
Opened SolarFlare interface p10p1 (MAC address 07:90:f5:ef:aa:67, MTU 1514)
engine: start app id14_NIC
Opened SolarFlare interface p10p1 (MAC address 0e:90:f5:ef:aa:67, MTU 1514)
engine: start app id3_Virtio
engine: start app id26_Virtio
Sending neighbor solicitation for next-hop 2::2
```

eugeneia, 2015-10-19:
@dpino Interesting! If I run the selftest using make like so

```
sudo SNABB_TEST_INTEL10G_PCIDEVA=0000:86:00.0 make program.snabbnfv.nfvconfig
```

it will produce the segfault as described. That's definitely a hint to follow.

dpino, 2015-10-22:
Apparently the other command works, but actually status code is 139 (SIGSEGV):

```bash
Opened SolarFlare interface p10p1 (MAC address 0e:90:f5:ef:aa:67, MTU 1514)
engine: start app id3_Virtio
engine: start app id26_Virtio
Sending neighbor solicitation for next-hop 2::2
dpino@grindelwald src (master) $ echo $?
139
```

eugeneia, 2015-11-17:
Fixed by #664

closed 2015-11-17 by eugeneia