snabbco/snabb #658: Bad error message when no such PCI device

lukego, 2015-11-11:
The error message is cryptic when trying to open a PCI device that does not exist:

```
sudo SNABB_TEST_INTEL10G_PCIDEVA="0000:04:00.0" SNABB_TEST_INTEL10G_PCIDEVB="0000:04:00.1" ./snabb snsh -t apps.intel.intel_app
selftest: intel_app
100 VF initializations:

Running iterated VMDq test...
lib/hardware/pci.lua:131: assertion failed!
```

kbara, 2015-11-12:
Agreed; that one bit me, though a bit of reading source code and poking around made it clear.

eugeneia, 2015-11-16:
Since 2015.11 the error message is as follows (not much better):

```
selftest: intel_app
core/lib.lua:54: Unable to open file: /sys/bus/pci/devices/0000:04:00.0/vendor
stack traceback:
	core/main.lua:126: in function <core/main.lua:124>
	[C]: in function 'error'
	core/lib.lua:54: in function 'firstline'
	lib/hardware/pci.lua:38: in function 'device_info'
	apps/intel/intel_app.lua:150: in function 'selftest'
	program/snsh/snsh.lua:29: in function <program/snsh/snsh.lua:29>
	core/lib.lua:405: in function 'dogetopt'
	program/snsh/snsh.lua:63: in function 'run'
	core/main.lua:58: in function <core/main.lua:32>
	[C]: in function 'xpcall'
	core/main.lua:155: in main chunk
	[C]: at 0x0044ee90
	[C]: in function 'pcall'
	core/startup.lua:1: in main chunk
	[C]: in function 'require'
	[string "require "core.startup""]:1: in main chunk
```

eugeneia, 2016-01-06:
Fixed in #679.

closed 2016-01-06 by eugeneia