snabbco/snabb #661: Snabb NFV and OpenStack Liberty-Mitaka

lukego, 2015-11-16:
There is a lot of stuff we can do to update and improve our OpenStack support!

If you are interested in that topic then this issue can act as a synchronization point to align your own interests with everybody else's.

#### Overview

These days we should have a relatively easy time working with OpenStack. The major features we wanted to add have all landed upstream (chiefly vhost-user in QEMU, Libvirt, Nova, and Neutron). We can now maintain our Neutron drivers however we want: Neutron "spun out" all of their drivers because having one centralized development process had become a frustrating bottleneck. The task now should rather be to simply maintain our own software, test it with the latest releases that come down the pipeline, revise our patch sets (drop patches that are upstream and released), and integrate with any interesting new features that other people have developed.

#### Opportunities

I am aware of one new upstream feature that is important to us: OpenStack has added NUMA support. Hopefully we can use this to ensure NUMA-affinity between NIC+Snabb+QEMU for optimal performance.

I believe we only have one patch that has not landed upstream (*woohoo!*): making QEMU support reconnections from vhost-user sockets. That is to allow Snabb Switch restarts/upgrades with only momentary interruption of VM traffic (instead of having to restart the VMs). This is a very important feature and it would be great to find a way to make it work without patching QEMU.

### Work

Here are the ideas for improvements that I have been talking with people about over the Mitaka cycle (due April 2016). If anybody is interested in taking the lead on these or contributing to them then comments here are the place to discuss that.

- [ ] Review which patches we do/don't need for Liberty release (2015).
- [ ] Upstream any remaining patches for Mitaka release (2016).
- [ ] Move our Neutron drivers to their own repository and make them availabile for packaging.
- [ ] Implement NUMA-affinity in our Neutron drivers (if this is now possible in OpenStack).
- [ ] Automate deployment of OpenStack+SnabbNFV test environments.
- [ ] Migrate from our [custom management architecture](https://github.com/SnabbCo/snabbswitch/blob/master/src/program/snabbnfv/doc/architecture.md) to a more traditional one based on agents and message queues.
- [ ] Add support for one or more standard Neutron network "types": `vlan`, `flat`, `vxlan`. (The driver currently defers on those network types and only implements the TeraStream `zone` type. That is a definite barrier to most users.)

We don't actually have to do everything on this list. I just wanted to list everything that I have heard people talking about to invite contributions and avoid duplicating efforts. I will add more items that are suggested for tracking.

nnikolaev-virtualopensystems, 2015-11-24:
To run a VM with hugepages and shared memory (which is a requirement for vhost-user to work) using the upstream Nova, it is enough that the used flavor has `hw:mem_page_size=large` set.
This will work provided that the `libvirt` on the compute node is at least version `1.2.8` and `qemu` is at least `2.1.0`.
This means that the rellated patches (https://github.com/SnabbCo/nova/commit/b45de6475699ae9952913843317a5f694cb52282 and https://github.com/SnabbCo/libvirt/commit/940dbda356c7c47b3f443410e2d8b04c049b999e) patches are not needed anymore.

This patch https://github.com/SnabbCo/nova/commit/7b16b300c2a8570126870de57ab697e3c9a08bd6 is upstream, although from another committer. Still there is a tricky part on the neutron side how to set the vhost-user socket path, because this default https://github.com/SnabbCo/nova/commit/7b16b300c2a8570126870de57ab697e3c9a08bd6#diff-a5d7d7912f51e1c8fb10c94d5ecdc2caR70 is no longer in the code.

lukego, 2015-11-26:
Great info @nnikolaev-virtualopensystems, thanks! cc @domenkozar.