snabbco/snabb #663: Gremlin ping failures during NFV selfest

eugeneia, 2015-11-16:
Sometimes pinging fails during NFV selftest runs for unknown reasons. A recent example could be witnessed in #653. See https://gist.github.com/SnabbBot/42cd610ced8705ecbe2f for the log.

Have to investigate this further.

dpino, 2015-11-17:
Thanks for the pointer. I tried to reproduce the bug but I got stuck in the NFV test. Is there any way of launching the CI bot when tests fail?

eugeneia, 2015-11-17:
No and it shouldn't be necessary. In odd cases I can do it manually by removing the cached result from the CI instance. In general, when this happens there is a regression in Snabb Switch or a bug in the CI. If you experience a failure that you can not reproduce you should mention @eugeneia (me) so I can investigate. If its really unrelated to your PR, its my responsibility and I need to know about it.

eugeneia, 2015-11-30:
Gremlin strikes again: https://gist.github.com/SnabbBot/c00ac257cbb83af3954e

eugeneia, 2016-01-11:
And again in #700 : https://gist.github.com/SnabbBot/6f03246d0344f55ff1a2

eugeneia, 2016-02-26:
#736: https://gist.github.com/SnabbBot/279f6999c99689e2b24c

eugeneia, 2016-04-26:
#892: https://gist.github.com/SnabbBot/654445d188bf5446d2825142085dda9b