snabbco/snabb #666: Confusing error when running non-root on first execution

lukego, 2015-11-23:
The first time you run `snabb` on a machine it will create `/var/run/snabb`. This fails with a confusing error message if the process is not running as root:

```
[luke@chur:~]$ snabb
core/shm.lua:86: shm open error (7724/engine/breaths):No such file or directory
stack traceback:
	core/main.lua:116: in function <core/main.lua:114>
	[C]: in function 'error'
	core/shm.lua:86: in function 'map'
	core/counter.lua:48: in function 'open'
	core/app.lua:28: in main chunk
	[C]: in function 'require'
	core/main.lua:106: in function 'initialize'
	core/main.lua:35: in function <core/main.lua:32>
	[C]: in function 'xpcall'
	core/main.lua:121: in main chunk
	[C]: at 0x0044f740
	[C]: in function 'pcall'
	core/startup.lua:1: in main chunk
	[C]: in function 'require'
	[string "require "core.startup""]:1: in main chunk
```

The root cause is a permissions problem without adequate error checking:

```
mkdir("/var/run/snabb", 01777)          = -1 EACCES (Permission denied)
```

This should either be a fatal error or the process should find a way to tolerate this problem and execute normally.

xrme, 2015-12-30:
For what it may be worth, this error can happen when following the directions in the top-level README.md file.

The `$` prompt in the "How do I get started" section implies that running  `src/snabb --help` as a non-root user should be OK, but this will fail if `src/snabb --help` is the first time the program is ever run.

eugeneia, 2016-01-06:
Fixed in #679.

closed 2016-01-06 by eugeneia