snabbco/snabb #671: Fixed-function PMU counters not working?

andywingo, 2015-12-03:
For this lua file:

```lua
local pmu = require('lib.pmu')
local has_pmu_counters, err = pmu.is_available()
if not has_pmu_counters then
   error('No PMU available: '..err)
end

local function count_cycles(f, iterations)
   pmu.setup({"cycles"})
   local set = pmu.new_counter_set()
   pmu.switch_to(set)
   local res = f(iterations)
   pmu.switch_to(nil)
   pmu.report(pmu.to_table(set))
   return pmu.to_table(set).cycles, res
end

local function print_loop(count)
   for i=1,count do print(i) end
end

print(count_cycles(print_loop, 10))
```

If I run it like `sudo taskset -c 0 ./snabb snsh /tmp/test.lua` I get:

```
$ sudo taskset -c 0 ./snabb snsh /tmp/test.lua
1
2
3
4
5
6
7
8
9
10
EVENT                                             TOTAL
cycles                                                0
ref_cycles                                            0
instructions                                          0
cpl_cycles.ring0                                     13
cpl_cycles.ring0_trans                               13
cpl_cycles.ring123                               34,448
cycle_activity.cycles_l1d_pending                     0
cycle_activity.cycles_l2_pending                  5,442
cycle_activity.cycles_ldm_pending                42,706
cycle_activity.cycles_no_execute                 73,052
dsb2mite_switches.penalty_cycles                    374
0	nil
```

The cycles/ref_cycles/instructions are borked.  What is going on, I wonder?

I should note that I have this local modification:

```diff
diff --git a/src/lib/pmu.lua b/src/lib/pmu.lua
index 9255c33..68fd5f4 100644
--- a/src/lib/pmu.lua
+++ b/src/lib/pmu.lua
@@ -199,7 +199,7 @@ function setup (patterns)
       table.sort(set)
    end
    local ndropped = math.max(0, #set - pmu_x86.ngeneral)
-   if ndropped > 0 then set[pmu_x86.ngeneral+1] = nil end
+   while #set > pmu_x86.ngeneral do table.remove(set) end
    local cpu = cpu_set()[1]
    -- Enable all fixed-function counters (IA32_FIXED_CTR_CTRL)
    writemsr(cpu, 0x38d, 0x333)

```

I think just setting the element to nil doesn't truncate the length unless that's the last element.

andywingo, 2015-12-03:
I should note this is on an E5-2620v3 system, the same as interlaken.

andywingo, 2015-12-03:
perhaps relevant: https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg1031365.html

andywingo, 2015-12-03:
Sadly, if I try with inst_retired.any I don't get anything either on cycles or on inst_retired.any.

andywingo, 2015-12-03:
Aaaaaaaaah, finally got it I think.  We weren't enabling IA32_PERF_GLOBAL_CTRL.  Adding this fixes it for me afaiu:

```diff
diff --git a/src/lib/pmu.lua b/src/lib/pmu.lua
index 68fd5f4..581282d 100644
--- a/src/lib/pmu.lua
+++ b/src/lib/pmu.lua
@@ -201,6 +201,11 @@ function setup (patterns)
    local ndropped = math.max(0, #set - pmu_x86.ngeneral)
    while #set > pmu_x86.ngeneral do table.remove(set) end
    local cpu = cpu_set()[1]
+   -- All available counters are globally enabled
+   -- (IA32_PERF_GLOBAL_CTRL).
+   writemsr(cpu, 0x38f,
+            bit.bor(bit.lshift(0x3ULL, 32),
+                    bit.lshift(1ULL, pmu_x86.ngeneral) - 1))
    -- Enable all fixed-function counters (IA32_FIXED_CTR_CTRL)
    writemsr(cpu, 0x38d, 0x333)
    for n = 0, #set-1 do
```

kbara, 2015-12-03:
Nice catch!

andywingo, 2015-12-04:
Somehow I had gotten this to work on another system without the writemsr patch.  I think it's because I ran `perf stat true` beforehand, so it already set the PMU MSRs.  That would also explain why the code worked for Luke -- he probably tried using perf on that system already.  Just a theory though, I haven't verified it.

andywingo, 2015-12-08:
Fixed in #672

closed 2015-12-08 by andywingo