snabbco/snabb #673: Max PPS ramp up time depends on lookup table size

mwiget, 2015-12-05:
Discovered an interesting behavior in snabb when using large per-packet lookup table: The larger the hash table, the longer it takes to reach maximum packets per second (PPS). I've used binding table size of 0 (bypassing the lookup), 50k, 100k and 200k. It takes up to 13 seconds on an Intel(R) Xeon(R) CPU D-1540 @ 2.00GHz to reach the maximum PPS:

[image]

For my lightweight 4 over 6 AFTR prototype (https://github.com/mwiget/snabbswitch/tree/lwaftr/), I need to lookup the binding for each packet. The look up key for the encapsulation path (IPv4 packets) is the IPv4 destination address and TCP/UDP (or ICMP Id) 16 Bit mask. The key size is 32+16=48 Bits (fits in a Lua number) and the value is an IPv6 address. Initially I created another lookup table for the encapsulation path with the IPv6 address as key, but turns out there is an easier way: Packets from the lightweight B4 client contain the IPv4 packet within an IPv6 header, so its possible to do a look up based on the IPv4 header information and compare the IPv6 packet source address with the hash result.

The test result was generated with a little snabb program lwaftrbench: https://github.com/mwiget/snabbswitch/tree/lwaftr/src/program/lwaftrbench, which basically sends the content of two PCAP files in each direction thru the lwaftr app and reports statistics every second:

```
ipv6_pcap_file ---> lwaftr ---> Statistics
    Statistics <--- lwaftr <--- ipv4_pcap_file
```

The slow ramp up effect can be seen when using an IPv4 PCAP file with 50k unique packets (hitting 50,000 unique binding table entries):

```
cd src/program/lwaftrbench/tests
$ grep CPU /proc/cpuinfo |head -1
model name	: Intel(R) Xeon(R) CPU D-1540 @ 2.00GHz
../../../snabb lwaftrbench -D 30 snabb-port-50k.cfg b4-icmp-request-0138.pcap lwaftr-ipv4-50k.pcap
Configuration file: snabb-port-50k.cfg
running for 30 seconds
offset=0 len=6 -> psmask=0xfc00
50000 bindings parsed, using 1778 kBytes
ipv4_to_ipv6: 1496 bpp 0.013 MPPS, 0.152 Gbps.
ipv6_to_ipv4:   98 bpp 0.013 MPPS, 0.010 Gbps.
ipv4_to_ipv6: 1496 bpp 0.012 MPPS, 0.142 Gbps.
ipv6_to_ipv4:   98 bpp 0.012 MPPS, 0.009 Gbps.
ipv4_to_ipv6: 1496 bpp 0.009 MPPS, 0.113 Gbps.
ipv6_to_ipv4:   98 bpp 0.009 MPPS, 0.007 Gbps.
ipv4_to_ipv6: 1496 bpp 0.010 MPPS, 0.125 Gbps.
ipv6_to_ipv4:   98 bpp 0.010 MPPS, 0.008 Gbps.
ipv4_to_ipv6: 1496 bpp 0.607 MPPS, 7.266 Gbps.
ipv6_to_ipv4:   98 bpp 0.607 MPPS, 0.476 Gbps.
ipv4_to_ipv6: 1496 bpp 1.232 MPPS, 14.748 Gbps.
ipv6_to_ipv4:   98 bpp 1.232 MPPS, 0.966 Gbps.
ipv4_to_ipv6: 1496 bpp 1.233 MPPS, 14.758 Gbps.
ipv6_to_ipv4:   98 bpp 1.233 MPPS, 0.967 Gbps.
ipv4_to_ipv6: 1496 bpp 1.234 MPPS, 14.764 Gbps.
ipv6_to_ipv4:   98 bpp 1.234 MPPS, 0.967 Gbps.
ipv4_to_ipv6: 1496 bpp 1.230 MPPS, 14.715 Gbps.
ipv6_to_ipv4:   98 bpp 1.230 MPPS, 0.964 Gbps.
. . .
```

I'm curious to learn/find out what exactly happens here that will explain the jump in performance. My guess its memory caching by the CPU core. Running the same command on a server with no L3 cache  
shows the same effect:

```
$ grep CPU /proc/cpuinfo |head -1
model name	: Intel(R) Atom(TM) CPU  C2750  @ 2.41GHz
$ sudo ../../../snabb lwaftrbench -D 30 snabb-port-50k.cfg b4-icmp-request-0138.pcap lwaftr-ipv4-50k.pcap
Configuration file: snabb-port-50k.cfg
running for 30 seconds
offset=0 len=6 -> psmask=0xfc00
50000 bindings parsed, using 1778 kBytes
ipv4_to_ipv6: 1496 bpp 0.005 MPPS, 0.058 Gbps.
ipv6_to_ipv4:   98 bpp 0.005 MPPS, 0.004 Gbps.
ipv4_to_ipv6: 1496 bpp 0.005 MPPS, 0.056 Gbps.
ipv6_to_ipv4:   98 bpp 0.005 MPPS, 0.004 Gbps.
ipv4_to_ipv6: 1496 bpp 0.002 MPPS, 0.018 Gbps.
ipv6_to_ipv4:   98 bpp 0.002 MPPS, 0.001 Gbps.
ipv4_to_ipv6: 1496 bpp 0.004 MPPS, 0.052 Gbps.
ipv6_to_ipv4:   98 bpp 0.004 MPPS, 0.003 Gbps.
ipv4_to_ipv6: 1496 bpp 0.002 MPPS, 0.024 Gbps.
ipv6_to_ipv4:   98 bpp 0.002 MPPS, 0.002 Gbps.
ipv4_to_ipv6: 1496 bpp 0.001 MPPS, 0.015 Gbps.
ipv6_to_ipv4:   98 bpp 0.001 MPPS, 0.001 Gbps.
ipv4_to_ipv6: 1496 bpp 0.005 MPPS, 0.055 Gbps.
ipv6_to_ipv4:   98 bpp 0.005 MPPS, 0.004 Gbps.
ipv4_to_ipv6: 1496 bpp 0.001 MPPS, 0.018 Gbps.
ipv6_to_ipv4:   98 bpp 0.001 MPPS, 0.001 Gbps.
ipv4_to_ipv6: 1496 bpp 0.003 MPPS, 0.040 Gbps.
ipv6_to_ipv4:   98 bpp 0.003 MPPS, 0.003 Gbps.
ipv4_to_ipv6: 1496 bpp 0.002 MPPS, 0.027 Gbps.
ipv6_to_ipv4:   98 bpp 0.002 MPPS, 0.002 Gbps.
ipv4_to_ipv6: 1496 bpp 0.001 MPPS, 0.017 Gbps.
ipv6_to_ipv4:   98 bpp 0.001 MPPS, 0.001 Gbps.
ipv4_to_ipv6: 1496 bpp 0.004 MPPS, 0.051 Gbps.
ipv6_to_ipv4:   98 bpp 0.004 MPPS, 0.003 Gbps.
ipv4_to_ipv6: 1496 bpp 0.002 MPPS, 0.021 Gbps.
ipv6_to_ipv4:   98 bpp 0.002 MPPS, 0.001 Gbps.
ipv4_to_ipv6: 1496 bpp 0.001 MPPS, 0.015 Gbps.
ipv6_to_ipv4:   98 bpp 0.001 MPPS, 0.001 Gbps.
ipv4_to_ipv6: 1496 bpp 0.004 MPPS, 0.052 Gbps.
ipv6_to_ipv4:   98 bpp 0.004 MPPS, 0.003 Gbps.
ipv4_to_ipv6: 1496 bpp 0.002 MPPS, 0.021 Gbps.
ipv6_to_ipv4:   98 bpp 0.002 MPPS, 0.001 Gbps.
ipv4_to_ipv6: 1496 bpp 0.262 MPPS, 3.130 Gbps.
ipv6_to_ipv4:   98 bpp 0.262 MPPS, 0.205 Gbps.
ipv4_to_ipv6: 1496 bpp 0.424 MPPS, 5.077 Gbps.
ipv6_to_ipv4:   98 bpp 0.424 MPPS, 0.333 Gbps.
ipv4_to_ipv6: 1496 bpp 0.425 MPPS, 5.082 Gbps.
ipv6_to_ipv4:   98 bpp 0.425 MPPS, 0.333 Gbps.
ipv4_to_ipv6: 1496 bpp 0.424 MPPS, 5.073 Gbps.
ipv6_to_ipv4:   98 bpp 0.424 MPPS, 0.332 Gbps.
ipv4_to_ipv6: 1496 bpp 0.424 MPPS, 5.076 Gbps.
ipv6_to_ipv4:   98 bpp 0.424 MPPS, 0.332 Gbps.
ipv4_to_ipv6: 1496 bpp 0.424 MPPS, 5.077 Gbps.
ipv6_to_ipv4:   98 bpp 0.424 MPPS, 0.333 Gbps.
ipv4_to_ipv6: 1496 bpp 0.424 MPPS, 5.075 Gbps.
ipv6_to_ipv4:   98 bpp 0.424 MPPS, 0.332 Gbps.
ipv4_to_ipv6: 1496 bpp 0.424 MPPS, 5.075 Gbps.
ipv6_to_ipv4:   98 bpp 0.424 MPPS, 0.332 Gbps.
ipv4_to_ipv6: 1496 bpp 0.425 MPPS, 5.081 Gbps.
ipv6_to_ipv4:   98 bpp 0.425 MPPS, 0.333 Gbps.
ipv4_to_ipv6: 1496 bpp 0.424 MPPS, 5.073 Gbps.
ipv6_to_ipv4:   98 bpp 0.424 MPPS, 0.332 Gbps.
ipv4_to_ipv6: 1496 bpp 0.425 MPPS, 5.081 Gbps.
ipv6_to_ipv4:   98 bpp 0.425 MPPS, 0.333 Gbps.
ipv4_to_ipv6: 1496 bpp 0.424 MPPS, 5.076 Gbps.
ipv6_to_ipv4:   98 bpp 0.424 MPPS, 0.333 Gbps.
```

The hash table key is built by the following function (https://github.com/mwiget/snabbswitch/blob/lwaftr/src/apps/lwaftr/lwaftr.lua)

```
local function ipv4idkey(ipv4, id, psmask)
  local psid = bit.band(id,psmask)
  local ipv4psid = bit.band(0xffffffffffffLL, ipv4)
  local key = tonumber(bit.bor(bit.lshift(ipv4psid,16), psid))
  return key
end
```

and the IPv6 address is stored as cdata:

```
local in_addr6  = ffi.new("uint8_t[16]")
local result = C.inet_pton(AF_INET6, binding_ipv6_addr, in_addr6)
local psidshifted = bit.lshift(psid,16 - psoffset - psid_len)
local ipv4psid = ipv4idkey(ipv4, psidshifted, 0xffff)
map_ipv4psid_to_ipv6[ipv4psid] = in_addr6
```

Is this a result of L1/L2 cache? How can I find out?

virtuallynathan, 2015-12-05:
Definitely not a SnabbSwitch expert, but I've been trying to follow along. I would guess that making use of the PMU module: https://github.com/SnabbCo/snabbswitch/blob/master/src/lib/pmu.lua would allow you to see what the L1/L2/L3 cache hit ratio looks like as the execution ramps up, and things potentially start to be kept in one of the cache levels. 

Cool idea for an app, by the way!

mwiget, 2015-12-05:
Great suggestion! Thanks. I added some basic reporting to the main program lwaftrbench and see the difference in L2 cache hit/miss, but I can't get L3 cache counters. Somehow I'm using pmu.setup() wrong:

```
    pmu.setup({
      "mem_load_uops_retired.l._hit",
      "mem_load_uops_retired.l._miss",
      "mem_load_uops_retired.l3_hit",
      "mem_load_uops_retired.l3_miss"})
```

Doing a test run with single packet versus 50k packets, I clearly see the expected L2 cache misses at the beginning for the large packet capture. Would be great to get L3 counters. 

```
 $ sudo ./r2.sh
model name	: Intel(R) Xeon(R) CPU D-1540 @ 2.00GHz
=========================================================
Running snabb-port-200k.cfg
PCAP files v6:b4-icmp-request-1502.pcap v4:aftr-icmp-reply-0098.pcap ...
taskset -c 0 ../../../snabb lwaftrbench -D 5 snabb-port-200k.cfg b4-icmp-request-1502.pcap aftr-icmp-reply-0098.pcap
Configuration file: snabb-port-200k.cfg
running for 5 seconds
offset=0 len=6 -> psmask=0xfc00
200000 bindings parsed, using 15633 kBytes
ipv4_to_ipv6:  138 bpp 2.749 MPPS, 3.035 Gbps.
ipv6_to_ipv4: 1460 bpp 2.749 MPPS, 32.105 Gbps.
ipv4_to_ipv6:  138 bpp 2.778 MPPS, 3.067 Gbps.
ipv6_to_ipv4: 1460 bpp 2.778 MPPS, 32.451 Gbps.
ipv4_to_ipv6:  138 bpp 2.774 MPPS, 3.062 Gbps.
ipv6_to_ipv4: 1460 bpp 2.774 MPPS, 32.397 Gbps.
ipv4_to_ipv6:  138 bpp 2.780 MPPS, 3.069 Gbps.
ipv6_to_ipv4: 1460 bpp 2.780 MPPS, 32.467 Gbps.
EVENT                                             TOTAL
cycles                                   12,999,707,304
ref_cycles                                            0
instructions                             19,565,368,417
mem_load_uops_retired.l1_hit              4,763,231,439
mem_load_uops_retired.l1_miss               266,653,176
mem_load_uops_retired.l2_hit                104,591,830
mem_load_uops_retired.l2_miss               162,061,346
---------------------------------------------------------
=========================================================
Running snabb-port-200k.cfg
PCAP files v6:b4-icmp-request-1502.pcap v4:lwaftr-ipv4-50k.pcap ...
taskset -c 0 ../../../snabb lwaftrbench -D 5 snabb-port-200k.cfg b4-icmp-request-1502.pcap lwaftr-ipv4-50k.pcap
Configuration file: snabb-port-200k.cfg
running for 5 seconds
offset=0 len=6 -> psmask=0xfc00
200000 bindings parsed, using 15633 kBytes
ipv4_to_ipv6: 1496 bpp 0.009 MPPS, 0.110 Gbps.
ipv6_to_ipv4: 1460 bpp 0.009 MPPS, 0.108 Gbps.
ipv4_to_ipv6: 1496 bpp 0.008 MPPS, 0.090 Gbps.
ipv6_to_ipv4: 1460 bpp 0.008 MPPS, 0.088 Gbps.
ipv4_to_ipv6: 1496 bpp 0.005 MPPS, 0.055 Gbps.
ipv6_to_ipv4: 1460 bpp 0.005 MPPS, 0.053 Gbps.
ipv4_to_ipv6: 1496 bpp 0.003 MPPS, 0.040 Gbps.
ipv6_to_ipv4: 1460 bpp 0.003 MPPS, 0.039 Gbps.
EVENT                                             TOTAL
cycles                                   13,177,342,212
ref_cycles                                            0
instructions                             13,404,706,258
mem_load_uops_retired.l1_hit              1,804,519,341
mem_load_uops_retired.l1_miss               267,961,568
mem_load_uops_retired.l2_hit                  2,129,026
mem_load_uops_retired.l2_miss               265,832,542
---------------------------------------------------------
```

Need to investigate further how pmu finds matching counters. The CPU is D-1540 has L3 cache:

```
$ lstopo|grep L3
  Socket L#0 + L3 L#0 (12MB)
```

virtuallynathan, 2015-12-05:
Interesting! I found these docs/files on the D-1500's PMU's from Intel: 
[332427-001.pdf](https://github.com/SnabbCo/snabbswitch/files/52978/332427-001.pdf)
[xeon-d-1500-uncore-performance-monitoring.zip](https://github.com/SnabbCo/snabbswitch/files/52979/xeon-d-1500-uncore-performance-monitoring.zip)

lukego, 2015-12-07:
My overall impression is that there is a fair amount of variation between that PMUs support from one CPU model to the next.

Intel publish some CSV files that list all of the counters for each CPU model (identified with CPUID instruction) and Snabb Switch has a script to translate these tables into Lua and lookup what is supported on the running machine.

Here is the Lua file containing the per-CPU definitions and the script that generates it:

- [lib/pmu_cpu.lua](https://github.com/SnabbCo/snabbswitch/blob/master/src/lib/pmu_cpu.lua)
- [scripts/generate-pmu.sh](https://github.com/SnabbCo/snabbswitch/blob/master/src/scripts/generate-pmu.sh)

The CPU-ID string is created with [these lines in pmu_x86.dasl](https://github.com/SnabbCo/snabbswitch/blob/master/src/lib/pmu_x86.dasl#L69-L93) using the CPUID instruction.

virtuallynathan, 2015-12-07:
Looks like the D-1500 (Broadwell-DE) lives at the bottom of the file: https://github.com/SnabbCo/snabbswitch/blob/master/src/lib/pmu_cpu.lua#L10510

It seems that L3 cache stats should exist:

``` lua
    ["mem_load_uops_retired.l1_hit"] = 0x01d1,
    ["mem_load_uops_retired.l2_hit"] = 0x02d1,
    ["mem_load_uops_retired.l3_hit"] = 0x04d1,
    ["mem_load_uops_retired.l1_miss"] = 0x08d1,
    ["mem_load_uops_retired.l2_miss"] = 0x10d1,
    ["mem_load_uops_retired.l3_miss"] = 0x20d1,
    ["mem_load_uops_retired.hit_lfb"] = 0x40d1,
    ["mem_load_uops_l3_hit_retired.xsnp_miss"] = 0x01d2,
    ["mem_load_uops_l3_hit_retired.xsnp_hit"] = 0x02d2,
    ["mem_load_uops_l3_hit_retired.xsnp_hitm"] = 0x04d2,
    ["mem_load_uops_l3_hit_retired.xsnp_none"] = 0x08d2,
    ["mem_load_uops_l3_miss_retired.local_dram"] = 0x01d3

```
(TIL xsnp = cross-core snoop)

lukego, 2015-12-07:
(Probably we should have a simple way to print all available counters for the current CPU...)

lukego, 2015-12-07:
Oh I think I see the problem here.

The CPU can only track a limited number of general purpose event counters at a time: four when hyperthreads are enabled or eight when disabled. Marcel's examples show four general purpose counters so if he has hyperthreads enabled (BIOS) then the others are likely being skipped due to lack of hardware resources.

We should print a warning about this! The `pmu.setup()` function does return the number of counters that were dropped due to lack of resources but this is undocumented and should really become a warning with higher-level functions that call `setup()`.

The Linux perf framework avoids this problem with sampling. It rotates the logical counters across the available physical counters and then also reports the % of time that the counter was active. I am not sure whether something like this would make sense for us and if so whether it should be done inside the pmu module or above in some higher-level code e.g. the engine.

mwiget, 2015-12-07:
Great catch! Thanks. Indeed, when I specify just the L2 and L3 hit/miss counters, I get what I'm currently interested in. The main question I'm trying to answer for myself is the following: Given random traffic, at what point is the lookup impacting processes on other cores. I hope to get an indication from the L3 hit/miss ratio, though I'm aware from the limited reading I've done so far, I'm not sure if such a conclusion can be made ;-).

Anyway, just run some tests again with the following 4 counters:

```
    pmu.setup({
      "mem_load_uops_retired.l2.hit",
      "mem_load_uops_retired.l2.miss",
      "mem_load_uops_retired.l3_hit",
      "mem_load_uops_retired.l3_miss"})
```

This reports the following, first running 4k unique packets thru  hash table with 200k binding entries, then with 50k unique packets:

```
snabb lwaftrbench -D 30 snabb-port-200k.cfg b4-icmp-request-0138.pcap lwaftr-ipv4-4k.pcap
...
ipv4_to_ipv6: 1496 bpp 1.410 MPPS, 16.880 Gbps.
ipv6_to_ipv4:   98 bpp 1.410 MPPS, 1.106 Gbps.
EVENT                                             TOTAL
cycles                                   73,734,875,518
ref_cycles                                            0
instructions                             74,380,017,968
mem_load_uops_retired.l2_hit                424,474,692
mem_load_uops_retired.l2_miss             1,340,172,842
mem_load_uops_retired.l3_hit                940,464,091
mem_load_uops_retired.l3_miss                         0

snabb lwaftrbench -D 30 snabb-port-200k.cfg b4-icmp-request-0138.pcap lwaftr-ipv4-50k.pcap
...
ipv4_to_ipv6: 1496 bpp 1.109 MPPS, 13.276 Gbps.
ipv6_to_ipv4:   98 bpp 1.109 MPPS, 0.870 Gbps.
EVENT                                             TOTAL
cycles                                   73,840,447,011
ref_cycles                                            0
instructions                             65,387,567,486
mem_load_uops_retired.l2_hit                182,810,566
mem_load_uops_retired.l2_miss             1,276,121,242
mem_load_uops_retired.l3_hit                894,374,142
mem_load_uops_retired.l3_miss                         0
```

The l3_miss counter doesn't seem to get updated. But the L2 hit/miss ratio clearly shows the impact of the larger packet base.

mwiget, 2015-12-07:
I do get correct l3_miss counters on a E5-2690v3. Anyhow, I need to split the packet injection part away from the app under test, because reading and repeating 50k packets is eating up a major part of the L3 cache. Even with an empty binding table I do get lots of L3 cache miss when loading 50k packets:

```
snabb lwaftrbench -D 30 snabb-port-50k.cfg b4-icmp-request-0138.pcap lwaftr-ipv4-50k.pcap
Configuration file: snabb-port-50k.cfg
running for 30 seconds
offset=0 len=6 -> psmask=0xfc00
50000 bindings parsed, using 6365 kBytes
. . .
ipv4_to_ipv6: 1496 bpp 1.072 MPPS, 12.825 Gbps.
ipv6_to_ipv4:   98 bpp 1.072 MPPS, 0.840 Gbps.
EVENT                                             TOTAL
cycles                                   104,719,843,882
ref_cycles                                   64,216,854
instructions                             61,424,683,495
mem_load_uops_retired.l2_hit                248,320,590
mem_load_uops_retired.l2_miss             1,276,794,789
mem_load_uops_retired.l3_hit                778,389,472
mem_load_uops_retired.l3_miss               498,403,630
```

Will try injecting the packets from another server over loopback cables. That will eliminate the PCAP file based packet injection.

mwiget, 2015-12-07:
Good news! Built a small app pcibench, which takes 2 PCAP files and 2 PCI addresses, then sends the content in a loop over both 10GE ports. Then I modified my lwaftrbench app to accept PCI addresses instead of PCAP files. Connecting both apps via 2 loopback cables helped me really test the lwaftr encap/decap function and its pretty good ;-).

Running the actual lwaftr app with a binding table of 200k entries:

```
./runlwaftrbench.sh snabb-port-200k.cfg
model name	: Intel(R) Xeon(R) CPU E5-2690 v3 @ 2.60GHz
../../../snabb lwaftrbench -D nil snabb-port-200k.cfg 0000:05:00.1 0000:0b:00.1
Configuration file: snabb-port-200k.cfg
offset=0 len=6 -> psmask=0xfc00
200000 bindings parsed, using 14971 kBytes
```

and the pcibench with the actual B4 and AFTR packets, of which the latter contains 50k distinct packets:

```
 ./runbenchpci-50k.sh
=========================================================
model name	: Intel(R) Xeon(R) CPU E5-2690 v3 @ 2.60GHz
PCAP files v6:b4-icmp-request-0138.pcap v4:lwaftr-ipv4-50k.pcap ...
taskset -c 0 ../../../snabb pcibench -D 30 b4-icmp-request-0138.pcap 0000:05:00.0 lwaftr-ipv4-50k.pcap 0000:0b:00.0
pci1_to_pci2:   98 bpp 0.012 MPPS, 0.009 Gbps.
pci2_to_pci1: 1496 bpp 0.012 MPPS, 0.142 Gbps.
pci1_to_pci2:   98 bpp 0.138 MPPS, 0.108 Gbps.
pci2_to_pci1: 1496 bpp 0.137 MPPS, 1.634 Gbps.
pci1_to_pci2:   98 bpp 2.216 MPPS, 1.737 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
pci1_to_pci2:   98 bpp 3.188 MPPS, 2.499 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
pci1_to_pci2:   98 bpp 3.251 MPPS, 2.549 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
pci1_to_pci2:   98 bpp 3.299 MPPS, 2.587 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
pci1_to_pci2:   98 bpp 3.296 MPPS, 2.584 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
pci1_to_pci2:   98 bpp 3.305 MPPS, 2.591 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
pci1_to_pci2:   98 bpp 3.309 MPPS, 2.594 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
pci1_to_pci2:   98 bpp 3.305 MPPS, 2.591 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
pci1_to_pci2:   98 bpp 3.306 MPPS, 2.592 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
pci1_to_pci2:   98 bpp 3.288 MPPS, 2.578 Gbps.
pci2_to_pci1: 1496 bpp 0.822 MPPS, 9.842 Gbps.
^C
```

So there is still some ramp up for 2 seconds, but then it settles at line rate for the large packets and over 3 MPPS in the other. Need to look now at L3 cache usage and increase the binding table to really push it harder.

mwiget, 2015-12-09:
I got the answers I was looking for. My initial test setup was flawed by reading large PCAP files into memory. Thanks also for the resolution on how to use pmu.lua. Very cool tool!

closed 2015-12-09 by mwiget