snabbco/snabb #674: Fast traffic throw to virtio VM

pavel-odintsov, 2015-12-05:
Hello, Team!

I have yet another weird idea inspired by IncludeOS project (https://github.com/hioa-cs/IncludeOS). 

I want to deploy my traffic capture toolkit inside IncludeOS. But for this task I need some way to read traffic from hardware NIC and drop it into VM. Could I use SnabbSwitch for this task?

mwiget, 2015-12-05:
Absolutely. That's how I got introduced to Snabb recently myself. The following getting started guide explains the steps to connect a VM via virtio to Intel 10GE NIC's managed by snabb:  https://github.com/SnabbCo/snabbswitch/blob/master/src/program/snabbnfv/doc/getting-started.md

pavel-odintsov, 2015-12-05:
Brilliant work! Awesome! :) Will check it soon and contribute feedback!