snabbco/snabb #676: AVX2 disassembly not working

andywingo, 2015-12-08:
```
$ cat /tmp/foo.dasl
local ffi = require'ffi'
local dasm = require'dasm'
|.arch ARCH
|.actionlist actions
local function gen(Dst)
	|  vzeroall
end
local Dst = dasm.new(actions)
gen(Dst)
local mcode, size = Dst:build()
if debug then
   print("mcode dump:")
   dasm.dump(mcode, size)
end
$ ../lib/luajit/usr/local/bin/luajit dynasm.lua -o /tmp/foo.lua /tmp/foo.dasl
$ taskset -c 0 ./snabb snsh /tmp/foo.lua
mcode dump:
7f86a3c33000  C5                (unknown)
7f86a3c33001  FC                cld
7f86a3c33002  77                (incomplete)
```

lukego, 2015-12-08:
Interesting. This was tested to be working in luajit/luajit#95. Did we miss a commit on the merge and/or is this a failure of that specific machine code?

andywingo, 2015-12-15:
I think on master this must be working fine.  I was working on an older branch and had back-ported the dynasm update but not the luajit update.  Closing, sorry for the noise!

closed 2015-12-15 by andywingo