snabbco/snabb #689: Cable-free test lab

lukego, 2015-12-30:
Today many of our test suites depend on access to ethernet ports that are physically cabled together. I propose that we improve our tests to break this dependency. I have a specific idea for how to accomplish this. It's related to the more flexible I/O model described in #687.

The reasons I want to relax the cabling requirement are:

1. Convenience. Want to run the performance regression tests in your production environment? Just go ahead: no need to go to site and reconnect the cables for the duration of your tests.
2. Physics. Cables take up space. Cross-connecting 20 x 10G ports on the same server is already cramped: quad-port 10G cards make this worse.
3. Economics. Cables are a boring way to use hardware budget.

We would still need some cabled ports for PHY testing. It may even be worth separately testing several SFP+ modules. However, only a few driver-development servers would need to be setup like this.

Specifically I imagine keeping the ports in our original lab cabled but skipping cables for the new servers at SWITCH.

### How to do it

Suppose we want to run a realistic performance test with Snabb NFV connected to one port and packetblaster to another. There are two ways to do this:

Option A:
- Snabb NFV gets port 1.
- Packetblaster gets port 2.
- Physical direct-attach cable is connected between the ports.

Option B:
- Snabb NFV gets RX on port 1 and TX on port 2.
- Packetblaster get RX on port 2 and TX on port 1.
- Both ports are put into MAC loopback mode by their device drivers.

Here is a diagram using the notation from #684:

[image]

Both scenarios have the same effect: packets sent by each process will pass through hardware transmit/receive and then be received by the other process. The second scenario works independent of how the NIC is cabled: the ports could be connected together, or connected to a switch, or not connected at all, and the test would work the same.

Thoughts?

hb9cwp, 2016-01-07:
Ack, clever & neat idea. Also motivates your other idea in #683 about splitting NIC driver apps into separate "half-duplex" tx and rx paths.

pavel-odintsov, 2016-01-13:
Brilliant idea! But what about full ixgbe emulation from hardware side? So we could run virtual PCI device and use it with upstream ixgbe driver or with snabbswitch.