snabbco/snabb #701: startup: unhandled exception

comotion, 2016-01-10:
I'm having issues installing. plaform is linux-musl-i486 and build completes correctly, but launching `snabb` results in:

```
  snabb
BINARY    2.0M snabb
make[1]: Leaving directory `/tmp/snabbswitch/src'
/tmp/snabbswitch $ ./src/snabb
startup: unhandled exception
syscall/linux/ffi.lua:16: module 'syscall.linux.x86.ffi' not found:
        no field package.preload['syscall.linux.x86.ffi']
        no file './syscall/linux/x86/ffi.so'
        no file '/usr/local/lib/lua/5.1/syscall/linux/x86/ffi.so'
        no file '/tmp/snabbswitch/lib/luajit/usr/local/lib/lua/5.1/syscall/linux/x86/ffi.so'
        no file '/usr/local/lib/lua/5.1/loadall.so'
        no file './syscall.so'
        no file '/usr/local/lib/lua/5.1/syscall.so'
        no file '/tmp/snabbswitch/lib/luajit/usr/local/lib/lua/5.1/syscall.so'
        no file '/usr/local/lib/lua/5.1/loadall.so'
```

eugeneia, 2016-01-11:
@justincormack Does this ring any bells to you? Seems to be ljsyscall related on first look.

lukego, 2016-01-12:
The fundamental issue here is that Snabb Switch only supports Linux/x86-64. That is, i486 is not a supported platform.

Are you interested in porting to i486? This should be reasonably straightforward and I can sketch the work likely involved. More challenging if your target CPU does not support huge pages.

(Curious: what are you working on here?)

lukego, 2016-01-12:
More specifically to this PR:

The top-level `Makefile` is [cherry picking](https://github.com/SnabbCo/snabbswitch/blob/master/Makefile#L15-L21) the relevant modules from ljsyscall. It takes x64 (supported platform) but not x86 (unsupported).

lukego, 2016-01-12:
(Sorry @justincormack for doing violence to ljsyscall in our downstream packaging like this. Perhaps we should change that to avoid creating support problems for you. And we should really print a message at startup to complain about running on an unsupported platform instead of hitting a random error.)

eugeneia, 2016-01-12:
Does anyone mind if I change the title to something reflecting “we should really print a message at startup to complain about running on an unsupported platform instead of hitting a random error”?

comotion, 2016-01-14:
I wouldn't mind attempting to port to i486. I'm currently targeting i486 and arm with my packaging work. It would be a start to include all ljsyscall stuff that is relevant for that arch :+1:

lukego, 2016-01-14:
Can you say some words about your packaging work? I am confused about why you want to package up software for platforms that it does not support :).

comotion, 2016-01-16:
I am packaging for a minimal musl-based distro in its early stages. Since snabbswitch is "the busybox for networking" I thought it interesting to package snabbswitch for it.  
The lack of explicitly missing architecture support is the source of your confusion.
Let's back up a bit. 
Marking this bug as a confusing error message is a bit of a stretch, so the bug title is now misleading. Nowhere on the web pages or in the readme does it say "x86_64 only". In fact, front page refers to x86. Unless there is a lot of code to support i486 (and conceivably arm) then it feels like you're setting arbitrary arch limits for no particular reason.

lukego, 2016-01-17:
@comotion Thanks for the feedback! Indeed we have done a terrible job of setting expectations with regards to portability. I have attempted to address this in #712. Feedback welcome!

closed 2016-03-14 by eugeneia