snabbco/snabb #716: Test for build problems after git pull

lukego, 2016-01-19:
In #705 we have the situation where it is necessary to `make clean` after you `git pull` or else the build will fail. This kind of problem seems generally worth detecting and resolving so that users can safely `make` after pulling a new Snabb release. However, in this specific instance it's due to an issue in a dependency (luajit/luajit#126) and no fix is available. In this case it seems desirable to detect the problem but not treat it as a blocking issue for CI.

How to solve?

Here is one short-term idea:

- Primary CI test run is done using a pristine checkout (from `git clone` or `git clean`). This runs the full test suite and blocks the PR from being accepted if it fails.
- Second CI test is made that switches from the previous release to the new code: `checkout master; make; checkout $pr; make`. This performs only minimal testing e.g. compilation only. Failure is treated as a warning.

Or alternatively we could somehow change the workflow such that it is standard practice to do a `make clean && make -j` at the top level after pulling new changes. The full build only takes about 5 seconds on a lab server so this should be no problem provided that it is convenient e.g. triggered automatically from `src/Makefile`.

eugeneia, 2016-01-19:
The CI only has two states: success / failure. I think we only ever check the logs of failed CI results, so your suggestion would effectively hide the error. Imho this kind of problem is rare enough to warrant going the extra mile and actually fixing the problem, in this case merging the fix in LuaJIT (if I read https://github.com/LuaJIT/LuaJIT/issues/126 correctly then there is actually a fix in LuaJIT's v2.1 branch, right?).

lukego, 2016-01-19:
@eugeneia You are right! I had misunderstood Mike's response as saying that we are stuck with this problem, but really yes he did add a new fix that we could pull from the luajit v2.1 branch: https://github.com/LuaJIT/LuaJIT/commit/2f6b2967c7312d867890df158fe6e0988fda3854.

lukego, 2016-01-19:
> The CI only has two states: success / failure.

Yes - but - there can be multiple CIs and the admin can say which ones are important enough to block merges. Ref: [Github: Protected branches and required status checks](https://github.com/blog/2051-protected-branches-and-required-status-checks).

Just clarifying for future discussions, not meaning to push this separate-build-upgrade-CI idea now.

closed 2017-06-23 by eugeneia