snabbco/snabb #728: Bring Snabb to OpenStack Nomad

hannibalhuang, 2016-01-24:
Hi Snabb folks,

As a long follower of Luke and definitely a fan of SnabbSwitch concept, I want to explore opportunities of expanding Snabb into more areas.

I'm currently serving as the rapporteur of ETSI NFV IFA004 which is the standardization of NFV acceleration management, as well as a contributor for OPNFV DPACC project focus on the management side of things, that is another word, OpenStack.

In IFA004 and DPACC, for mgmt we have developed a concept of a standalone acceleration resource pool, that all NFV related accelerators should be able to form a individual resource pool as any other existing ones such as Compute, Storage, Network.

That is to say VNF should be able to hook to accelerators just like to attach a volume or bind to a port.

We've initiated a OpenStack project called [Nomad](https://github.com/openstack/nomad) to start trying out this idea. Currently I just uploaded the cookiecutter so the project is empty, we will have a team soon working on writing up modules that would provide compute-centric acceleration lifecycle management (for example Crypto accelerators).

I think Snabb should be a great candidate for network-centric accelerators to work on, and I hope Nomad could provide a mechanism to do lifecycle management and other things of snabb for VNFs, alongside working with Neutron and Nova which focus on  mgmt of snabb functionalities.

Actually i think we could design an abstraction layer proposed in [#720](https://github.com/SnabbCo/snabbswitch/issues/720) to have Nomad provide a way to connect up different networking related acceleration apps (through a DAG maybe ?)

Anyways if you guys are interested, we could start from scratch to design this in Nomad and any ideas are welcomed

lukego, 2016-01-25:
Howdy @hannibalhuang!

Cool if somebody here in the community is also involved in these projects and can cooperate with you.

Generally the Snabb community is quite remote from ETSI, OPNFV, etc. I think technical discussions will work best if we keep the vocabulary familiar e.g. in terms of Snabb Switch code or specifications from more familiar organizations e.g. IETF RFCs.

Personally I am not active in any of these projects like OpenStack, OPNFV, OpenDaylight, ETSI NFV, etc but I think it would be awesome if people who are could use Snabb Switch for interesting things.

nnikolaev-virtualopensystems, 2016-01-25:
Hello, @hannibalhuang, we at Virtual Opens Systems as a member of ETSI NFV, closely following IFA activities, as well as DPACC, and having contributed to OpenStack integration with SnabbSwitch, we can play with you the role of bridging through these technologies. We are very interested to discuss in more details a suitable joint working plan for these matters.

hannibalhuang, 2016-01-25:
Hi @lukego , thx for the feedback and 100% agree with you that with regard to Snabb related work,we should concentrate on more tech details, that why I start an issue here rather than doing it in standardization bodies : P

@nnikolaev-virtualopensystems thx for the reaching out ! We could definitely start to discuss ideas on design by proposing specs for Nomad : )