snabbco/snabb #729: Clarify copyright notice policy

wingo, 2016-01-25:
Snabb currently has norms about how to determine the copyright of a file: everything is Apache 2.0 and the copyright belongs to the authors.  However we should make this explicit in a CONTRIBUTING.md.  @lukego has expressed an aversion to copyright headers, which I understand.  We could at least encode this convention into a document.  @nnikolaev-virtualopensystems has expressed a desire for a copyright header; I understand the social dynamics about not wanting to list all authors in a header though.  Perhaps we could have a standard short header.

```lua
-- Copyright 2016 the Snabb Switch project authors.  Use of this source
-- code is governed by the Apache 2.0 license; see COPYING.
```

Or not :)  I am happy any way.

I think that there is one way in which we can improve our record-keeping here, which is to provide a way to associate an alternate copyright holder with some code.  By default we can assume that the @-domain of the email address committing the code has copyright -- snabb.co, example.com, etc, so no need for excessive labelling.  But I don't know how to indicate that e.g example@gmail.com just did work on behalf of BigCorp and so it's actually BigCorp with the copyright; ideas?

nnikolaev-virtualopensystems, 2016-01-25:
As stated here https://github.com/SnabbCo/snabbswitch/blob/master/COPYING#L178 ,the Apache 2.0 license clears this for us:

```
APPENDIX: How to apply the Apache License to your work.

   To apply the Apache License to your work, attach the following
   boilerplate notice, with the fields enclosed by brackets "[]"
   replaced with your own identifying information. (Don't include
   the brackets!)  The text should be enclosed in the appropriate
   comment syntax for the file format. We also recommend that a
   file or class name and description of purpose be included on the
   same "printed page" as the copyright notice for easier
   identification within third-party archives.

Copyright [yyyy] [name of copyright owner]

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at

    http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
See the License for the specific language governing permissions and
limitations under the License.
```

This is how one aplies the Apache lincense. I don't think we have much space left here for arguments.

wingo, 2016-01-25:
I respectfully disagree that it is necessary to attach any specific kind of header to a file in order to indicate its license.  The example header that you quote is just an example from a non-normative appendix.  We can make other decisions.

wingo, 2016-01-25:
As an external resource, I think https://softwarefreedom.org/resources/2012/ManagingCopyrightInformation.html is fairly sensible.  Snabb currently uses the centralized copyright notice; scan down that page for "centralized" for more details.  We could add file-level notices referring to a top-level notice, but this is not required.

eugeneia, 2016-01-25:
Silly question: what difference does it make who has copyright on what line of code? I only know copyright in conjunction with commercial licensing and royalties. Since Snabb Switch is Apache licensed I don't see a point. Are people inclined  commercially re-license parts of the Snabb Switch code base, but without the bug fixes and/or improvements of third-party authors?

 From Andy's link:

> But be careful when removing the notices of other developers. Since free software licenses require licensees to preserve notices, wrongfully removing one is a violation of the license from that contributor and may be copyright infringement. If it’s absolutely clear that every remnant of a developer’s contribution has been removed, then it is probably OK to remove the associated copyright notice; otherwise, it’s best to keep it around. However, a requirement to “preserve” or “reproduce” a developer’s copyright notice does not necessarily require that the notice be kept in exactly the same place it started; it’s usually acceptable to move notices from individual source files to a central attribution file, for example.

I read this as “its a liability to have copyright information, and you can incriminate yourself if you mess up”.

wingo, 2016-01-25:
> Silly question: what difference does it make who has copyright on what line of code?

Well :)  Let's say I do some work on Snabb for Microsoft.  (I haven't; it's an example.)  They agree to submit the result upstream under the apache license (yay) but they keep the copyright (well OK, doesn't matter to me, the copy in Snabb is Apache-licensed).  Now, as copyright owner, they can relicense their code.  If I were owner, they couldn't.  Being able to relicense your code can be important.  Maybe they wanted to re-contribute that code to an MIT-licensed repo of Lua code.  Dunno.  Lots of possibilities.  Some parts of Snabb aren't very Snabb-specific and can stand on their own.

Assigning copyright to a customer is a common contractual obligation, even when releasing the result as free software.  As a Snabb contractor I am under some of these obligations right now, even.

eugeneia, 2016-01-25:
Thanks for pointing this out to me, I remember now that I read about an open source project having trouble to re-license to a different open source license once because they could not reach consensus between the many different copyright holders, if I remember correctly.

Snabb Switch code should in my opinion be able to land in MIT (or GPL etc.) licensed projects.

lukego, 2016-01-25:
Thanks @wingo and @nnikolaev-virtualopensystems for raising this discussion. Thanks @wingo also for the link, I have read it and it seems very level-headed to me.

Thoughts...

#### License

I agree that it is crucial for everybody to understand that Snabb Switch is an apache-licensed open source project, and that contributed code is also apache licensed unless clearly marked otherwise. This is what makes Snabb Switch open source and gives us all the right to use each others' code. The only reason it *is* open source is that we say it is when we contribute code, and so it is important to be crystal clear and avoid misunderstandings.

Currently we spell out these license terms in the top-level file `COPYING`. It seems reasonable to also point this out in a `contributing.md` file (that Github would automatically show when creating a PR) and perhaps in source files.

I can imagine for example having a standard two-line header at the start of source files along the lines that Andy suggested:

```
-- This is the <foo> module that does <bar>.
-- Use of this source code is governed by the Apache 2.0 license; see COPYING.
```
in order to further minimize the risk of misunderstandings. I am not really convinced that longer per-file boilerplate is worth the fuss (see Andy's link).

#### Copyright

Copyright notices seem less fundamental to me. We are all using the code under the terms of the apache license and it doesn't make much practical difference who owns the copyright to each line of code. Copyright is not actually assigned by writing a note in a file anyway, and the copyright status can change after the note is written e.g. you might sell your copyright, or you are sued and your copyright is awarded to somebody else as damages, or a former employer challenges your right to the copyright in the first place, etc.

It is possible that one day copyright ownership will become relevant, e.g. because we want to unanimously relicense the code under some different open source license, but this will be a huge administrative mess independent of whether comments are available in the source files.

Having said that, if people want to make notes about copyright assignment that it would be nice to have an *optional* mechanism to do so. One way would be to have a top-level `COPYRIGHT` file where people can write notes like `Joe Blogg's contributions in 2016-2018 are copyright Big Co Ltd.` It could also be reasonable to put notes directly into source files but, like all source code comments, these will need to be maintained and so it would be nice to write them in a way that makes this easy.

One idea could be:

```
-- Portions of this file are copyright Big Co Ltd.
```

which seems easier to maintain when the file is further developed by the author and extended by others than the more traditional:

```
-- Copyright 2016 Big Co Ltd.
```

wingo, 2016-01-26:
Thanks for this discussion.  Having a spelled-out norm will help future contributions go more smoothly.

I think that for the same reason that we should avoid per-file copyright notices for individuals, we should also avoid them for corporations..  My reasoning is that per-file notices establish a kind of ownership that is negative to the project.  Someone could get resentful if BigCorp dumped some code on a project but then Jane Hacker was left to maintain it.  Also, having notices just for corporate contributions values corporations over people.  Finally, if we have a policy of avoiding copyright notices in files, we can avoid this discussion in the future always.

I do think a central COPYRIGHT file is a good idea though.  Ideally it would start by listing all contributors, as determined by git, and then add on some optional notes about how work by Jane Hacker in 2016-2018 was copyright Big Co Ltd.

nnikolaev-virtualopensystems, 2016-01-26:
Question: If I am submitting a whole module contribution, can I choose to put my Copyright line on the top of the module files? 

Also the central COPYRIGHT file should be in `src` folder, as the `lib` folder imports source from other projects which are not a subject of this COPYRIGHT.

wingo, 2016-01-26:
@nnikolaev-virtualopensystems the question of copyright notice policy is precisely what this issue/question is about :)  I have argued that any per-file notice has a negative effect on the project; if you propose to allow notices, you would need to argue for them instead of re-stating the question :)

Let's cut to the chase though: companies want to put their name out there.  I'm pretty sure that's why you want per-file notices.  But it's bad for the project, I think (reasons above) and not necessary for companies either: your company was hired to do virtio-net in Snabb not because of copyright notices, but because your company does respected work in this field, and everyone who is capable of doing a git log knows this.  You don't need to stamp an immutable line in a source file to get the word out!

nnikolaev-virtualopensystems, 2016-01-26:
@wingo - OK let's just put the rules in place and see what happens :)

lukego, 2016-01-26:
> companies want to put their name out there

Here are some suggestions for perfectly fine ways for people and companies to be visible in the community:

- Maintain valuable subsystem branches on a `MyCompany/snabbswitch` Github fork.
- Give public presentations about Snabb Switch.
- Blog about Snabb Switch.
- Represent Snabb Switch in other communities (OpenStack, OPNFV, ETSI, FOSDEM, etc).
- Sponsor work on a project that is related to Snabb Switch (LuaJIT, NixOS, etc).
- Donate important equipment to the lab.
- Buy Google Adwords for "snabb switch" searches.

These all seem like valuable activities that can be very naturally associated with the person doing them and their employer.

However, I agree that it is negative for the project to inject direct advertising into the development process e.g. by trying to maximize the number of times that other developers type your company name or look at your company logo and so on. This comes across as antisocial to me, like if my neighbour puts up a Coca Cola billboard that I have to look at from my living room window. There are better ways to promote a brand.

Just a personal opinion:

I think it is also a reality of open source projects that individuals get a large share of the credit for their work, more so than in less transparent communities where nobody knows who is "making the sausage." This is empowering to individual developers and companies need to accept that.

Over time as Snabb Switch grows the experienced developers will have better and better opportunities available to them and their employers will need to work harder and harder to keep them. For example, a senior Linux kernel hacker can pick and choose between any number of employers large or small. I see this as a good thing: a virtuous circle. I don't think it would reflect well on a company to be seen as fighting this by trying to transfer credit from its employees to its corporate entity.

lukego, 2016-01-27:
@nnikolaev: Sorry but I am going to stop typing the "-virtualopensystems" suffix on your handle since no reasonable justification for requiring me to was ever given. I think your company should embrace the image of "home of famous Snabb Switch hacker Nikolay Nikolaev" as their brand instead of these other funny things.

lukego, 2016-01-28:
@wingo Thanks again for starting this discussion.

The next action I see for closing this issue is for somebody to PR a [CONTRIBUTING.md](https://github.com/blog/1184-contributing-guidelines) file that captures the outcome of this discussion into actionable advice for people sending contributions.

To me it seems like we have rough consensus that adding this line to source files would be valuable:

```
-- Use of this source code is governed by the Apache 2.0 license; see COPYING.
```

and that we could create a `src/COPYRIGHT` file to record notes about copyright assignment. The exact purpose and contents of this file is not entirely clear and would need to be spelled out in CONTRIBUTING.md. I imagine that the COPYRIGHT file would be used to write "notes to our future selves" in case one day we need to establish the current copyright status of the code e.g. for relicensing, etc. In this case it could be useful to get feedback on the contents from people who have done such things. (I don't think the purpose of COPYRIGHT is to be a proxy for measuring contributions to Snabb Switch and if that is the only reason people want it then perhaps we don't need it at all.)

Generally discussions about how best to promote Snabb Switch products and services is a very welcome topic. Everybody who would like to discuss this and help establish community norms is very welcome to create a new Issue dedicated to that subject.

justincormack, 2016-01-31:
As I now work for a company, in which I have had to raise the issue of
providing attributions for code we use, having a central source file for
copyrights is hugely more convenient than writing code to scrape every file.

On the other side, if I sign off my commits with my work email that is all
the attribution they want.

closed 2016-05-04 by eugeneia