snabbco/snabb #738: "Snabb"

lukego, 2016-02-04:
I propose that we shorten the name of our project to simply *Snabb*.

The longer name *Snabb Switch* comes from the early days of the project when the NFV application (a "hypervisor switch" or "vswitch") was the main focus. However, it has become something of a misnomer now that people are building many new and wonderful other applications. (It's also kind of long to type and abbreviations like *SnS* have not caught on.)

I think it's okay for the name to be slightly ambiguous i.e. for *Snabb* to refer to the core software, the whole software ecosystem including applications, the community, etc. This seems to work fine for "Linux."

What do we think?

#### How

This change would seem to involve a little bit of work:

- Update the source and documentation to `s/Snabb Switch/Snabb/`.
- Rename the Github repo from `snabbswitch` to `snabb`.
- Could alias `snabb-switch` to Snabb NFV if that reduces confusion somehow (or might increase it..)

(The renaming of the default binary from `snabbswitch` to `snabb` was already done long ago.)

#### Related

Couple of related issues:

- Define what the *Snabb* brand stands for i.e. what can users/developers safely assume about an application called *Snabb Foo* (where Foo = packetblaster, nfv, etc). Example: that it is apache licensed, accepts contributions, is engineered to be small/compact, and that the code syncs with the master branch.
- Agree on suitable/unsuitable uses of the project name. (For example I should rebrand my own professional activities which for historical reasons are also under the name "Snabb".)

wingo, 2016-02-04:
Nice!  Before, I was hesitant to use the name "Snabb" because of potential conflicts with SnabbCo; having some clarity will help the project I think.

Regarding *Snabb Foo* and branding, the points you mention make sense to me.  I might relax them to be a bit more aspirational -- e.g. someone could start their project called Snabb Foo with an aspiration to be small/compact, sync with master, etc, but if it's early days you never know how the thing is going to go.  Apache licensing should be a must, though.

Regarding other uses of the word Snabb -- I don't have experience in this domain, but https://wiki.gnome.org/FoundationBoard/Resources/LicensingGuidelines is a not-long document that can hint to what problems might arise.  Have you registered the trademark?  I guess it's a good idea, though who knows :)

Specifically: we should encourage the use of Snabb in open source product names which meet the *Snabb Foo* criteria above.  We should prohibit Snabb in company names  (besides yours, if you decide to keep it) or in proprietary product or service names.  We should prohibit "snabb" in domain names, besides snabb.co, perhaps unless they are subsidiary to something else (e.g. snabb.microsoft.com).   We should adapt this rule: "Do not use GNOME Trademarks more prominently than your company, product or service name.".  Probably we should prohibit creation of a Snabb logo, without authorization from the copyright holder.  Maybe this is too much, feel free to take/leave/modify/etc :)

lukego, 2016-02-04:
I would be interested to know what people consider good/bad role models for open source project brands. Particularly ones that are administrated in a simple way.

"Linux", "GNU", and "Apache" are all good strong open source brands for me.

"OpenStack" not so much. The upstream community has used the brand as a stick to wave at their contributors to make them run hither or thither for fear of being de-branded. This has created a lot of animosity, politicing, and gaming of the system.

I do have a Snabb trademark here in Switzerland. I should subscribe to a trademark monitoring service to be notified if somebody would apply for a confusing trademark elsewhere in the world and perhaps also register in a couple of important places. Practical tips on how to do this in a quick and simple way would be welcome if somebody knows a good service.

lukego, 2016-02-06:
> We should prohibit Snabb in company names (besides yours, if you decide to keep it)

Just to be clear: My company (Snabb GmbH in Switzerland) is serving as a bootstrapping vehicle for the Snabb project. In the beginning it has been practical to have a well-known entity that can sign contracts to provide services, sponsor relevant work on Snabb and related projects, buy and operate lab equipment, etc. Over time these activities are spreading within the community: other people and companies are selling reputable Snabb services, SWITCH is hosting and maintaining the new lab, Intel and Mellanox and other sponsors are providing equipment, and so on. So the community dependence on that one company is decreasing all the time and eventually I imagine that it will fade into the background and discretely stop trading under the name "Snabb".

This is fine with me. I have no desire to exercise unfair/asymmetric control over the "Snabb" brand: it is something for the community to share. Snabb GmbH is safeguarding it while we work out how we want to do things in the long term.

Further notes:

- Snabb GmbH has no other shareholders or other obligations that I could imagine creating a conflict of interest during the bootstrapping era. Nor does it own any intellectual property besides the trademark (not even the copyright to the code that I write).
- In the *very* beginning of the project there were other paths available too e.g. to follow Hwaci (SQLite), Apache, Mozilla, BerkeleyDB, MySQL, etc. I consulted with some of the people who created those projects to get help in working out a good path for Snabb. I am very satisfied with the track we are on and the progress we are making.

eugeneia, 2016-02-16:
>  I was hesitant to use the name "Snabb" because of potential conflicts with SnabbCo

Me too. Wouldn't be opposed to drop the “Switch”

kbara, 2016-02-16:
I also like this set of ideas.

lukego, 2016-02-16:
Great! So we need volunteers to do these things:

1. [ ] PR to update "Snabb Switch" to "Snabb" in the source tree.
2. [ ] Make a plan for renaming `snabbswitch` to `snabb` in whatever is the smoothest way.
3. [ ] Propose simple and unambiguous rules for using the Snabb brand (name/logo).
4. [ ] Create a project homepage e.g. landing page that explains what "Snabb" means and links to Snabb-branded projects and professional services.

I also need to come up with a simple and unconfusing plan for rebranding my own professional activities.

lukego, 2016-03-13:
Just moving forward here a little bit...

Github seems to handle repository renames very smoothly. Existing checkouts will work, the old name and URL can still be used, website will redirect quietly. So it seems we can simply rename the repository from `snabbswitch` to `snabb` once we merge a PR to update the content.

In the background I am setting up a separate brand for my own professional activities and I plan to complete that this year. This will free up the name *Snabb* and the domain `snabb.co` to refer specifically to the open source project. Then we can all use the *Snabb* brand on the same fair and equitable terms that we will thrash out here on Github. I imagine that I will continue to hold and administrate the trademark for the foreseeable future. Happy to discuss any concerns, I have no sinister plans.

How should we define the rules for using the Snabb name and logo? Generally my feeling is that a *Snabb* product or service is one closely connected with the upstream community here on Github. This is a grassroots project so it should be easy for well-intentioned people to create projects, sell services, organize events, start user groups, sell t-shirts, etc. Specifically though I think we need to be prepared to evolve the details: we could start with a simple statement like the [PostgreSQL one](https://wiki.postgresql.org/wiki/Trademark_Policy) and if needed then evolve toward something more elaborate like the [Apache one](http://www.apache.org/foundation/marks/) over the coming years/decades. I would suggest we maintain a `TRADEMARKS.md` file where we keep track of all the uses that we have defined as okay or not-okay i.e. maintain the trademark policy the same way we maintain everything else.

plajjan, 2016-04-05:
Heh, biggest downside to this is probably that "Snabb" is in no way unique. "SnabbSwitch" is kind of unique combination whereas "snabb" is already used lots. It's still easy to find snabb.co (Google ranks it high) but I think it will be difficult to find other related things, like email threads, discussions, comments or similar.

eugeneia, 2016-05-13:
Done in v2016.04.

closed 2016-05-13 by eugeneia