snabbco/snabb #740: Document a coding style for use of object-orientation in Snabb

wingo, 2016-02-04:
Snabb is a growing project and there is room for many things in it :)  Still, if there are solutions that we agree upon as being particularly snabby, we should document them.

One of these areas is object orientation.  My proposal is to write up a document with the conclusions of this issue and PR it.  However at present it's just a discussion.  Here's a draft:

----

An object in Snabb is data with associated methods.  If you implement a data type in Snabb, we encourage you to set a common Lua metatable on all instances of that type, and to add methods to that metatable exposing the object's functionality.

It's also possible to expose object functionality via functions exported from a module which take an object as an explicit argument.  Doing this or not is up to you.

If a module exports functionality that does not logically operate on an object, such functionality *should* however be exported as a plain function.  Notably, constructors fall into this category.  So for example if you implement a trie:

```lua
local trie = require('lib.trie')

-- The constructor doesn't operate on an object, so it's just a function exported by the module.
local t = trie.new()

-- On the other hand, an "add" method on the trie does operate on an object, so it's a method.
t:add('foo')

-- Maybe there is a trie.add method; if you feel like it.
trie.add(t, 'bar')
```

Thoughts @alexandergall ?

alexandergall, 2016-02-05:
I agree (and pledge guilty for having abused "class methods" for non-object related functions). I see your point concerning the constructor, too. That could be easily implemented with `lib.lua.class` as well (just a pain to change all the modules that currently use the method style, but that's on me).

wingo, 2016-02-05:
Yeah probably we should include some implementation notes as well, for example how to provide this api using `lib.lua.class` and so on.  No need for urgent changes, but agreement is a good thing to look for :)

lukego, 2016-02-09:
Good discussion!

I would like to flag this whole topic as controversial. It is recommending a coding style that is unlike the core modules e.g. packet and link. This needs to be addressed somehow.

The broader discussion could take place under the umbrella of #734 including e.g. should we update the `packet` and `link` modules to follow this example too?

I will advocate for Occam's Razor i.e. the technically simplest and most minimal solution that actually covers our requirements. If that requires procedures, classes, subclasses, class methods, multimethods, predict disaptch, etc, then so be it. But I would presume everything to be unnecessary complexity until proven otherwise.

The risk I see is that we will introduce concepts that make one part of the community comfortable at the expense of another part of the community, and if we do this inconsistently over time then everybody will end up uncomfortable. To understand the whole Snabb source tree should you already be familiar with Java, Python, Smalltalk, CLOS, etc and recognize ideas that are borrowed from these languages? That is the kind of transitive complexity that I want to avoid.

I would also like to be explicit about where the rules apply and where they don't. For example documented APIs may be expected to strictly follow the recommendations, and core modules too, but not internal code within apps or programs where we defer to the author's creative impulses instead.

I observe that the `lib.lua.class` code has been controversial and I think it's because the first version PR'd already used classes and subclasses and people didn't have the chance to "see the working out" and understand what value those concepts are bringing to the table.

alexandergall, 2016-02-10:
Maybe it's helpful to the discussion if I explain my motivation to go that particular path in the `lib.protocol` code. It was obvious to me that there will be lots of opportunities for code-reuse in the handling of protocol headers and the class model with actual inheritance seemed like a good fit. I'd be curious how people who don't like this approach would have written that code. I don't have any strong feelings towards any style, but I do care a lot about re-using code.

AFAICS, there are currently not many other parts of the Snabb codebase that would benefit from a non-trivial class hierarchy, but I do observe that at least the `intel10g` app uses inheritance internally and I actually find that code hard to read because the hierarchy is not obvious. I think that a more modular approach for writing drivers would be beneficial in general, e.g. separating generic from hardware-specific code. That might give us a better idea about what coding style to prefer.

lukego, 2016-02-10:
> I'd be curious how people who don't like this approach would have written that code.

I observe that when people have dug into that code they tend to develop a better appreciation for it. It is very carefully and thoughtfully written. I can believe that it would be awkward to factor that code in a more "cave man" style and that the object-orientation is really solving a problem.

I also like reuse. My fear with `class.lua` is that we will casually create object-oriented class hierarchies as one does in other languages like Java even when they are not required. That would not seem idiomatic for Snabb ("Snabby") to me.

>  at least the intel10g app uses inheritance internally and I actually find that code hard to read because the hierarchy is not obvious

I struggle with that code and am sometimes surprised about which functions are really being called in different configurations and which registers are really available. This leads to awkwardness in changes like 9c8b1025f049d3803b7aa03dc4ed2b754a4459d5.

With the benefit of hindsight I feel that inheritance complicates the code in `intel10g` for me and that is why in the I350 driver I factored everything into one app and used `if` on configuration parameters to decide e.g. whether to initialize the device or not. That driver has fewer features and so I may be cheating a bit, I am not sure yet. I really enjoy the creative process of looking for ways to eliminate clever ideas and make things more low-brow/cave-man/down-to-earth. This is not always possible though... "as simple as possible, but not simpler" as the saying goes.

In the meta-sense though I am also really glad that we have not prematurely standardized all of this stuff and so we have an interesting assortment of programming techniques in our repository that we can talk about. I think we need to preserve the feeling that the person actually solving the problem has a lot of latitude to make design decisions.

wingo, 2016-03-07:
> I will advocate for Occam's Razor i.e. the technically simplest and most minimal solution that actually covers our requirements. If that requires procedures, classes, subclasses, class methods, multimethods, predicate dispatch, etc, then so be it. But I would presume everything to be unnecessary complexity until proven otherwise.

I think this is a straw man argument :)  This PR advocates attaching FFI metatables to data types, that's all.  It's done in in such a way that you can use functions or methods, and the resulting procedural interface is harmonious with those in `core.packet` etc AFAICT.

However I would like to go further, and to advocate globally for preferring method syntax where it's obvious what the data types are.  For example if you know P is a packet, then use `p:free()` rather than `packet.free(p)`.  You might choose to use `packet.free(p)` if you feel that it's not obvious that P is a packet and you want to help the reader know where the `free` function is coming from.

The reason for this is to reduce interface ceremony.  As a use example:

```lua
local histogram = require('core.histogram')
...
local h = histogram.new(1e-6, 1)
h:add(42)
for count, lo, hi in h:iterate() do
  ...
end
```

It's clear that h is a histogram, so the method syntax reduces the degree to which the user is battered with the word "histogram".  Compare to:

```lua
local histogram = require('core.histogram')
...
local h = histogram.new(1e-6, 1)
histogram.add(h, 42)
for count, lo, hi in histogram.iterate(h) do
  ...
end
```

or worse:
```lua
local histogram = require('core.histogram')
-- someone will do this eventually
local histogram_add = histogram.add
...
histogram_add(h, 42)
```

or worse:
```lua
local histogram = require('core.histogram')
-- someone will do this too
local add = histogram.add
...
add(h, 42)
```

And of course if there's no local require, like #734 proposes, then there's no ceremony at all:
```lua
local h = snabb.histogram.new(1e-6, 1)
h:add(42)
for count, lo, hi in h:iterate() do
  ...
end
```

And it's all super-low-tech.  I find this very snabby but others may disagree :)

lukego, 2016-03-07:
> However I would like to go further, and to advocate globally for preferring method syntax where it's obvious what the data types are.

Good!

Let's decide on a canonical style that works for everybody. This will require some due diligence on JIT efficiency to make people comfortable but hopefully we understand things well enough to do that now.