snabbco/snabb #741: Troubleshooting/performance should be better documented

kbara, 2016-02-04:
I read Luke's comment on a single API manual ( https://github.com/SnabbCo/snabbswitch/issues/733#issuecomment-179634478 ) and think he's right. I also think a single manual would be good for cross-Snabb-application troubleshooting/performance tweaking. It should cover things like not receiving packets due to IOMMU settings, numactl+taskset, etc.

eugeneia, 2016-03-14:
#824 might be related.

wingo, 2016-03-15:
cc @kbara for her recent document where she pulled in some machine-level performance tuning suggestions from @mwiget

wingo, 2016-03-15:
Haha, I neglected to see that the ticket was made by @kbara :)  Ignore me please :)

eugeneia, 2016-03-15:
#827

eugeneia, 2016-05-13:
Closing because #827 was merged, please reopen on demand.

closed 2016-05-13 by eugeneia