snabbco/snabb #745: Enabled mention-bot

lukego, 2016-02-06:
I enabled [mention-bot](https://github.com/facebook/mention-bot) as an experiment. The idea is that when a Pull Request is opened this bot will automatically look at the diff and post a comment that @mentions people who are likely to know about the code being changed.

This is meant as a simple "heads up" notification to make relevant code changes more visible to people. There is no obligation to respond when mention-bot tags you.

If you have thoughts on the matter then please leave a comment below :).

lukego, 2016-02-10:
@eugeneia mentioned on #746 that mention-bot is annoying him enough that he blocks it. If this is a common sentiment then we can do away with it. Please vent here :).

eugeneia, 2016-02-10:
My observation: in #747 mention-bot mentioned, dpino, alex and me. Imho that's a 2/3 failure rate. I think dpino wrote the code in question, so one might argue he would like to know about this, and I would probably have mentioned him anyways. Alex and me on the other hand... not so much. E.g. I think mention-bot is overzealous, but maybe I am just mad because the machines are taking away my job. ;-)

wingo, 2016-02-10:
To me it's not a failure to mention someone who's not a domain expert in the code at hand, dunno.  I think mentioning you @eugeneia is a success in some way, as you actually maintain an integration branch, and it's conceivable that alex might also have a good comment as an experience snabber.  Your mileage evidently varies :)

wingo, 2016-03-04:
OK I now side with @eugeneia :)  At the very least if the original issue or PR @-mentions a person, having mention-bot chime in is not necessary and a bit annoying :)

lukego, 2016-03-06:
@wingo @eugeneia Yes, now I agree too, mentionbot is mostly a distraction. Disabled now.

Thanks all for bearing with that experiment.

closed 2016-03-06 by lukego