snabbco/snabb #749: Support stripping VLAN tags on non-virtualized 82599 NICs

wingo, 2016-02-10:
When configured with `vmdq=true` and `vlan=1234`, the `intel10g` driver will strip VLANs.  It enables this option by setting the VME bit in RXDCTL:

[image]

https://github.com/SnabbCo/snabbswitch/blob/master/src/apps/intel/intel10g.lua#L913

However in non-virtualized mode, the VME bit is never set AFAIU and so even if you specify `vlan=1234`, you get a vlan filter but no stripping.  Proposal: we add a `strip_vlan_tag` conf option to the intel10g driver.  Sadly it would have to default to `false` on `vmdq=false` and `true` on `vmdq=true`.  WDYT @lukego?

wingo, 2016-02-10:
Gaaaah, VLAN *filtering* only works if VMDq is enabled also.  The documentation does not mention this!

wingo, 2016-02-10:
AFAIU, without VMDq, an 82599 app will accept any incoming MAC address, and will not stamp outgoing packets with the MAC, and won't insert a VLAN tag on outgoing packets either.

wingo, 2016-02-11:
Apparently app:report() on intel apps also needs vmdq enabled to work: https://github.com/Igalia/snabbswitch/pull/260

lukego, 2016-02-11:
> Apparently app:report() on intel apps also needs vmdq enabled to work: Igalia#260

Ouch. This is the kind of issue I was alluding to in https://github.com/SnabbCo/snabbswitch/issues/740#issuecomment-182327203. I think the PF/SF/VF factoring of the code is error prone (inheriting methods between objects that have quite different state e.g. available register definitions) and that we should try refactoring in terms of INIT/TX/RX like in the newer I350 driver (see also #687 "I/O 2.0"). See commentary in `intel1g.lua` on master for more on that idea.

The next step that I have planned in that direction is to finish my Intel FM10K (100G) driver make a nice integration between that an Snabb NFV.