snabbco/snabb #759: Update lab for glibc security vulnerability

wingo, 2016-02-17:
glibc had an incredible security vulnerability disclosed yesterday: https://googleonlinesecurity.blogspot.fr/2016/02/cve-2015-7547-glibc-getaddrinfo-stack.html

We should update our lab servers.  For Ubuntu I guess this will happen mostly automatically.  For NixOS I don't know what the story is.  @domenkozar do you know what we need to do?  I assume a full rebuild of NixOS will take forever, and then there's the issue of people's profiles that all need to be updated too.

eugeneia, 2016-02-17:
As far as I know our NixOS fleet can be trivially redeployed when a patch is available (might even happen automatically scheduled, bit sure).

domenkozar, 2016-02-17:
I'll update the lab now.

domenkozar, 2016-02-17:
I've updated the lab and supporting server. Note that you'll have to run `nix-env -u` to update your imperatively installed packages.

closed 2016-03-01 by wingo