snabbco/snabb #789: core.packet: shiftleft is unsafe

eugeneia, 2016-02-26:
`packet.shiftleft` as well as `packet.shiftright` (from #781) can underflow and overflow packets.

wingo, 2016-02-26:
I don't know that this is a bug, fwiw.  Lots of things in Snabb are unsafe in general although they may be safe in particular.  As a user of shiftleft I certainly wouldn't want to propose adding an assertion to it without analyzing perf impact.

kbara, 2016-02-27:
I think 'safe' variants should be used by default, with unsafe variants with 'unsafe' in the name being available when raw speed is necessary. Call it my Haskell bias. I'm in favor of speed at the expensive of safety only when it's _really_ necessary, never as a default.

lukego, 2016-02-29:
The real problem in my eyes is lack of an immediate way to measure the impact of the assertions on real applications via end-to-end performance measurements. Removing assertions is only an optimization if it makes something important run faster: otherwise it's a bug.

wingo, 2016-02-29:
@lukego No argument there; this bug is to *add* assertions, which is fine but requires an analysis of the perf impact IMO.  Just trying to avoid the situation where we could merge in the next release and have to spelunk a perf regression.

eugeneia, 2016-02-29:
I would personally like the core API and libraries to be memory safe. I see this as one of the major benefits of using Lua instead of C. If that is not possible due to performance requirements then the unsafe API should be explicitly documented as such. So at the very least this is a documentation bug since the documentation of `shiftleft` does not document the dangers:

> — Function **packet.shiftleft** *packet*, *length*
> 
> Truncates *packet* by *length* bytes from the front.

I think this would be a good general rule for the documentation: if its not safe the extent of unsafeness should be documented. A function whose documentation that does not explicitly state unsafeness should be safe.

lukego, 2016-02-29:
@eugeneia On the one hand that sounds like an entirely reasonable position and I tend to agree. On the other hand contract with the "No hand holding!" statement in the [LuaJIT FFI documentation](http://luajit.org/ext_ffi_semantics.html) that sets a very different tone and is describing the level of abstraction that we are operating at.

kbara, 2016-02-29:
I don't see this as 'hand holding' so much as having non-stupid defaults that we can override when strictly necessary.

eugeneia, 2016-02-29:
@lukego I am aware that LuaJIT's FFI API is unsafe but that does not mean our core API has to be. E.g. if I am writing an IPsec/ESP app I want to be able to use (a subset of) an API that is memory safe. There should be a simple way to write safe-by-default Snabb code.

lukego, 2016-02-29:
In a perfect world I would like to only support safe variants and to have sufficient performance tests upstream to demonstrate that the advantage of removing assertions is negligible.

The historical picture is that we have started with making things safe and then people have removed the assertions during optimization campaigns.

I am not sure how I feel about supporting both safe and unsafe variants. People will all end up using the unsafe ones in practice if they believe that they are faster. Could make things worse if one person writes the code using safe routines and then somebody else comes along and switches them when they are trying to shave cycles.

I do think that clearly documenting the preconditions that these functions depend on is an unambiguously excellent idea.

eugeneia, 2016-02-29:
> I do think that clearly documenting the preconditions that these functions depend on is an unambiguously excellent idea.

I agree that this is the easiest solution right now, and in some way also the most correct. I will treat this as a documentation bug, and take lead on this.

lukego, 2016-02-29:
Note: Checking overflow/underflow like this should be effectively free for the CPU. If there is any performance overhead it is surely due to the way the test is implemented e.g. implications for the JIT when the tests are done in the middle of long traces.

This suggests to me that *if* the assertions are too expensive then we should find a new implementation strategy rather than removing them. For example between breaths we could have a small loop that scans through packets and ensures that all freed packets had a valid length and/or that known-good "canary" values are found to be valid leading/trailing the packet struct. This would not detect errors quite so immediately but it may be a reasonable compromise.

lukego, 2016-03-01:
Just wanted to summarize a bit more reflections since this will be a recurring theme in the future.

On the one hand:

- I am firmly in the camp of wanting safety features built-in and always available. I am reminded of the computer science classic [The Emperor's Old Clothes](http://worrydream.com/refs/Hoare%20-%20The%20Emperors%20Old%20Clothes.pdf).
- I see much more benefit in chasing 1000% speedups like multicore vs 1% speedups by removing valuable safety features like bounds-checks.

On the other hand:

- Wearing my upstream hat I don't want to stand between people and their application-level performance targets. If assertions in the `packet` module prevent an application from reaching a performance target then I am inclined to accept a patch that removes them.
- I am not in favor of safe/unsafe variants of functions and expecting application developers to toggle back-and-forth based on benchmarks. This is a complicated interface and in practice I would expect to see Pull Requests asking for every safe call to be replaced with an unsafe one as byproduct of people's product optimization efforts.

Overall I think the onus is on people who want safety, including myself, to find implementation strategies that have universally acceptable costs. If this requires moving checks out of the inner loops of apps, or adopting compiler intrinsics, or using hardware breakpoints, etc, then so be it.

I do see the statistically-significant benchmarking as one defense against overdoing the dangerous optimizations. Then we can at least avoid removing safety features when they don't *actually* cause problems but people are tempted to remove them anyway due to gut feeling or on the general principle that they care more about performance than safety.

wingo, 2016-03-01:
On my side I apologize for FUDding the issue -- I know it can be irritating when you just try to make things better and people come with all kinds of questions and no data.

I would like to avoid a pattern of performance degradation over time though and I can see that happening if we start trying to "add" some degree of assertions/safety to unsafe interfaces *with which people have already tested their performance targets* and for which there are no automated perf tests.  I also don't think our perf infrastructure is precise enough to track 0.5% performance losses -- correct me if I'm wrong.  In the web browsers world we do sweat 0.5% perf losses because it's easy for them to add up to a big loss over time.

How about, if someone wants to add safety to any packet interfaces, I agree to test them for performance on my workload and give you good feedback :)  See you on the PRs, then?

lukego, 2016-03-01:
@wingo I believe the basic way forward is for upstream to do performance tests that are much more sophisticated than people would bother to do themselves. This would incentivise people to upstream their tests (or reasonable proxies thereof) which would be invaluable for performance optimization efforts.

I believe that I do now have the tools to run an A/B test of a change for any reproducible benchmark to test a hypothesis such as "Performance impact is less than 0.5% with 99% confidence." I am happy - even eager - to do this with tests that people contribute to the project.

I don't want to get into the habit of waving sticks at contributors. I do however feel that expecting upstream to defend invariants that are ill-defined and non-reproducible is heaping technical debt onto the project and is not the path to nirvana.

lukego, 2016-03-01:
@wingo also note: Safety on those interfaces was there from the beginning but has been removed. The benchmarks that lead to this removal are not immediately reproducible and that is an example of technical debt i.e. we can't check whether there was a good reason to remove them in the first place nor whether that reason still holds now e.g. after LuaJIT upgrades and so on.

I was conscious of this when I merged the PRs to remove the checks and I would be much happier if there were (say) a completely reproducible Hydra build showing the test results that the decision was informed by.

lukego, 2016-03-01:
> Performance impact is less than 0.5% with 99% confidence

Could also tests slightly different hypotheses like "99% confidence that there is no performance impact." I am sure it will be an interesting process to define our tests in meaningful terms that lead the code in the right direction (and are computationally feasible).

eugeneia, 2016-03-14:
@wingo @lukego While thinking about how to document `shiftleft` and `shiftright` I noticed its not as easy as we might have thought:

- `shiftleft` - this one is indeed “easy”, 0 <= `bytes` <= `packet.length(p)`.
- `shiftright` - this one on the other hand: 0 <= `bytes` and `bytes+packet.length(p)` <= `max_payload`. I do not want to refer to `max_payload` in the documentation as it is not really a public concept...

I would definitely welcome some confirmation on these assertions hurting performance, as the involved API cruft is not as miniscule as I would like it to be. @wingo Could you help me out in running the relevant benchmark(s)?

drbo, 2016-04-08:
@eugeneia I think the fact that the precondition for `shiftright` being unclear makes it _especially_ important to check.

I think that **security** is the most important reason to ensure safety on public APIs.  Buffer overflows are exploitable, and Snabb Switch usually runs with superuser privileges and direct access to the hardware.  Furthermore, Snabb Switch runs on machines that could do a lot of damage should they become compromised, if nothing else due to the volume of traffic that passes through them.  A buffer overflow in Snabb Switch is just as exploitable &ndash; and with the same consequences to the vulnerable machines &ndash; as a buffer overflow in the Linux kernel's network code.

domenkozar, 2016-07-12:
During [the inspection fo benchmarking failures](https://github.com/snabbco/snabb/issues/960#issuecomment-232027967) I've also hit following error:

```
core/packet.lua:84: packet payload overflow
stack traceback:
	core/main.lua:128: in function <core/main.lua:126>
	[C]: in function 'assert'
	core/packet.lua:84: in function 'append'
	lib/virtio/net_device.lua:148: in function 'buffer_add'
	lib/virtio/virtq_device.lua:77: in function 'get_buffers'
	lib/virtio/net_device.lua:128: in function 'receive_packets_from_vm'
	lib/virtio/net_device.lua:114: in function 'poll_vring_receive'
	apps/vhost/vhost_user.lua:78: in function 'method'
	core/app.lua:83: in function 'with_restart'
	core/app.lua:301: in function 'thunk'
	core/histogram.lua:135: in function 'breathe'
	...
	program/snabbnfv/traffic/traffic.lua:63: in function 'run'
	program/snabbnfv/snabbnfv.lua:17: in function 'run'
	core/main.lua:47: in function <core/main.lua:34>
	[C]: in function 'xpcall'
	core/main.lua:157: in main chunk
	[C]: at 0x00451ac0
	[C]: in function 'pcall'
	core/startup.lua:3: in main chunk
	[C]: in function 'require'
	[string "require "core.startup""]:1: in main chunk
```

See https://hydra.snabb.co/build/128546 as an example failure.

eugeneia, 2016-07-21:
@domenkozar its a separate issue (the error comes from `packet.append`’s bounds check), using `shiftleft` wouldn't have left us with this detailed error message. ;-)

eugeneia, 2016-08-23:
Closing because #913 landed.

closed 2016-08-23 by eugeneia