snabbco/snabb #792: Making Snabb Switch more friendly to new users

eugeneia, 2016-02-29:
It is not well defined which privileges are required to run which parts of Snabb Switch, and when Snabb Switch crashes due to insufficient privileges the error messages are usually unhelpful to new users. Examples of this include #791, in which we decided to recommend generally running Snabb Switch as root because it was the simplest option. (Related: some of our test scripts (`apps/tap/selftest.sh`) use `sudo`.)

Currently, even trivial failure cases print stack traces which might be overwhelming to non-developers. See #679 for a related discussion.

Cc @lukego

lukego, 2016-03-01:
I see related issues here:

1. Identifying when - if ever - it is okay to run as non-root.
2. Printing a clear error message when running with insufficient privileges.

I suspect that (1) is a really narrow use case at the moment e.g. running `--help` if even that.

For (2) we already have a solution which is to call `core.lib.root_check()` before executing a privileged operation.

Strawman first step: Simply call `check_root()` on startup and never run as a non-superuser? (Then relax once we identify well-defined functional subsets that will work reliably?)

kbara, 2016-03-01:
I don't see why that would be a strawman. It would be a huge improvement in many cases.

eugeneia, 2016-03-01:
> Simply call `check_root()` on startup and never run as a non-superuser? 

Did we not even do that at some point? I agree with @kbara that this would be a good thing for now, but the downside is that we hide the real bugs with this. We probably should work on defining granular privilege requirements and treat inconsistencies as bugs. If we just refuse to start up these “bugs” are invisible.

I think in reality the cases where we need privileges are actually limited (and snabb can do many useful things that do not require them), I see:

1. Access to `/var/run/snabb`: we could test if we can write to `/var/run/snabb` or `/var/run` and if not, fall back to a different root (`~/.snabb/run` for instance)
2. Access to DMA: we could have the `check_root()` in initialization functions of modules that actually use DMA and report better messages, e.g. “XYZ needs privileges”.
3. Allocate huge pages: I believe this can be done unprivileged, depending on the OS configuration. We could detect if we are unable to and print a helpful message.

Anything I missed?

lukego, 2016-03-01:
@eugeneia In my mind we have never supported non-root operation and so depending on root access is not a bug (except insofar as it is not checked). IIRC there was some push-back when I tried to make root a hard requirement and so I added the `check_root()` instead which has now become a maintenance burden.

lukego, 2016-03-01:
... could also be that we really do depend on certain commands like `neutron2snabb` to be available without root access. This is at best inadequately documented though. I am in favor of making root a hard-requirement and then adding exceptions when they are well-defined and well-motivated on an application-by-application basis.

Could be a good change to introduce early in the next release cycle so that people have a chance to see whether requiring root on `next` will cause them issues and have a chance to send in patches before that is released.

kbara, 2016-03-02:
In practice, Snabb needs to be run as root to do pretty much anything useful at the moment. A colleague of mine wrote a packet generator on Snabb; even that needed to be root, for incidental reasons.
Philosophically, I prefer software to use minimal permissions. Pragmatically, Snabb needs significant hacking to be usable as non-root, and bailing out fast if it's not would make it more usable.

I'm happy to take patches to make permissions more granular on kbara-next.

fsaintjacques, 2016-03-25:
I just tried to run the tests and it failed because `/var/run/snabb` didn't exists. Since I don't have root on the host (I am aware this is not ideal to test such package), I had to manually edit `core/shm.lua` for something else. Might I suggest you add a runtime flag to snabb command-line, e.g. `--run-path` defaulting to the current value. This would allow to run tests in a localized fashion.

drbo, 2016-04-08:
One feature that would be very helpful is to add assertions that ensure that the Snabb API is used correctly.  An assertion failure is much more helpful than a random segfault (or worse) due to memory corruption.