snabbco/snabb #808: PMU musings

lukego, 2016-03-08:
The PMU is a powerful piece of hardware. We are only scratching the surface of how it can be used. This issue is meant to remind us of this fact and perhaps inspire the next step in our usage.

The idea I came up with is simple: to run the `snabbmark basic1` benchmark many times, each time with a different set of PMU events, and each time with `/packet` counts reported.

Script:

```lua
-- Specify groups of PMU events by name pattern.
-- The CPU can track up to 8 events at a time.
-- See lib/pmu_cpu.lua for available counters.
local event_groups = {
   {"^l2_rqsts"},
   {"^l2_trans"},
   {"^dtlb_store"},
   {"^dtlb_load"},
   {"^offcore_requests"},
   {"^itlb_misses"},
   {"^cycle_activity"},
   {"^uops_executed_port"},
   {"^mem_trans_retired[.]load_latency_gt_"},
   {"^mem_uops_retired"},
   {"^mem_load_uops_retired"}
}

local snabbmark = require("program.snabbmark.snabbmark")
local pmu = require("lib.pmu")

local npackets = 10e6

-- Run basic1 benchmark separately for each group of events.
for _, events in ipairs(event_groups) do
   engine.configure(config.new())
   pmu.profile(
      function () snabbmark.basic1(npackets) end,
      events,
      {packet = npackets})
end
```

Output:

```

Processed 10.2 million packets in 0.35 seconds (rate: 29.2 Mpps).
EVENT                                             TOTAL     /packet
cycles                                    1,235,445,846     123.545
ref_cycles                                            0       0.000
instructions                              2,733,223,244     273.322
l2_rqsts.all_code_rd                          1,073,783       0.107
l2_rqsts.all_demand_data_rd                  51,378,621       5.138
l2_rqsts.all_demand_miss                      1,534,342       0.153
l2_rqsts.all_demand_references               97,109,913       9.711
l2_rqsts.all_pf                              21,668,114       2.167
l2_rqsts.all_rfo                             44,657,508       4.466
l2_rqsts.code_rd_hit                            770,365       0.077
l2_rqsts.code_rd_miss                           291,776       0.029
packet                                       10,000,000       1.000
load: time: 0.34s  fps: 58,662,064 fpGbps: 32.381 fpb: 510 bpp: 60   sleep: 0   us

Processed 10.0 million packets in 0.34 seconds (rate: 29.5 Mpps).
EVENT                                             TOTAL     /packet
cycles                                    1,193,594,857     119.359
ref_cycles                                            0       0.000
instructions                              2,649,570,794     264.957
l2_trans.all_pf                               4,956,849       0.496
l2_trans.all_requests                       187,960,552      18.796
l2_trans.code_rd                                998,689       0.100
l2_trans.demand_data_rd                      50,124,949       5.012
l2_trans.l1d_wb                              65,830,399       6.583
l2_trans.l2_fill                              4,488,218       0.449
[... 122 lines cut ...]
mem_uops_retired.all_stores                 577,123,036      57.712
mem_uops_retired.lock_loads                          44       0.000
mem_uops_retired.split_loads                        991       0.000
mem_uops_retired.split_stores                         0       0.000
mem_uops_retired.stlb_miss_loads                      0       0.000
mem_uops_retired.stlb_miss_stores                     0       0.000
packet                                       10,000,000       1.000
load: time: 0.35s  fps: 57,998,525 fpGbps: 32.015 fpb: 510 bpp: 60   sleep: 0   us

Processed 10.2 million packets in 0.35 seconds (rate: 29.1 Mpps).
EVENT                                             TOTAL     /packet
cycles                                    1,227,718,482     122.772
ref_cycles                                            0       0.000
instructions                              2,717,272,860     271.727
mem_load_uops_retired.hit_lfb                32,476,430       3.248
mem_load_uops_retired.l1_hit                810,899,999      81.090
mem_load_uops_retired.l1_miss                50,662,084       5.066
mem_load_uops_retired.l2_hit                 50,150,753       5.015
mem_load_uops_retired.l2_miss                         0       0.000
mem_load_uops_retired.l3_hit                          0       0.000
mem_load_uops_retired.l3_miss                         0       0.000
packet                                       10,000,000       1.000
```

No interpretation for now :-). However, this seems like an approach that could be used to gather data for more interesting benchmarks (like `mp-ring` in #804) and that could also be analyzed for interesting patterns and differences with an R script (like #755) to help explain results and validate intuitions.

lukego, 2016-03-08:
cc @xrme @kbara in case this is handy as a reference one day.

lukego, 2016-03-08:
I ran this with `sudo taskset -c 0 ./snabb snsh ./snabbmark_pmu.lua` and actually running this in the lab depends on https://github.com/snabblab/snabblab-nixos/issues/23. I did a sneaky way to enable the `msr` module for PMU on `lugano-1` like this:

```
$ nix-env -i module-init-tools
$ sudo insmod /nix/store/gxk7qq6s8hx2c20dch634xf2bh5hx6l9-linux-4.3.3/lib/modules/4.3.3/kernel/arch/x86/kernel/msr.ko
```

mwiget, 2016-03-08:
any reason you run it on cpu 0? That cpu gets the most work from the kernel by handling all interrupts.

lukego, 2016-03-08:
Finger memory! Looking forward to merging and then following the new recommendations from you and @kbara :)

lukego, 2016-03-09:
I braindumped some thoughts on lukego/blog#15. Basically I am wondering whether the PMU can work as a `wireshark` for catching expensive MESIF interactions (or at least a `netstat -s`). Seems to me like all performance issues around multicore are likely to be interactions with the L3 cache.

lukego, 2016-03-09:
Reflection from lukego/blog#15: I suspect that L3 cache access is the key thing to optimize for any multiprocess application (including Snabb+VM over virtio-net). The L3 cache is "the network" that connects the cores together and has much higher latency than anything they do locally.

Here is a side-by-side comparison of some L3 PMU counters for the `mp-ring` benchmark (#804 #809) comparing single-process operation (left) with multi-process operation (right):

```
[luke@lugano-1:~/git/snabbswitch/src]$ pr -w 160 -m -t a b                                                                                                                         
Benchmark configuration:                                                        Benchmark configuration:
       burst: 100                                                                      burst: 100
  writebytes: 0                                                                   writebytes: 0
   processes: 1                                                                    processes: 2
   readbytes: 0                                                                    readbytes: 0
     packets: 10000000                                                               packets: 10000000
        mode: basic                                                                     mode: basic
   pmuevents: mem_load_uops_l3                                                     pmuevents: mem_load_uops_l3
  69.20 Mpps ring throughput per process                                           5.26 Mpps ring throughput per process
PMU report for child #0:                                                        PMU report for child #0:
EVENT                                             TOTAL     /packet             EVENT                                             TOTAL     /packet
cycles                                      500,751,679      50.075             cycles                                    6,651,599,204     665.160
ref_cycles                                            0       0.000             ref_cycles                                            0       0.000
instructions                                562,015,029      56.202             instructions                                593,294,818      59.329
mem_load_uops_l3_hit_retired.xsnp_hit               447       0.000             mem_load_uops_l3_hit_retired.xsnp_hit           432,740       0.043
mem_load_uops_l3_hit_retired.xsnp_hitm                4       0.000             mem_load_uops_l3_hit_retired.xsnp_hitm        9,410,279       0.941
mem_load_uops_l3_hit_retired.xsnp_miss              280       0.000             mem_load_uops_l3_hit_retired.xsnp_miss            9,189       0.001
mem_load_uops_l3_hit_retired.xsnp_none              567       0.000             mem_load_uops_l3_hit_retired.xsnp_none           11,856       0.001
mem_load_uops_l3_miss_retired.local_dram              0       0.000             mem_load_uops_l3_miss_retired.local_dram              0       0.000
mem_load_uops_l3_miss_retired.remote_dram             0       0.000             mem_load_uops_l3_miss_retired.remote_dram             0       0.000
mem_load_uops_l3_miss_retired.remote_fwd              0       0.000             mem_load_uops_l3_miss_retired.remote_fwd              0       0.000
mem_load_uops_l3_miss_retired.remote_hitm             0       0.000             mem_load_uops_l3_miss_retired.remote_hitm             0       0.000
packet                                       10,000,000       1.000             packet                                       10,000,000       1.000
```

(also as a [gist](https://gist.github.com/lukego/87362ba844baedcd77f3) in case that is easier to read.)

I see two interesting and likely related things here:

1. The single-process version is executing about 10x faster in terms of instructions-per-cycle.
2. The single-process version has no L3 access while the multi-process version has one `xsnp_hitm` event per packet. I believe this event means that the L3 cache served a *hit* from a line that was *m*odified in another core.

From the paper cited in #735 I would expect an L3 `hitm` to have >100 cycles of latency and that could partly explain the whopping 665 cycles/packet performance that the benchmark is reporting. So I predict that if somebody is able to identify that `hitm` and reduce its frequency then that will significantly increase the performance of this benchmark.

Now if we had working PEBS support (#631) hooked into a profiler we could also have a report telling us which instruction is triggering that hitm and which data structure it is accessing at the time. This is firmly on my wishlist for the future.