snabbco/snabb #812: Github PR/issue queries

lukego, 2016-03-09:
Creating this issue to work out what are interesting ways to query Github pull requests and issues. I am particularly interested to have some queries that I could execute e.g. daily or weekly to find upstreaming work that I am responsible for.

First off the cuff:

- [`is:pr is:open no:assignee`](https://github.com/SnabbCo/snabbswitch/pulls?utf8=%E2%9C%93&q=is%3Apr+is%3Aopen+no%3Aassignee) open PRs that are not yet assigned to an upstream.
- [`is:pr is:open base:next`](https://github.com/SnabbCo/snabbswitch/pulls?utf8=%E2%9C%93&q=is%3Apr+is%3Aopen+base%3Anext) open PRs that specify `next` as their base branch.
- [`is:pr is:open -label:merged`](https://github.com/SnabbCo/snabbswitch/pulls?utf8=%E2%9C%93&q=is%3Apr+is%3Aopen+-label%3Amerged) open PRs that are not `merged`.
- [`is:issue is:open label:bug`](https://github.com/SnabbCo/snabbswitch/issues?q=is%3Aissue+is%3Aopen+label%3Abug) open bugs.

The basic thing I am lacking is a way to express time like "in the past week". This can be done with literal dates like `created:>2016-03-01` but that is not bookmark'able.

Could be that we could use queries to more definitely define a division of labor for upstream activities? at least I would quite like a well-understood way to identify my required actions so that I know what to do and other people know how to make me do stuff. For now the best idea I have is to search for open PRs that have one of my branches as the `base:`.

cc @eugeneia @wingo @kbara.

eugeneia, 2016-03-30:
I would like to replace the `[sketch]` and `[wip]` tags with labels so that they can be queried. Can we pull this off? I read that collaborators can apply labels, do people have to be manually elected to be collaborators for public repositories like this one?

eugeneia, 2016-03-30:
I could label PRs opened by people who do not have the required permissions yet, given that this group of people will be small.

lukego, 2016-03-31:
The can of worms here is Github permissions. They are confusing. Historically at least I don't think it is possible to give somebody access to issues without general read/write access to the whole repo. I am not an expert though and this is a moving target lately as Github adds features.

The reason I suggest putting `[wip]` and `[sketch]` in the PR title is that this is under the control of the contributor. Labels are not: only maintainers can manipulate those AFAIK. If there would be a mechanism for contributors to add/remove the `sketch` and `wip` labels from their own PRs then that would be neat but I am not aware of such a possibility.

> I would like to replace the [sketch] and [wip] tags with labels so that they can be queried.

Note: This can already be queried today e.g. `[wip] in:title`. Does that solve the problem anyway?

eugeneia, 2016-04-28:
Actually I need the reverse: show me everything that's not `[wip]`. Can't find docs about title queries... :/

plajjan, 2016-08-04:
@eugeneia "is:issue is:open NOT wip in:title" should do that. Here's an example (using 'NFV' instead of wip because there are no wip issues): https://github.com/snabbco/snabb/issues?utf8=%E2%9C%93&q=is%3Aissue%20is%3Aopen%20NOT%20NFV%20in%3Atitle%20

Note that "NOT" must be all capital - syntax is described here: https://help.github.com/articles/searching-issues/

plajjan, 2016-08-04:
@lukego yeah, no relative date AFAICT but how about sorting by date? append "sort:updated-desc" for latest by update which is what makes sense most of the time.

Where should these queries be documented?