snabbco/snabb #814: Single Snabbnfv traffic process for Multiple VM

harishiitd, 2016-03-10:
I want to pass traffic between two VMs by creating  vmdqs  from the same physical port.

            In the current testcases, each snabbnfv traffic process is created with vmdqs from different  PCI device .
                We can pass multiple ports in nfvconfig file and start a single traffic backend process. 
I tried to launch two VMs with same socket path. In this case, one VM is able to communicate with snabb process other VM is not able to 

How multiple VMs can communicate with this traffic process? Can they (single traffic process and two VMs) communicate using the same socket? Is it possible?

Harish

lukego, 2016-03-11:
Hi Harish,

This is supported and the reason you are confused is a documentation problem.

On the snabbnfv command-line you can specify the socket path like this:

```
/somewhere/%s.socket
```

and the `%s` will be replaced by the `id` for the port in the configuration file. Then you can start your VMs with different paths and the Snabb process will connect to both.

Does that help?

harishiitd, 2016-03-11:
Hi Luke,

   Thanks for the reply, it works perfectly :).
I have a doubt ..
    Does the packet switching between VMs happens in software or through NIC's L2 switch?

Regards
Harish

lukego, 2016-03-11:
Great!

L2 switching, VLAN tag insert/remove, and MAC anti-spoofing are all being done in the NIC by VMDq. The rest is done in software.