snabbco/snabb #825: Formatting function/method/variable documentation

lukego, 2016-03-14:
Question for @eugeneia mostly:

Is it important that we format the documentation for function/method/variable definitions the way that we do? I reflect that there are many possible styles and ours is notably different than the Lua and LuaJIT reference manuals.

Snabb style:

[image]

Lua style:

[image]

LuaJIT style:

[image]

Just wanted to bring this up because I don't really remember the context of how we chose this style and wondered whether this is something we want to revisit while revising the docs or not.

eugeneia, 2016-03-14:
Its only a convention I chose because I think its readable, and useful to note the type of the described object.

lukego, 2016-03-15:
I like that idea for showing return values!

Thanks for clarifying re: history. Some fine day I think we should consider switching to a more Lua-esque format. I find that easier to parse and it seems like there are not so many different kinds of symbols to differentiate between. (Unlike e.g. Lisp universe where you need to carefully distinguish between function / variable / constant / macro / symbol-macro / special-form / compiler-macro / type / ... and where I suspect we picked up this convention from).

lukego, 2016-03-15:
(Can also be that I underestimate the value of the type specifier e.g. for spelling out configuration items, counters, etc. I only say that we should consider switching if we don't have a good reason to be different.)

wingo, 2016-03-15:
I think there's still an open question of how to document methods versus functions, at least for me.  If a data type has a method that is not available as a free procedure -- i.e., if it were only `pkt:free()` and never `packet.free(pkt)` -- then I don't know how to refer to the data type in the manual.

To state it another way: before the parenthesis, to me it seems that all of the names should denote bindings accesible in the normal Lua namespace.  For example `packet.free` is such a binding, `packet` being the conventional way we denote modules and `free` being a property of that module; but in `pkt:free` there is no such thing as `pkt`.  `packet:free` is in some way the worst of all worlds: the `packet` denotes a binding (the module itself), but not one which has a `free` method (unless the `packet` module has a `free` procedure, of course).

wingo, 2016-03-15:
Somewhat related to #740.