snabbco/snabb #850: Maintainers push to snabbco branches directly

lukego, 2016-03-31:
Github has [extended](https://github.com/blog/2137-protected-branches-improvements) their protected branches feature so that you can say who is allowed to push to which branch. This seems very useful: now we could tell Github who is responsible for each branch (e.g. that @kbara owns `kbara-next`) and then everybody could push to their branches directly.

This would allow us to retire an awkward bit of infrastructure i.e. the shell script that I run to mirror everybody's known branches onto the SnabbCo repo. The main reason that script exists is because when people push directly to SnabbCo it has been easy to make awful mistakes like accidentally making a release by pushing to `master`. (I have done this a couple of times in the past and I didn't want those mistakes to scale up linearly with the number of subsystem maintainers.)

Sound good @eugeneia @kbara @wingo @capr? The specific difference is you would publish your long-lived branches (the ones listed in `branches.md`) by pushing them to the SnabbCo repo rather than your own fork (and Github would now helpfully stop you from pushing to other people's branches). If so then I will setup the permissions and we can make the switch.

capr, 2016-03-31:
Ok, let us know when it's ready so we can switch the remote.

lukego, 2016-04-07:
Done!

Now you should please publish your branches by pushing directly to the snabbco repository. These are all "protected branches" which means (a) they don't accept force pushes and (b) each branch has a list of users who are allowed to push to it (I made the assignments that seemed obvious to me, let me know if you have trouble!)

Note that the mirroring script is disabled now so if you push to your own fork then that will not automatically arrive on the snabbco repository.

If you accidentally screw up a public branch and need to force push then please raise an issue in the spirit of #817 and we can work it out together. Github does not seem to have any option to allow force-pushes to protected branches, which is harsh on maintainers (difficult to undo mistakes) but easy on users (won't get their merges messed up by an upstream rebase).

Questions welcome!

cc subsystem maintainers @capr @eugeneia @kbara @wingo @domenkozar

closed 2016-05-13 by eugeneia