snabbco/snabb #853: Snabb crashes on app network reconfig

plajjan, 2016-03-31:
I'm seeing crashes of Snabb when reconfiguring the app network.

We can use this code as example: https://github.com/plajjan/snabbswitch/commit/86c785b1ea339b67275bbf1fa4856a859484bcc7

As can be seen I'm doing three runs of Snabb with different configuration and the apps have unique names. If we try to reuse the same app, i.e. instead of calling it "sink1" in first run and "sink2" in the second, we use the same name, that will lead to a crash of Snabb.

Since the selftest of core.app does reconfig, but with kind of bogus apps, and this works, I assume the problem is with the apps itself. How should an app be rewritten to handle reconfig?

eugeneia, 2016-04-01:
@plajjan Could you add a command to reproduce the crash and attach the error message / stack trace?

plajjan, 2016-04-01:
Are you not able to reproduce this?

Which command?

eugeneia, 2016-04-04:
@plajjan I believe I have managed to reproduce the issue, it is a bug in `core.shm` that is fixed in `max-next` (I assume in the course of the changes made by @wingo in #794).

plajjan, 2016-04-04:
Ah, okay, nice to hear it's fixed already. When is max-next hitting master next time? :)

What's the process here, do we keep issues until it's on master or can we close it as soon as it's on one of the -next branches ?

eugeneia, 2016-04-05:
I will do the release right now, so it should be in master in half an hour or so. :) I tend to keeping issues open until they are fixed on master. This way you can query for a list of open bugs for the current release easily.

eugeneia, 2016-04-05:
Fixed in `v2016.04`.

closed 2016-04-05 by eugeneia